sarxos/webcam-capture #568: record video problem

justingq, 2017-10-24:
When I use java swing to display the camera and record video, the video screen will be stuck.What is the problem?

---

```
package com.sdbank;

import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.File;

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import com.xuggle.mediatool.IMediaWriter;
import com.xuggle.mediatool.ToolFactory;
import com.xuggle.xuggler.ICodec;
import com.xuggle.xuggler.IPixelFormat;
import com.xuggle.xuggler.IVideoPicture;
import com.xuggle.xuggler.video.ConverterFactory;
import com.xuggle.xuggler.video.IConverter;

import javax.swing.JButton;
import javax.swing.JOptionPane;
import javax.swing.JLabel;

import java.awt.event.ActionListener;
import java.awt.event.ActionEvent;
import java.awt.Font;

public class FaceVedio {

	private final static Webcam webCam = Webcam.getDefault();

	private static void initFrame(){
		webCam.setViewSize(WebcamResolution.QVGA.getSize());
		final WebcamPanel panel = new WebcamPanel(webCam);
		panel.setBounds(0, 35, 461, 272);
		panel.setMirrored(true);
		panel.stop();
		final JFrame window = new JFrame("record");
		window.getContentPane().setLayout(null);
		window.getContentPane().add(panel);
		
		JButton btn_vedio = new JButton("record");
		btn_vedio.setFont(new Font("微软雅黑", Font.PLAIN, 12));
		btn_vedio.addActionListener(new ActionListener() {
			public void actionPerformed(ActionEvent e) {
				try {
					if(webCam.isOpen()){
						getVedio();
					}else{
						JOptionPane.showMessageDialog(null, "please open camera");
					}
					
				} catch (InterruptedException e1) {
					e1.printStackTrace();
				}
			}
		});
		btn_vedio.setBounds(191, 321, 93, 30);
		window.getContentPane().add(btn_vedio);
		
		JLabel lab_warning = new JLabel("Please face the camera");
		lab_warning.setFont(new Font("微软雅黑", Font.PLAIN, 14));
		lab_warning.setBounds(176, 10, 123, 15);
		window.getContentPane().add(lab_warning);
		
		JButton btn_open = new JButton("open camera");
		btn_open.setFont(new Font("微软雅黑", Font.PLAIN, 12));
		btn_open.addActionListener(new ActionListener() {
			public void actionPerformed(ActionEvent e) {
				panel.start();
			}
		});
		btn_open.setBounds(10, 321, 108, 30);
		window.getContentPane().add(btn_open);
		
		JButton btn_close = new JButton("close camera");
		btn_close.setFont(new Font("微软雅黑", Font.PLAIN, 12));
		btn_close.addActionListener(new ActionListener() {
			public void actionPerformed(ActionEvent e) {
				panel.stop();
			}
		});
		btn_close.setBounds(346, 321, 108, 30);
		window.getContentPane().add(btn_close);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.setSize(467, 392);
		window.setLocationRelativeTo(null);
		window.setResizable(false);
		window.setVisible(true);
	}

	public static void getVedio() throws InterruptedException{
		File file = new File("E:"+File.separator+"output.mp4");
		IMediaWriter writer = ToolFactory.makeWriter(file.getAbsolutePath());
		Dimension size = WebcamResolution.QVGA.getSize();

		writer.addVideoStream(0, 0, ICodec.ID.CODEC_ID_H264, size.width, size.height);

		long start = System.currentTimeMillis();

		for (int i = 0; i < 80; i++) {

			System.out.println("Capture frame " + i);
			
			BufferedImage image = ConverterFactory.convertToType(webCam.getImage(), BufferedImage.TYPE_3BYTE_BGR);
			IConverter converter = ConverterFactory.createConverter(image, IPixelFormat.Type.YUV420P);

			IVideoPicture frame = converter.toPicture(image, (System.currentTimeMillis() - start) * 1000);
			frame.setKeyFrame(i == 0);
			frame.setQuality(0);

			writer.encodeVideo(0, frame);

			Thread.sleep(10);
		}

		writer.close();

		System.out.println("Video recorded in file: " + file.getAbsolutePath());
		JOptionPane.showMessageDialog(null, "record success");
	}
	
	public static void main(String[] args) {
		initFrame();
	}
	
}
```

sarxos, 2017-10-24:
Can you try, when opening webcam:

```java
public void actionPerformed(ActionEvent e) {
	webCam.open(true); // open webcam in non-blocking mode
	panel.start();
}
```

And when closing:

```java
public void actionPerformed(ActionEvent e) {
	webCam.close();
	panel.stop();
}
```

Please let me know if this fixed your problem.

justingq, 2017-10-24:
sorry , it didn't change . 0_0

justingq, 2017-10-25:
Thanks sarxos. I fixed it with thread. 
When record:

```
public void actionPerformed(ActionEvent e) {
	if(webCam.isOpen()){
		VideoThread vt = new VideoThread(webCam);
		vt.start();	
	}else{
		JOptionPane.showMessageDialog(null,"please open camera");
	}
}
```
class VideoThread :

```
        private Webcam webCam=null; 
	
	public VideoThread(Webcam webcam){
		this.webCam = webcam;
	}
	
	private void getVideo(){
	...
	...
	}
	public void run(){
		getVideo();
	}
```

sarxos, 2017-10-25:
Very nice solution :) Take care!

closed 2017-10-25 by sarxos