sarxos/webcam-capture #570: hey i am getting this issue in my project and i am new in java so can you plss help me

Ayush0601, 2017-10-28:
can you plss help me i am getting this error in my project and i am a beginner

15:08:53.403 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
15:08:53.419 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
15:08:54.372 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Integrated Webcam 0
15:08:54.372 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
15:08:54.372 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Integrated Webcam 0
15:08:54.372 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam Integrated Webcam 0
15:08:54.372 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device Integrated Webcam 0
15:08:54.372 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=640,height=480]
15:08:55.273 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
15:08:55.306 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Clear memory buffer
15:08:55.759 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@31bb2612 is now open
15:08:55.760 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Integrated Webcam 0
15:08:55.824 [main] DEBUG com.github.sarxos.webcam.Webcam - Closing webcam Integrated Webcam 0
15:08:55.832 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamCloseTask - Closing Integrated Webcam 0
15:08:55.833 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Closing webcam device
15:08:56.007 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Unlock Webcam Integrated Webcam 0
15:08:56.023 [webcam-lock-[Integrated Webcam 0]] DEBUG com.github.sarxos.webcam.WebcamLock - Lock updater has been interrupted
15:08:56.023 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam Integrated Webcam 0 has been closed
Exception in thread "main" java.lang.UnsupportedOperationException: Not supported yet.
        at FD$ImagePanel.<init>(FD.java:77)
        at FD$ImagePanel.<init>(FD.java:71)
        at FD.<init>(FD.java:31)
        at FD.main(FD.java:64)

sarxos, 2017-10-29:
Hi, thank you for your interest in Webcam Capture API. Unfortunately I see no problems in regards to webcam related aspects in the logs you provided. Neither ```FD``` nor ```ImagePanel``` are classes used in Webcam Captuer project and therefore I cannot tell you why ```FD$ImagePanel``` constructor throws ```UnsupportedOperationException```. I suggest you to look into the code of this class and find out what is happening there.

Ayush0601, 2017-10-30:
here is my code and i not able to understand what is error in it and why webcam is not opening
can you plsss help me sir to solve this problem and guide what can i do in it

 
import java.awt.FlowLayout;
 import java.awt.image.BufferedImage;
 import java.io.IOException;
 import java.util.Iterator;
 import java.util.List;
 import javax.swing.JFrame;
 import org.openimaj.image.FImage;
 import org.openimaj.image.ImageUtilities;
 import org.openimaj.image.processing.face.detection.DetectedFace;
 import org.openimaj.image.processing.face.detection.HaarCascadeDetector;
 import com.github.sarxos.webcam.Webcam;
 import com.github.sarxos.webcam.WebcamResolution;
import java.awt.Dimension;
 
public class FD extends JFrame 
{
  private static final long serialVersionUID = 1;
  private static final HaarCascadeDetector detector = new HaarCascadeDetector();
  private Webcam webcam = null;
  private BufferedImage img = null;
  private List < DetectedFace > faces = null;

  public FD() throws IOException 
  {
   webcam = Webcam.getDefault();
   webcam.setViewSize(WebcamResolution.VGA.getSize());
   webcam.open();
   img = webcam.getImage();
   webcam.close();
   ImagePanel panel;
      panel = new ImagePanel(img);
   panel.setPreferredSize(WebcamResolution.VGA.getSize());
   add(panel);
   setTitle("Face Recognizer");
   setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
   pack();
   setLocationRelativeTo(null);
   setVisible(true);
  }
  
 public void detectFace() 
 {
   JFrame fr = new JFrame("Discovered Faces");
   faces = detector.detectFaces(ImageUtilities.createFImage(img));
   if (faces == null) 
   {
    System.out.println("No faces found in the captured image");
    return;
   }
   Iterator < DetectedFace > dfi = faces.iterator();
   while (dfi.hasNext()) {
    DetectedFace face = dfi.next();
    FImage image1 = face.getFacePatch();
    ImagePanel p = new ImagePanel(ImageUtilities.createBufferedImage(image1));
  
   }
   fr.setLayout(new FlowLayout(0));
   fr.setSize(500, 500);
   fr.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
   fr.setVisible(true);
  }
  
  public static void main(String[] args) throws IOException {
   new FD().detectFace();
  }

    private void add(ImagePanel panel) {
        throw new UnsupportedOperationException("Templates."); 
    }

    private static class ImagePanel {

        public ImagePanel() {
        }

        private ImagePanel(BufferedImage img) {
            throw new UnsupportedOperationException("Templates."); //To change body of generated methods, choose Tools | Templates.
        }

        private void setPreferredSize(Dimension size) {
            throw new UnsupportedOperationException("Not supported yet."); //To change body of generated methods, choose Tools | Templates.
        }
    }
 }

sarxos, 2017-10-30:
@Ayush0601, 

And here is a tutorial chapter for Swing (this is Java toolkit to create window applications):

https://www.javatpoint.com/java-swing

Please note that Webcam Capture API is heavily using Swing so it's another aspect to be familiar with if you are willing to display video/image feed from webcam in a window.