sarxos/webcam-capture #572: SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".

Rotockter, 2017-11-03:
```
 List<Webcam> webcams = Webcam.getWebcams();
        for (Webcam w c: webcams)
        {
            msg(wc.getName());
        }
```
External Libraries:
bridj-0.7-20140918.jar
slf4j-api-1.7.2.jar
webcam-capture-0.3.10.jar

Get this error, then i try to use the lib.
-----------------------------------------------------------------------------------
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so'
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:1061)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:1038)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:607)
        at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:765)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1016)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:898)
        at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
        at org.bridj.BridJ.initialize(BridJ.java:1117)
        at org.bridj.NativeObject.<init>(NativeObject.java:50)
        at org.bridj.StructObject.<init>(StructObject.java:46)
        at org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:789)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:768)
        at de.rotock_dev.Main.main(Main.java:23)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at java.util.concurrent.FutureTask.report(FutureTask.java:122)
        at java.util.concurrent.FutureTask.get(FutureTask.java:192)
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
        ... 4 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.RuntimeException: Failed to allocate new instance of type class com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
        at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:806)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1016)
        at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:898)
        at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
        at org.bridj.BridJ.initialize(BridJ.java:1117)
        at org.bridj.NativeObject.<init>(NativeObject.java:50)
        at org.bridj.StructObject.<init>(StructObject.java:46)
        at org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        ... 3 more
Caused by: java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/tmp/BridJExtractedLibraries7888148236174114925/OpenIMAJGrabber.so'
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:1061)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:1038)
        at org.bridj.BridJ.getNativeLibrary(BridJ.java:607)
        at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:765)
        ... 13 more

sarxos, 2017-11-03:
Did you check BridJ 0.6.3-SNAPSHOT?

https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.nativelibs4java&a=bridj&v=0.6.3-SNAPSHOT

Rotockter, 2017-11-03:
Switched with "BridJ 0.6.3-SNAPSHOT"  and same Problem. Then I use fswebcam driver, i have no error, but the Image is Black. 

In "motion" i get a Picture. 

But i get not one of the other drivers work correctly. 

```
The following capture drivers are available for 0.3.10:

webcam-capture-driver-vlcj | maven | zip
 java.lang.NoClassDefFoundError: uk/co/caprica/vlcj/discovery/NativeDiscovery

webcam-capture-driver-openimaj | maven | zip
java.lang.ClassNotFoundException: org.openimaj.video.capture.VideoCapture

webcam-capture-driver-lti-civil | maven | zip
java.lang.NoClassDefFoundError: com/lti/civil/CaptureException

webcam-capture-driver-jmf | maven | zip
java.util.concurrent.ExecutionException: java.lang.NoClassDefFoundError: javax/media/Format

webcam-capture-driver-javacv | maven | zip
java.lang.NoClassDefFoundError: org/bytedeco/javacv/FrameGrabber$Exception

webcam-capture-driver-ipcam | maven | zip
Webcam.getDefault() ==  java.lang.NullPointerException

webcam-capture-driver-gstreamer | maven | zip
java.lang.NoClassDefFoundError: org/gstreamer/Gst

webcam-capture-driver-fswebcam | maven | zip
Only Black Image
```

motion output:
```
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motion/motion.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0:motion] [NTC] [ENC] ffmpeg_init: ffmpeg libavcodec version 57.64.101 libavformat version 57.56.101
[0:motion] [NTC] [ALL] main: Camera 0 is from /etc/motion/motion.conf
[0:motion] [NTC] [ALL] main: Camera 0 is device: /dev/video0 input -1
[0:motion] [NTC] [ALL] main: Stream port 8081
[0:motion] [NTC] [ALL] main: Waiting for threads to finish, pid: 2562
[1:ml1] [NTC] [ALL] motion_init: Camera 0 started: motion detection Enabled
[1:ml1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
[0:web_control] [NTC] [STR] http_bindsock: listening on 127.0.0.1 port 8080
[0:web_control] [NTC] [STR] httpd_run: Started motion-httpd server on port 8080 (auth Disabled)
[1:ml1] [NTC] [VID] v4l2_get_capability:
------------------------
cap.driver: "uvcvideo"
cap.card: "USB 2.0 PC Camera"
cap.bus_info: "usb-3f980000.usb-1.2"
cap.capabilities=0x84200001
------------------------
[1:ml1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] v4l2_select_input: - CAMERA
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1:ml1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (0) MJPG (Motion-JPEG)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 0 - Motion-JPEG (compressed : 1) (0x47504a4d)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: (1) YUYV (YUYV 4:2:2)
[1:ml1] [NTC] [VID] v4l2_set_pix_format: 1 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1] [NTC] [VID] v4l2_set_pix_format Selected palette YUYV
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YUYV (320x240)
[1:ml1] [NTC] [VID] v4l2_do_set_pix_format: Using palette YUYV (320x240) bytesperlines 640 sizeimage 153600 colorspace 00000008
[... 32 lines cut ...]
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174927-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174927-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174928-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174928-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174929-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174929-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174930-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174930-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174932-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174933-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174938-00.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174939-01.jpg
[1:ml1] [NTC] [EVT] event_newfile: File of type 1 saved to: /tmp/motion/01-20171103174940-00.jpg
^C[0:web_control] [NTC] [STR] httpd_run: motion-httpd - Finishing
[0:web_control] [NTC] [STR] httpd_run: motion-httpd Closing
[0:web_control] [NTC] [STR] motion_web_control: motion-httpd thread exit

[1:ml1] [NTC] [ALL] motion_loop: End of event 1
[1:ml1] [NTC] [ALL] motion_loop: Thread exiting
[1:ml1] [NTC] [STR] stream_stop: Closing motion-stream listen socket & active motion-stream sockets
[1:ml1] [NTC] [STR] stream_stop: Closed motion-stream listen socket & active motion-stream sockets
[1:ml1] [NTC] [VID] vid_close: Closing video device /dev/video0
[0:motion] [NTC] [ALL] main: Threads finished
[0:motion] [NTC] [ALL] main: Motion terminating
```

sarxos, 2017-11-06:
In log you have:

```
... libjpeg.so.8: cannot open shared object file: No such file or directory
```

Install ```libjpeg8``` and check again with default driver.

Rotockter, 2017-11-06:
It seems it has helped a bit, but errors are still present.

From Maven:
bridj-0.7.0.jar
slf4j-api-1.7.2.jar
v4l4j-0.9.1-r507.jar
webcam-capture-0.3.11.jar
webcam-capture-driver-v4l4j-0.3.11.jar

---------------------------------------------------------------

USB Camera + default Driver:
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Segmentation fault
```
---------------------------------------------------------------

Raspberry Pi Camera + default Driver:
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Segmentation fault
```

---------------------------------------------------------------

USB Camera + v4l4j also not working correct:
`[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN`
black image

---------------------------------------------------------------

Raspberry Pi Camera + v4l4j:
```
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l2-query.c:49 ] Error looking up V4L2 format 34363248. Please submit
a bug report on the v4l4j mailing list.[ v4l2-query.c:484 ] libvideo has skipped an unsupported image format:
[ v4l2-query.c:485 ] H.264 (0x34363248)
[ v4l2-query.c:486 ] Please let the author know about this error.
[ v4l2-query.c:487 ] See the ISSUES section in the libvideo README file.
```
Extreme small image and always the same each  .getImage()

sarxos, 2017-11-13:
Again, did you try with BridJ 0.6.3-SNAPSHOT after installing ```libjpeg8```? I see you had BridJ 0.7 on a deck.

sarxos, 2018-10-08:
Is there any progress with this issue? Can I close it?