sarxos/webcam-capture #574: Multiple camera stream

Miraeld, 2017-11-07:
Good morning, 

I've been trying to adapt the code of an example given into this library in order to be able to stream two USB cameras over internet (JSON method). The truth is, I've been able to do it with one, working fine, but I have a problem when I try to do it with two. It seems to be problem with the Thread, which the server awaits for a connection, and at this moment, the second "camera" does not awaits which means when there is a connection from a client, the client doesn't get the second camera feed which make it impossible to display it. 
Did someone have been able to stream two cameras over internet (Server/Client) ? If Yes, which method did you use ? Thank you for your answers,
Regards,
Miraeld.

sarxos, 2017-11-07:
Are you opening webcam in blocking or non-blocking mode?

Miraeld, 2017-11-07:
Actually I don't really know, 
But I am using the "EntryPoint" project name as base for mine. I am trying to adapt it in order to stream two cameras feed.

sarxos, 2017-11-07:
I have no idea what the EntryPoint project is. Can you be more specific?

sarxos, 2017-11-07:
This is what I get when I search webcam-capture repository for "EntryPoint":

[image]

Miraeld, 2017-11-07:
Okay, 
Alright, I did a mistake. I was using a project from another person based on this API. The example, working with one camera, is available at https://github.com/quangdng/simple-java-webcam-streamer but, when I try to adapt the code in order to stream two cameras, the second one is not working. 
Do you suggest me to look into another solution? If yes, which one ?

Thank you,
Regards.

sarxos, 2017-11-07:
There is an example which demonstrates practical possibility to obtain video feeds from two different cameras at once (1st camera was built-in and the second one was connected through USB port):

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-manycams

The only limitation is USB 2.0 capacity which limits the number of cameras which can be open in the same time. This is usually 3 UVC devices open in the same time and running on VGA resolution. The lower resolution is the more devices you can open in the same time, and vice-versa, the higher resolution is the less devices can be open in the same time.

Or this example which demonstrates practical possibility to stream video feed over WebSocket from several IP cameras:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets

However, in regards to the issue you reported, the bottom line is the following - there is a possibility to open more than one camera in the same time and this functionality is working well. Therefore, if you have any problems with streaming video feed from two different cameras using [simple-java-webcam-streamer](https://github.com/quangdng/simple-java-webcam-streamer), you need to make sure that this specific project is supporting this. I suggest to open new issue in [simple-java-webcam-streamer](https://github.com/quangdng/simple-java-webcam-streamer/issues) and verify this with the project owner.

In regards to your question:

> Do you suggest me to look into another solution? If yes, which one ?

Any solution is good as long as it works fine, but I'm not familiar with any solution that is free of bugs and ready to be used in production. It may be a good opportunity to write your own tool if you would like to use it in e.g. DYI stuff, school project, etc.

Miraeld, 2017-11-09:
Thank you for your answer. Actually, I looked into the "Webcam-capture-websockets" example in order to understand. I tried to use it as well, but, I have a problem. I am not using Eclipse, and there is an import for org.eclipse.jetty.api.**** . I tried to find the .jar of the jetty library but .. I couldn't find it. 
I am using Netbeans, which might be a problem, isn't it ?

Sorry for my stupid question.. 

Regards.

Miraeld, 2017-11-09:
Good afternoon, 

I'll be more specific about the project I am working on, maybe it will help to understand and find a way to achieve my goal. 
I have a Raspberry Pi 3, and I am developing a client/server software. The Raspberry Pi host the Server software and any other computer can have the client software. I did a version of the client software working without any video stream over internet. But it means, my cameras has to be plugged to the client computer in order to work. Now I am looking to improve this part in order to be able to plug my two USB camera to the raspberry pi, and stream over internet. It would allow the user to watch the video stream on the client software from anywhere. 
I tried to adapt the code of simple-java-webcam-streamer but it seems to not be adapted for two cameras streaming. 

So I am looking for a solution allowing the two camera streaming. I tried to use the one Sarxos provided which is Jetty and websockets, available at this link: https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets
But I don't why, I get an error, so I cannot get it working. 

I am stuck on this since a month and half already.. So I am asking for help ..

Thank you,
Regards.

sarxos, 2017-11-10:
Hi @Miraeld, what you described can be done with this project alone. You don't need any 3rd party stuff.

### On Raspberry Pi

This code will start streaming from all detected webcams and expose MJPEG stream on different ports, starting from 38080, for example:

* 1st webcam - port 38080
* 2nd webcam - port 38081
* 3rd webcam - port 38082
* etc...

Just **make sure** to get the newest Webcam Capture API snapshot JAR because I had to fix few things (e.g. binding to network interfaces) which you will need for this to work:

https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.12-SNAPSHOT

You will be able to access video feed by opening ```http://{ip.address}:{port}```, for example:

```
http://192.168.0.52:38080
```

The code is pretty straightforward:

```java
import java.io.IOException;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamStreamer;

public class MiraeldStreaming {

	public static void main(String[] args) throws InterruptedException, IOException {

		int port = 38080; // starting port, incremented it by +1 for every webcam
		int fps = 1; // 1 frame per second (MJPEG is not very fast, keep this in mind)

		for (final Webcam webcam : Webcam.getWebcams()) {
			new WebcamStreamer(port++, webcam, fps, true);
		}

		System.out.println("Press any key to stop webcams streaming...");
		System.in.read();
	}
}
```

### On The Client Side (via Browser, e.g. Firefox or Chrome)

Just open the browser open URL pointing to your Raspberry Pi domain name or IP address, e.g. ```http://192.168.0.52:38080``` and enjoy the video.

I tested this on Firefox running on my Samsung smartphone an it works well:

[image]

_NOTE! On this picture you can see image streamed from my PC webcam and displayed on Samsung phone, not a picture from the smartphone camera._

I can access it from the PC as well:

[image]

### On The Client Side (via Java)

Use IP camera capture driver from here (get ZIP file, all required JARs are in a libs directory in the archive):

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam

Just **make sure** to change the code to provide correct (your) IP addresses. The ones in code are mine and won't work in your case.

The code:

```java
import java.awt.Dimension;
import java.awt.GridLayout;
import java.net.MalformedURLException;
import java.util.ArrayList;
import java.util.List;

import javax.swing.BorderFactory;
import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamPanel.DrawMode;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class MiraeldStreamingClient {

	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public static void main(String[] args) throws MalformedURLException {

		// MAKE SURE to provide correct IP addresses and ports
		IpCamDeviceRegistry.register("RasPi Cam 1", "http://192.168.0.52:38080", IpCamMode.PUSH);
		IpCamDeviceRegistry.register("RasPi Cam 2", "http://192.168.0.52:38081", IpCamMode.PUSH);
		IpCamDeviceRegistry.register("RasPi Cam 3", "http://192.168.0.52:38082", IpCamMode.PUSH);

		JFrame f = new JFrame("Miraeld Streaming Client");
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		f.setLayout(new GridLayout(0, 2, 1, 1));

		List<WebcamPanel> panels = new ArrayList<WebcamPanel>();

		for (Webcam webcam : Webcam.getWebcams()) {

			WebcamPanel panel = new WebcamPanel(webcam, new Dimension(256, 144), false);
			panel.setDrawMode(DrawMode.FIT);
			panel.setFPSLimited(true);
			panel.setFPSLimit(0.2); // 0.2 FPS = 1 frame per 5 seconds
			panel.setBorder(BorderFactory.createEmptyBorder());

			f.add(panel);
			panels.add(panel);
		}

		f.pack();
		f.setVisible(true);

		for (WebcamPanel panel : panels) {
			panel.start();
		}
	}
}
```

It result in:

[image]

You may notice that there are errors in two of my cameras (RasPi Cam 2, RasPi Cam 3). That's because I have only one camera connected to my streamer. That is why the other two gives me an error in webcam panels.

The code is very simple. Just **make sure** you won't mess with dependencies (JARs) and use the ones which are provided in ZIP archives and not the ones you found somewhere in the Internet, because JARs versions from internet will most likely be different and you will and up with nothing.

closed 2017-11-10 by sarxos

Miraeld, 2017-11-13:
Good morning,

Thank you for your explanations, really clear. It will help a lot. 
But, even when every libraries are included, I have an error on these imports:
`import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;`

Which I don't understand the reason, because everything are included.
[image]

sarxos, 2017-11-13:
@Miraeld, that is's great! Good luck with your project!