sarxos/webcam-capture #575: Access to CCTV Cameras

Cabusa123, 2017-11-07:
Hello.

is there a way to access to a CCTV camera with this API?

Im currently working with your API and im using webcams to test the software functionality, but actually i need to get access to cctv cameras in order to finish it, is there a way to do that with your API?

coolmanbat2, 2017-11-07:
Are you connecting them through wirelessly or wired? It should detect the cameras either way I believe.

sarxos, 2017-11-07:
@Cabusa123, you can access CCTV as long as your CCTV cameras expose video feed as a MJPEG stream.

@coolmanbat2, in case of IP cameras the Webcam Capture API is unable to detect them automatically. You have to provide URL into the capture driver for them to work.

sarxos, 2017-11-13:
@Cabusa123, I think I answered you question in another issue, namely in https://github.com/sarxos/webcam-capture/issues/579. Therefore I'm closing this one.

closed 2017-11-13 by sarxos