sarxos/webcam-capture #580: This lib works for web apps like Spring Boot?

brunochichava, 2017-11-11:
Hi, this lib works for java web? If yes how can i implement it?

Thanks in advance.

sarxos, 2017-11-11:
Hi @brunochichava,

Your question is too general.

First of all - almost every Java library (including this one) can work in Java web app because of the very simple reason - Java web app is just a normal Java application, but running on the server side and exposing some abstract API (e.g. SOAP, REST, HTTP, etc) for the clients over the network.

I would also like to note that Spring Boot is not a web app as you stated in the title. Spring Boot is application launcher for applications which uses Spring framework. You can use Spring Boot in Java web application in the same way as in Java desktop application.

If you are asking if you will be able to access webcams of users who access you web app, then the answer is no. This library can work with webcams connected to the computer which is running the process (in case of Java web app this would be the server).

Here is example I wrote some time ago which starts Jetty and expose images from webcams connected to the server with WebSocket interface.

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-websockets

I also played with ```HttpServlet``` in the past to expose MJPEG stream over the network and it was working well.

In regards to your question:

> how can i implement it?

There is a load of examples on the [project main page](https://github.com/sarxos/webcam-capture).

brunochichava, 2017-11-11:
> webcam-capture-examples/webcam-capture-websockets

Many thanks for your answer and for the clarification, i want a way to
connect a java app using Spring Boot with webcam, i will keep researching.

Thanks.