sarxos/webcam-capture #582: Image processing

Miraeld, 2017-11-14:
Good morning,

First of all, thank you for solving my previous problem with the cameras streaming, it works perfectly. 
But now I have a new question. I plan to do some image processing, and I need to analyse each pixels of the current image. Unfortunately, I tried different solution, but it always take a lot of time. Which is not good for my project. Is there someone with an idea of a fast way to do so. Mainly convert the image from RGB to Grayscale, and then to binary ... 
All of that is to implement a celestial object tracking system. 

Thank you for your help,
Regards,
Miraeld.

sarxos, 2017-11-14:
Hi @Miraeld,

In regards to your question. This is how I did binarization in one of my other projects. Not sure if you can call it "fast" because it takes ~70ms on VGA image (640x480), however you can speed it up by reusing ```BufferedImage``` instances in a long running process (see second argument to _filter_ method). I'm using image filters by Jerry Huxtable:

http://www.jhlabs.com/ip/filters/

Code samples can be found below. I'm using 128 as a threshold value (1/2 of max luminance) for binarizing filter, but you need to choose value which is more appropriate in your use case.

```java
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;

import javax.imageio.ImageIO;

import com.github.sarxos.webcam.Webcam;
import com.jhlabs.image.ThresholdFilter;

public class MiraeldProcessing {

	public static void main(String[] args) throws IOException {

		// threshold is a value in range of [1, 255]
		final ThresholdFilter binarizer = new ThresholdFilter(128);
		final Webcam webcam = Webcam.getDefault();
		webcam.open();

		final BufferedImage image = webcam.getImage();
		final BufferedImage binary = binarizer.filter(image, null);

		ImageIO.write(binary, "PNG", new File("binary.jpg"));
	}
}
```

In my case doing grayscale conversion was a waste of time since I needed only binary, but if you need grayscale, then you can use other filter:

```java
import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;

import javax.imageio.ImageIO;

import com.github.sarxos.webcam.Webcam;
import com.jhlabs.image.GrayscaleFilter;
import com.jhlabs.image.ThresholdFilter;

public class MiraeldProcessing {

	public static void main(String[] args) throws IOException {

		// threshold is a value in range of [1, 255]
		final ThresholdFilter binarizer = new ThresholdFilter(128);
		final GrayscaleFilter grayscaler = new GrayscaleFilter();
		final Webcam webcam = Webcam.getDefault();
		webcam.open();

		final BufferedImage image = webcam.getImage();
		final BufferedImage gray = grayscaler.filter(image, null);
		final BufferedImage binary = binarizer.filter(gray, null);

		ImageIO.write(gray, "PNG", new File("gray.jpg"));
		ImageIO.write(binary, "PNG", new File("binary.jpg"));
	}
}
```

Files written to disk:

[image]

[image]

sarxos, 2017-11-14:
@Miraeld, 

Just a note. You need to add this JAR into the classpath to make it work:

http://repo1.maven.org/maven2/com/jhlabs/filters/2.0.235-1/filters-2.0.235-1.jar

Or this dependency if you are using Maven:

```xml
<dependency>
    <groupId>com.jhlabs</groupId>
    <artifactId>filters</artifactId>
    <version>2.0.235-1</version>
</dependency>
```

sarxos, 2017-11-16:
@Miraeld, the other option, and probably much faster, would be to use ```webcam.getImageBytes()``` and work on bytes instead of image, but this may be harder.

Miraeld, 2017-11-17:
Good morning, 
Thanks a lot, you really helped me for my project.

sarxos, 2017-11-17:
The pleasure is mine :)

I'm closing this ticket since this is resolved. Please reply here in case of any additional questions in regards to the topic.

closed 2017-11-17 by sarxos