sarxos/webcam-capture #583: GetDault)' throws exception

rubenkon, 2017-11-14:
Just at the initialisation of the Webcam I get the below exception.
It seems to happen when webcams cannot be found. If trhe webcam can be found everything works fine. I would not expect the exception to be thrown.... I want to test for null in case the webcam is not found..

The first call is:  webcam = Webcam.getDefault();
log4j:WARN No appenders could be found for logger (com.github.sarxos.webcam.Webcam).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
nov 14, 2017 11:21:50 AM uk.chromis.pos.customers.CustomersView <init>
SEVERE: null
com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:879)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:856)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:834)
        at .... ()

sarxos, 2017-11-14:
Can you please provide whole stack trace? From the stack fragment you posted I can see that it's unable to list webcams, but I can't see why, because the stack trace is not complete.

namitsharma99, 2017-11-14:
Hi @sarxos  

Sorry for jumping into the discussion.
I have just become your follower and trying to explore this wonderful code.

I am stuck on a similar issue.

While trying to run the code on Dell Windows machine, 
 >> Webcam webcam = Webcam.getDefault(); 
 >> using these in pom.xml for driver related search, but no help 

>>  I got this log - 
_01:22:23.254 [main] INFO  c.g.sarxos.webcam.WebcamDriverUtils - Searching driver: com.github.sarxos.webcam.ds.openimaj.OpenImajDriver
01:22:23.273 [main] INFO  c.g.sarxos.webcam.WebcamDriverUtils - Searching driver: com.github.sarxos.webcam.ds.civil.LtiCivilDriver
01:22:23.273 [main] INFO  c.g.sarxos.webcam.WebcamDriverUtils - Searching driver: com.github.sarxos.webcam.ds.jmf.JmfDriver
01:22:23.275 [main] INFO  c.g.sarxos.webcam.WebcamDriverUtils - Webcam driver has been found: com.github.sarxos.webcam.ds.jmf.JmfDriver
Exception in thread "main" com.github.sarxos.webcam.WebcamException: No webcam available in the system
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:252)_

And also, no webcam is found -
	System.out.println(">>>> "+Webcam.getWebcams());
	output = []

Though, the code is working fine on Lenovo Fedora machine.

Looks like OS related issue so far, but no workaround.

Request you to share your suggestions.

Regards
Namit

sarxos, 2017-11-15:
Hi @namitsharma99, 

From the log I see that ```JmfDriver``` is used. Are you using it on purpose or it "just happen"? This specific driver requires [JMF](http://www.oracle.com/technetwork/java/javase/download-142937.html) from Oracle to be installed in operating system.

If you are not trying to do something unusual I would recommend you to use default driver. Just do not include any additional drivers into the classpath and default will be selected by default.

I think we can work this out. Can you please share the list of JARs you are using?

sarxos, 2017-11-15:
@namitsharma99,

Just remove these two dependencies from your POM:

```
dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture-driver-ipcam</artifactId>
			<version>0.3.4</version>
		</dependency>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture-driver-jmf</artifactId>
			<version>0.3.9</version>
		</dependency>
```

Also, please use the newest stable version of webcam capture which is 0.3.11. This is valid for drivers as well if you want to use non-default one. All of them need to be 0.3.11 when main JAR is 0.3.11.

namitsharma99, 2017-11-15:
Hi @sarxos 
I added the drivers' dependencies in pom.xml intentionally, thinking that the drivers are not being picked up during the first time when getDefault() was not able to find the system camera.
I will try again with the new version as you have suggested and will update back here.

Thanks & Regards.

sarxos, 2017-11-15:
@namitsharma99,

Please enable debug logging as described here, if possible. This will give us better insight in what is really happening in your runtime.

https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Debug-Logs

namitsharma99, 2017-11-15:
Thanks a lot @sarxos 

This time I just added the 0.3.11 version dependency, and it worked fine for me.

<dependency>
	<groupId>com.github.sarxos</groupId>
        <artifactId>webcam-capture</artifactId>
	<version>0.3.11</version>
</dependency>

Exploring further. 

Thanks & Regards.

sarxos, 2017-11-16:
Great :) Good luck with your project!