sarxos/webcam-capture #584: Need help with video overlay?

zzzsharazzz, 2017-11-16:
My question is: Can I use a video as a background, then put the Webcam preview on top of it?

_ I have a project which allows the user to stand in front of a green background, then the Webcam preview will show him/her with a video playing in the back (chromakey effect, like they do on weather forecast channels).
_ I already read through all the examples and wonder if the above is possible or not. Anyone with experience can help me?

sarxos, 2017-11-17:
@zzzsharazzz,

I wanted to answer "yes, you can", but I tried to implement this with [vlcj](https://github.com/caprica/vlcj) by putting [```EmbeddedMediaPlayerComponent```](https://github.com/caprica/vlcj/blob/master/src/main/java/uk/co/caprica/vlcj/component/EmbeddedMediaPlayerComponent.java) below [```WebcamPanel```](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/WebcamPanel.java) and making the webcam panel non-opaque (with some small changes in the code to support translucent painting). It would normally work fine, but no matter the effort, is was impossible :(

Everything was working almost as it should, but the webcam panel was not transparent when put on top of video panel.

I spent some time yesterday to play with the code and do some google searches and I found out that there are two kinds of graphical components in Java:

* Heavyweight - this is whole Java AWT including classes like ```Canvas```, ```Panel```, etc.
* Lightweight - this is whole Swing (based on AWT though) including classes like ```JPanel```, ```JLabel```, etc.

In general - the [vlcj](https://github.com/caprica/vlcj) video rendering panel is created from heavyweight component and, from the other hand, the webcam panel from Webcam Capture API is a lightweight component. And the problem is the following - when heavy component is put under the light component, then the light component **will no be transparent** (source: [link 1](https://stackoverflow.com/questions/4759983/transparent-jpanel-over-canvas-vlcj), [link 2](http://www.oracle.com/technetwork/articles/java/mixing-components-433992.html)).

[image]

The bottom line is that:

* It's impossible with predefined components of vlcj, but
* It's probably possible with direct rendering which can be summarised by the following: first you get ```BufferedImage``` from video and then get another ```BufferedImage``` from webcam and then replace some specific color range with alpha and finally paint the second one on the first one. This will get you want you need.

If you find some other framework which render video on lightweight component then please let me know so I implement necessary changes into the Webcam Capture API to support translucent panel painting.

zzzsharazzz, 2017-11-20:
Thanks a lot for your reply.
Yes I also tried the vlcj method before but as you said, I couldn't make it work. I'm still looking for a way to do this, will post it here when I'm done.
By the way, I really love your work, hope you could make it even better.

abv2017, 2017-11-22:
Hello, just want to update, I finally did this using your suggestions, maybe it will help others like me.

I use [JavaCV](https://github.com/bytedeco/javacv), this library has the class [FFmpegFrameGrabber](https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/FFmpegFrameGrabber.java) which is what I need to grab the frames and convert them to `BufferedImage`, then mix with the `BufferedImage` returned from Webcam and that's it.

The result is pretty smooth and I think it's better than translucent panel painting. Thanks a lot @sarxos

zzzsharazzz, 2017-11-22:
Hello, just want to update, I finally did this using your suggestions, maybe it will help others like me.

I use [JavaCV](https://github.com/bytedeco/javacv), this library has the class [FFmpegFrameGrabber](https://github.com/bytedeco/javacv/blob/master/src/main/java/org/bytedeco/javacv/FFmpegFrameGrabber.java) which is what I need to grab the frames and convert them to `BufferedImage`, then mix with the `BufferedImage` returned from Webcam and that's it.

The result is pretty smooth and I think it's better than translucent panel painting. Thanks a lot @sarxos

sarxos, 2017-11-22:
Hi @zzzsharazzz,

Thank you for the feedback! So you finally used direct rendering approach. I had a hunch this will be the best approach in terms of this issue.

Just a side note for the ones who would like to do it with vlcj instead of JavaCV - the same is also possible with vlcj by using [```getSnapshot(int, int)```](http://caprica.github.io/vlcj/javadoc/3.1.0/uk/co/caprica/vlcj/player/MediaPlayer.html#getSnapshot(int,%20int)) method from [```EmbeddedMediaPlayer```](http://caprica.github.io/vlcj/javadoc/3.1.0/uk/co/caprica/vlcj/player/embedded/EmbeddedMediaPlayer.html), but this specific method saves snapshot to a temporary file before it's returned as ```BufferedImage``` and therefore cause I/O to be a bottleneck in this solution (which will effectively cause video FPS degradation).