sarxos/webcam-capture #586: Java 9

trauchhaupt, 2017-11-21:
Running with Java 9 does not work, as
* The jar file (and therefore the automatic module) is not an open module, but wants BridJ to load the DLLs from itself
* BridJ is not Java 9 compatible, as the loading of the resources (dlls) uses the wrong classloader

sarxos, 2017-12-05:
Hi @trauchhaupt, thank you for reporting this, but I'm afraid I can't move forward with Java 9 before fix for BridJ is available. I would love to take a look a this bug it on my own, but with current workload on my head I have no time to play with bugs :( I'm sure, however, that Olivier (BridJ owner) will love to see pull request with a fix for this issue :)

msgilligan, 2019-04-29:
BridJ is looking like it is unmaintained. Are there any plans to migrate to something newer or that is maintained?

msgilligan, 2019-04-30:
So, to be clear Java 9+ will work if webcam-capture is loaded on the classpath. (At least that what I'm seeing in my basic testing.) This issue is about not working with the Java Platform Module System. Right?

msgilligan, 2019-04-30:
So, unmerged [BridJ PR #103](https://github.com/nativelibs4java/BridJ/pull/103) seems to fix the problem for me.

I forked BridJ and created a https://jitpack.io build that I successfully used to build and run webcam-capture on JDK 11/12 in a modular app.

Here are the relevant bits of my `build.gradle`:

```
repositories {
    maven { url 'https://oss.sonatype.org/content/groups/public' }
    maven { url 'https://jitpack.io' }
}

dependencies {
    implementation "com.github.msgilligan:BridJ:msgilligan-master-SNAPSHOT"

    implementation ("com.github.sarxos:webcam-capture:0.3.12") {
       // The exclude is necessary so we don't have 2 versions of the same packages
        exclude group: 'com.nativelibs4java', module: 'bridj'
    }
}

```

msgilligan, 2019-08-23:
@sarxos -- let's make a fork/release of my patch to **BridJ**! Are you game?

msgilligan, 2019-10-04:
OK, I've created a fork of **BridJ** in the **ConsensusJ** organization:

https://github.com/ConsensusJ/BridJ

This has PR #103 and also pulls in essential config from what was previously a parent POM. It can be accessed through `jitpack.io`, but I can push it to a Bintray maven repo after a little more testing.

doct0rX, 2023-04-01:
any answer regarding this issue till now? I'm trying to use the library in modular gradle application and it seems it doesn't support JPMS yet.

Vampire, 2024-06-29:
Finally my PR to BridJ was merged, now they "just" need to do a release.