sarxos/webcam-capture #589: Captured image squished in 320x240

neilyoung, 2017-11-27:
I'm using a brand new Logitech C920 HD. The thing worked out of the box using your excellent packages and the inbuilt camera driver. However, if I go down to 320 x 240 resolution, the aspect ratio is no longer maintained and the image taken looks a bit like squished. VGA resolution as well as the QCIF looks OK.

Any idea?

sarxos, 2017-12-05:
Hi @neilyoung,

What OS are you using? Can you list all formats supported by your webcam? In past I assumed that all cameras support 320x240, but this may be wrong... If this is a case then you can find exact resolutions supported by your webcam and register them with ```setCustomViewSizes(Dimension[])``` (from ```Webcam``` class).

neilyoung, 2017-12-05:
Sure. OS is Mac OS High Sierra 10.13.1.

Here some tests:

Supported dimensions are:

INFO - Supported size: 176 x 144
INFO - Supported size: 320 x 240
INFO - Supported size: 640 x 480

However, the USB camera is capable to record HD, but for that I need to feed `webcam.setCustomViewSizes`

The test code:

```

        try {
            SimpleLayout layout = new SimpleLayout();
            ConsoleAppender consoleAppender = new ConsoleAppender( layout );
            logger.addAppender( consoleAppender );
            FileAppender fileAppender = new FileAppender( layout, "logs/logfile.log", false );
            logger.addAppender( fileAppender );
            logger.setLevel( Level.ALL );

        } catch( Exception ex ) {
        }

        List<Webcam> l = Webcam.getWebcams();
        final Webcam webcam = Webcam.getDefault();

        for (Dimension supportedSize : webcam.getViewSizes()) {
            logger.info(String.format(Locale.US, "Supported size: %.0f x %.0f", supportedSize.getWidth(), supportedSize.getHeight()));
        }
        
        webcam.setViewSize(WebcamResolution.QVGA.getSize());
 
        final WebcamPanel panel = new WebcamPanel(webcam);
        panel.setFPSDisplayed(true);
        panel.setImageSizeDisplayed(true);
        panel.setMirrored(true);
        final JCheckBox flip = new JCheckBox();
        flip.setSelected(true);
        flip.setAction(new AbstractAction("Flip") {

            @Override
            public void actionPerformed(ActionEvent e) {
                panel.setMirrored(flip.isSelected());

            }
        });
        final JComboBox<String> comboBoxCamera= new JComboBox<String>();
        for (Webcam cam : l) {
            comboBoxCamera.addItem(cam.getName());
        }

        JFrame window = new JFrame("Test webcam panel");
        window.setLayout(new FlowLayout());
        window.add(panel);
        window.add(flip);
        window.add(comboBoxCamera);
        window.setResizable(true);
        window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
        window.pack();
        window.setVisible(true);

```

INFO - Supported size: 176 x 144
INFO - Supported size: 320 x 240
INFO - Supported size: 640 x 480

Here some reference images with a roll of tape as circular reference. The last image has been taken with the internal camera of the MacBook. You see the squeeze in the first image, taken from the webcam in 320x240.

[image]

[image]

[image]

neilyoung, 2017-12-05:
Could it be because Logitech doesn't claim to support 4:3 but 16:9 only? If that's the case I'm wondering, why 640x480 is working

from http://support.logitech.com/en_us/product/hd-pro-webcam-c920/specs

Image Capture (4:3 SD) | N/A
-- | --
Image Capture (16:9 W) | 2.0 MP, 3 MP*, 6 MP*, 15 MP*
Video Capture (4:3 SD) | N/A
Video Capture (16:9 W) | 360p, 480p, 720p, 1080p

Image Capture (4:3 SD) | N/A
-- | --
Image Capture (16:9 W) | 2.0 MP, 3 MP*, 6 MP*, 15 MP*
Video Capture (4:3 SD) | N/A
Video Capture (16:9 W) | 360p, 480p, 720p, 1080p