sarxos/webcam-capture #593: Ambiguity warning

neilyoung, 2017-12-05:
While running my Java project, which uses webcam and native code using OpenCV I'm getting this at runtime printed on console after 

`
final Webcam webcam = Webcam.getDefault();
`

`objc[1337]: Class CaptureDelegate is implemented in both /usr/local/lib/libopencv_videoio.3.3.dylib (0x12917b918) and /var/folders/l6/_fk71njs2r52jbvp4b9y52_h0000gp/T/BridJExtractedLibraries8002541723609415571/libOpenIMAJGrabber.dylib (0x10f6a59e0). One of the two will be used. Which one is undefined.
`

Any ideas what this might mean? And how to prevent?

neilyoung, 2017-12-05:
As a consequence it seems I'm not getting called back 

Following your example: 
https://github.com/sarxos/webcam-capture/issues/581

Having done this:

        mCameraAPI = Webcam.getDefault();
        mCameraAPI.setViewSize(WebcamResolution.VGA.getSize());
        mCameraAPI.setImageTransformer(new WebcamImageTransformer() {
            @Override
            public BufferedImage transform(BufferedImage bufferedImage) {
                return null;
            }
        });
        mCameraAPI.open();

The expectation is to get called back with every frame. Should that happen?

sarxos, 2017-12-06:
Hi @neilyoung,

It says that implementation of ```CaptureDelegate``` exist in two different native binaries:

* libOpenIMAJGrabber.dylib
* libopencv_videoio.3.3.dylib

One comes from OpenIMAJ extracted by BridJ and the other comes from OpenCV. Excluding BridJ from dependencies should be enough IMO, so you can try adding BridJ exclusion into your ```pom.xml```:

```xml
<dependency>
	<groupId>com.github.sarxos</groupId>
	<artifactId>webcam-capture</artifactId>
	<version>${project.version}</version>
	<exclusions>
		<exclusion>
			<groupId>com.nativelibs4java</groupId>
			<artifactId>bridj</artifactId>
		</exclusion>
	</exclusions>
</dependency>
```

If you are not using Maven then simply get rid of BridJ from your classpath.

Please also make sure to set ```JavaCvDriver``` at the very beginning, ideally in static block in your main class, to prevent default driver from being initialized.

```java
static {
  Webcam.setDriver(new JavaCvDriver());
}
```

sarxos, 2017-12-06:
@neilyoung I forgot to ask you - which driver are you using? Default or JavaCvDriver? Because when you exclude BridJ you will be using JavaCvDriver, but this may not be what you need.

neilyoung, 2017-12-06:
@sarxos Default

neilyoung, 2017-12-06:
@sarxos Well, I'm not using maven to build, I'm using the dependency management of IntellijIdea. Indeed, once I add a dependency to com.github.sarxos:webcam-capture:0.3.112, these JARs are added automatically:

bridj-0.7.0.jar
slf4j-api-1.7.2.jar
webcam-capture-0.3.11.jar

If I remove the first manually I'm catching an exception like so:

```
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:874)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:937)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:914)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:892)
	at com.mycompany.myproject.sdk.myclass.<init>(myclass.java:81)
	at com.mycompany.myproject.myclassMain.preparemyclass(myclassMain.java:185)
	at com.mycompany.myproject.myclassMain.<init>(myclassMain.java:263)
	at com.mycompany.myproject.myclassMain.main(myclassMain.java:301)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 8 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NoClassDefFoundError: org/bridj/cpp/CPPObject
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Caused by: java.lang.ClassNotFoundException: org.bridj.cpp.CPPObject
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 17 more

This happens as a direct response to Webcam.getDefault()

Regards

```

sarxos, 2017-12-06:
@neilyoung 

I see. In this case, if you are using default driver, removing BridJ is not an option. There are two ways to solve your issue imo:

1) Remove OpenCV from your Mac, or
2) Prevent BridJ from loading libopencv_videoio.3.3.dylib.

Option 1) is obvious, but let us discuss option 2).

I took a brief look at the BridJ source code and I found the following lines:

https://github.com/nativelibs4java/BridJ/blob/f221b3650ebe0d74931898dbaa88b5fe18b2d10c/src/main/java/org/bridj/BridJ.java#L697-L702

As we can see in the code there are two ways of adding library to the BridJ search paths. First is by setting ENV variable and the second one is by using -D property of Java. Let's start by finding which property/ENV variable is causing OpenCV dylib to be added to the BridJ:

```java
System.out.println("java.library.path ---------------- ");
System.out.println(System.getProperty("java.library.path");
System.out.println("sun.boot.library.path ---------------- ");
System.out.println(System.getProperty("sun.boot.library.path");
System.out.println("gnu.classpath.boot.library.path ---------------- ");
System.out.println(System.getProperty("gnu.classpath.boot.library.path"");
```

And for ENVs:

```java
System.out.println("LD_LIBRARY_PATH ------------------ ");
System.out.println(System.getenv("LD_LIBRARY_PATH"));
System.out.println("DYLD_LIBRARY_PATH ------------------ ");
System.out.println(System.getenv("DYLD_LIBRARY_PATH"));
System.out.println("PATH ------------------ ");
System.out.println(System.getenv("PATH"));
```

Run these snippets and take a close look at the output. Your goal is to find out which one of them is adding the following directory (with file, or without file):

```
/usr/local/lib
```

Right now all properties and ENV variables which contains this directory need to be modified (but **not** permanently). Property can be reset very simple by setting it to some empty value. This need to be done in a static block at the very beginning of your main class:

```java
static {
  System.setProperty("property.name", ""); // replace property.name with a real property
}
```

The ENV is a bit more tricky. If you are using InteliJ then you have to edit your Run/Debug Configuration (somewhere in menu) and set ENV variable to empty value. Thing's are more tricky when you are running your application from console because you will have to save previous values of ENV variables, run your application, and then change ENV variable to what it was before.

neilyoung, 2017-12-06:
@sarxos Thanks for your patience. Currently this problem is not that important for me, because I don't have to use the image transformer callback (as I intended to to first). I'm happy with what getImageBytes delivers on a separate thread.

However, in order to find the root of this problem, let's continue to analyze:

Here is the output of the snippets above:

```
java.library.path ---------------- 
./lib
sun.boot.library.path ---------------- 
/Library/Java/JavaVirtualMachines/jdk1.8.0_60.jdk/Contents/Home/jre/lib
gnu.classpath.boot.library.path ---------------- 
null

LD_LIBRARY_PATH ------------------ 
null
DYLD_LIBRARY_PATH ------------------ 
null
PATH ------------------ 
/Users/me/.atom/packages/platformio-ide/penv/bin:/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/git/bin:/usr/local/heroku/bin:/Users/me/tools:/Users/me/.rvm/gems/ruby-2.2.4/bin:/Users/me/.rvm/gems/ruby-2.2.4@global/bin:/Users/me/.rvm/rubies/ruby-2.2.4/bin:/opt/local/bin:/opt/local/sbin:/opt/subversion/bin/:/Users/me/android-sdk-mac_x86/tools:/Users/me/android-sdk-mac_x86/platform-tools:/usr/local/bin/:/cal/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/share/dotnet:/usr/local/git/bin:/Library/Frameworks/JRuby.framework/Versions/Current/bin:/Users/me/.rvm/bin:/Users/me/bin:/Users/me/Documents/ESP-32/xtensa-esp32-elf/bin:/usr/local/Cellar/mysql/5.6.27/bin

```

AFAIK there is no `/usr/local/lib ` entry, isn't it?

sarxos, 2017-12-06:
Dammit. Yes, indeed, there is no ```/usr/local/lib``` :(

sarxos, 2017-12-06:
Another dive into BridJ code and I found it:

https://github.com/nativelibs4java/BridJ/blob/f221b3650ebe0d74931898dbaa88b5fe18b2d10c/src/main/java/org/bridj/BridJ.java#L734

Unfortunately this one cannot be workaround ;(

sarxos, 2017-12-06:
But lets go back to your initial comment:

> Having done this:

```
    mCameraAPI = Webcam.getDefault();
    mCameraAPI.setViewSize(WebcamResolution.VGA.getSize());
    mCameraAPI.setImageTransformer(new WebcamImageTransformer() {
        @Override
        public BufferedImage transform(BufferedImage bufferedImage) {
            return null;
        }
    });
    mCameraAPI.open();
```

> The expectation is to get called back with every frame. Should that happen?

No. This won't be called with every frame. This will be called every time you invoke ```getImage()```. In async mode (when open(true) is used) this will be, indeed, invoked on every frame, because webcam updater silently call getImage() in a background.

If you want this method to do nothing you should return source image instead of null:

```
    mCameraAPI.setImageTransformer(new WebcamImageTransformer() {
        @Override
        public BufferedImage transform(BufferedImage bufferedImage) {
            return bufferedImage;
        }
    });
```

sarxos, 2017-12-06:
In regards to the warning I think you will either have to live with it or remove OpenCV.

neilyoung, 2017-12-06:
@sarxos OK, thanks for clarification. As said, I'm happy with what I have now and I can live with the warning.

Thanks

closed 2017-12-06 by neilyoung

mdagraca, 2018-03-06:
Hi, I'm having this error. Did you find a solution? I'm using opencv openpnp and I can not remove it.
The error message is:
objc[6899]: Class CaptureDelegate is implemented in both /private/var/folders/j1/_q1t_00d0fg66bsgtgf8zjgc0000gp/T/opencv_openpnp5786236370894680416/nu/pattern/opencv/osx/x86_64/libopencv_java320.dylib and /var/folders/j1/_q1t_00d0fg66bsgtgf8zjgc0000gp/T/BridJExtractedLibraries4705987337836921372/libOpenIMAJGrabber.dylib. 
One of the two will be used. Which one is undefined.
Thanks