sarxos/webcam-capture #594: Capturing image from webcam currently displaying input stream

reginavaleria96, 2017-12-06:
I am working on a web app and it involves displaying an video input stream from webcam and taking/capturing photos from the same webcam. I successfully get access to the webcam and display it on the web page using JavaScript getUserMedia function and a <video> tag on HTML5. Then I need a button to allow user capture and save the image, so I involved your library in the servlet. 
```
public void capture(String path) throws IOException {
		Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
		} else {
			System.out.println("No webcam detected");
}
		webcam.setViewSize(new Dimension(640,480));
		BufferedImage image = webcam.getImage();
		ImageIO.write(image, "JPG", new File(path));
		System.out.println("Captured & Saved");
		webcam.close();
	}
}
```
However my trial result in image==null, while webcam is not null. Is there any way to capture an image from a webcam that's currently already on?

neilyoung, 2017-12-06:
Missing webcam.open()

neilyoung, 2017-12-06:
@reginavaleria96 Hmm. Sorry, no

sarxos, 2017-12-06:
Hi @reginavaleria96,

We have to consider two different cases:

1) You are running servlet and the browser on the **same** computer, or
2) You are running servlet add the browser on a **different** computers.

In case 1), when your Java servlet and your browser runs on the very same computer the Webcam Capture API won't be able to get images from your webcam because it already owned by a different process. You will have to close webcam in the browser first, and then open in from Java (by webcam.open()).

In case 2), when your servlet runs on a server and browser runs on the client, with both server and client being two different computers, I have to disappoint you since what you are trying to do will not be possible with Webcam Capture. This is because JavaScript is executed in a browser, that is on the client-side, and thus it's able to access hardware available on the computer which runs given browser (e.g. Chrome, Firefox).

The servlet, from the other hand, is running on the server-side and is only able to access webcam hardware connected to the server (not the client, which is physically a different computer).

In this case (2) the only way to capture user's webcam with Webcam capture is to use either Java Applet or Java Web Start. I strongly advice against using Applet because this technology is practically dead and very hard to use. The Web Start from the other had is an option, but you have to note that it requires classic Java application to be started on the user-side, which may not be acceptable by your design and users themselves.

In regards to your use-case. What I would recommend is not to use Webcam Capture project for this task because it does not have functionalities you require. I think the best for you will be to capture image in JavaScript and send it with AJAX request to your servlet which will then save it into the file.

Consider using this library by Adam Wróbel:

https://github.com/amw/jpeg_camera

Or this one by Joseph Huckaby:

https://github.com/jhuckaby/webcamjs

reginavaleria96, 2017-12-06:
Hi @sarxos , thanks for responding to my inquiry.
Just to clarify, my case is (1) where I run the servlet and the browser on same computer (on localhost).

Based on your answer, I came up with the solution by closing the current stream of webcam / stopping the webcam access from JavaScript on the browser, then I can call webcam.open() and capture an image. It does cause a slight delay, however for now I am still taking this approach as the solution to be able to do what I want.
Also this might also be an answer to issue #556

closed 2017-12-06 by reginavaleria96

sarxos, 2017-12-06:
Hi @reginavaleria96,

I'm very glad you worked this out :) Take care and good luck with your project!

_PS. I still think the better option would be to use JavaScript to capture image into blob and send it to the servlet. By doing this you could get rid of this unwanted delay you mentioned._