sarxos/webcam-capture #598: Problem with ubuntu 16 arm (orange pi)

hertzon, 2017-12-10:
Hi, i run https://github.com/rakeshbhatt10/WebCamJavaFXFXMLSample/blob/master/WebCamCaptureFXMLExample/src/fxml/controller/WebCamPreviewController.java

in windows runs perfectly...

but in linux embedded ubuntu 16 , java 1.8, orange pi, give error:

Exception in thread "JavaFX Application Thread" java.lang.NoClassDefFoundError: javafx/embed/swing/SwingFXUtils
        at cammeratest.FXMLDocumentController$3$1.run(FXMLDocumentController.java:136)
        at com.sun.javafx.application.PlatformImpl.lambda$null$174(PlatformImpl.java:295)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$175(PlatformImpl.java:294)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
        at com.sun.glass.ui.gtk.GtkApplication.lambda$null$50(GtkApplication.java:139)
        at java.lang.Thread.run(Thread.java:748)

please helpme to solve this

Best regards...

Nelson

sarxos, 2017-12-18:
This missing class should be in jfxrt.jar. Can you verify you have it in your classpath? If yes, can you check if this JAR contains this class?

cjgriscom, 2017-12-20:
ARM JavaFX does not contain SwingFXUtils.  You can make a local copy of it in your project.

hertzon, 2018-05-29:
I use vrl-0.4.3.2.3.jar and works fine, thanks