sarxos/webcam-capture #600: packages ch.qos.logback.classic.LoggerContext and ch.qos.logback.classic.joran.JoranConfigurator

aploteo, 2017-12-11:
Hi Bartosz,
Congratulations for your java api webcam.. I've been using it with great results, I have however a few minor issues with it. There are 2 classes missing:
packages ch.qos.logback.classic.LoggerContext and ch.qos.logback.classic.joran.JoranConfigurator 
in package com.github.sarxos.webcam.log in this java class file /Projecto/src/com/github/sarxos/webcam/log/WebcamLogConfigurator.java 
Do you have a solution for this? 
Also when running my app I get the following warnings:
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.

If you have a solution for these issues tell us about it. Thank you and keep up the good work.
luca
aploteo@gmail.com

sarxos, 2017-12-11:
Hi @aploteo, 

Yes, it's not [NP-Hard](https://en.wikipedia.org/wiki/NP-hardness) (sorry, I watch too much [Computerphile](https://www.youtube.com/watch?v=dJUEkjxylBw) lately) and of course there is a solution for this :smile:

There errors are due to missing [Logback](https://logback.qos.ch/) classes (Logback is a logging framework).

Download [logback-core](http://search.maven.org/remotecontent?filepath=ch/qos/logback/logback-core/1.0.9/logback-core-1.0.9.jar) and [logback-classic](http://search.maven.org/remotecontent?filepath=ch/qos/logback/logback-classic/1.0.9/logback-classic-1.0.9.jar) JARs and include them in your application's classpath.

After this is done create file named ```logback.xml``` and put it in the application's main directory. Here is the file content:

```xml
<configuration scan="true" scanPeriod="30 seconds">
	<appender name="STDOUT" class="ch.qos.logback.core.ConsoleAppender">
		<layout class="ch.qos.logback.classic.PatternLayout">
			<Pattern>%d{HH:mm:ss.SSS} [%thread] %-5level %logger{36} - %msg%n</Pattern>
		</layout>
	</appender>
	<root level="info">
		<appender-ref ref="STDOUT" />
	</root>
</configuration>
```

To load this configuration simply call:

```java
WebcamLogConfigurator.configure("logback.xml");
```

After this is done you should have nice logs printed into the console. If you want to see more logs you can change `level="info"` to `level="debug"`, but be aware - this may flood you console with tons of unnecessary data.

aploteo, 2017-12-12:
Hi Barthosz,

Thank you for the prompt reply. I was having a hard time looking for those
classes. I've included them in the classpath of my app and now everything
is running smooth.
Thanks again and take care
luca