sarxos/webcam-capture #601: [Question] access DroidCam

reginavaleria96, 2017-12-18:
Is it possible to access and take picture from a phone connected through [DroidCam](http://www.dev47apps.com/) using this project?

sarxos, 2017-12-18:
Hi @reginavaleria96,

It seems you can, but I don't know the exact URL to be used with ```IPCamDevice```. This is the claim from their web page:

> [image]

When you already know what is the URL to access MJPEG stream from the camera you can take a look at the IP camera examples of Webcam Capture to learn more on how to integrate it into your solution:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

Please let me know if you have any further questions.

sarxos, 2017-12-19:
@reginavaleria96, FYI, I found information on MJPEG here:

https://www.dev47apps.com/droidcam/connect/

Check this section: **Connect via Internet Browser or another Device (IP Camera)**.

neilyoung, 2017-12-19:
@sarxos Just out of curiosity: Would it be able to `getImageBytes` from such a cam?

sarxos, 2017-12-20:
@neilyoung, yes, but the performance will be much below what you can get with UVC (USB). This is due to:

1. MJPEG frame rate is much lower than UVC (usually ~2 FPS),
2. The `getImageBytes()` invokes `getImage()` and converts it to RGB bytes adding some few milliseconds on top of this.

In addition to 2, you can skip this and get RAW bytes, but these won't be RGB, but a JPEG frame, which you will have to convert to RGB to have compatibility with `getImageBytes()` (thus doing the same what `getImageBytes()` is doing underneath).

neilyoung, 2017-12-20:
@sarxos Clear and understandable. Thanks for the explanations. Merry Christmas to you and thanks too for all your efforts.

reginavaleria96, 2018-01-10:
I have included webcam-capture-0.3.12-20171103.095135-4.jar in my library but I'm getting **IpCamDeviceRegistry cannot be resolved** and **IpCamMode cannot be resolved to a variable**
Where can I find those?

sarxos, 2018-01-10:
@reginavaleria96, use these JARs: [libs.zip](https://github.com/sarxos/webcam-capture/files/1619648/libs.zip)

sarxos, 2018-01-10:
And remove webcam-capture-0.3.12-20171103.095135-4.jar beforehand. The zip I provided you with already contains newest webcam-capture dependency.

reginavaleria96, 2018-01-18:
Thanks for the JARs!
Do you mind explaining why is it necessary to have
`for (int i = 0; i < webcams.size(); i++) {`
(https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/LignanoBeachPushModeIpCameraExample.java line 57 onward)?

and do webcam.open() and webcam.close() work the same way for IPcam?

sarxos, 2018-01-18:
Hi @reginavaleria96,

In regards to your questions:

> why is it necessary to have

https://github.com/sarxos/webcam-capture/blob/13b0b61ed8a05e79bb845ac5a9865864182aa4f6/webcam-capture-drivers/driver-ipcam/src/examples/java/LignanoBeachPushModeIpCameraExample.java#L57-L62

This is the code which is executed when action is performed on a "Snapshot" button. When you click this button the ```for (int i = 0; i < webcams.size(); i++)``` will iterate over all ```Webcam``` instances in ```webcams``` list, get image and save it to file with ```ImageIO```. The file names will be ```test-0.jpg```, ```test-1.jpg```, ```test-2.jpg```, etc (as many as you have webcams). Without this code you would not have ability to save images into files.

And in regards to:

> do webcam.open() and webcam.close() work the same way for IPcam?

No. The ```open()``` and ```close()``` works in a different for IP cameras when you compare to _classic_ webcam.

In case of classic webcam (the [UVC device](https://en.wikipedia.org/wiki/USB_video_device_class)) when you invoke ```open()``` then hardware device is turned on (LED is powered) and USB capacity is allocated to stream image frames. After you invoke ```close()``` the USB capacity is released and UVC device is turned off. There is a direct connection between UVC being turned on/off and the webcam state (unless something is broken).

The [IP camera](https://en.wikipedia.org/wiki/IP_camera) from the other hand is always turned on, unless you turn it off with a manual switch, a power cord or administration panel. In this case when you invoke ```webcam.open()``` a persistent HTTP connection is made which is used to stream MJPEG frames from camera to your computer, but this is done with the IP/TCP instead of USB (as in case of UVC). Then, when you invoke ```webcam.close()```, the webcam is not turned off, but the HTTP connection is closed instead and IP camera device remains active.

This is a difference. I hope I was able to describe it in a understandable way.

sarxos, 2018-01-18:
@reginavaleria96 

Just for your information, if you would like to avoid managing JARs manually, I suggest you to take a look at [Maven](https://maven.apache.org/).

Maven is a tool to build Java projects and manage JAR dependencies. It may be a little hard to learn at the beginning, but there a dozens of tutorials on the internet. The ZIP bundles I sent you are also a list of files prepared by Maven.

The more JARs you have in the project the higher chance you will have to use some dependency manager (Maven, Gradle or Ivy). When Java project grows in time it comes into the point where managing JARs manually is impossible (due to massive amount of incompatible libraries).

My Webcam Capture API project is also managed and build with Maven.

reginavaleria96, 2018-01-18:
Thank you. I was confused about the webcam iteration, that was why I had to ask. I hope you don't mind.

If I have the default onboard webcam on my laptop and an IP camera connected via droidCam, how do I tell the program to specifically take one picture from my DroidCam camera?

And thank you for sharing about Maven, I have heard of it and had considered using it. My mistake was thinking that my small project won't have so many libraries and decided to manage them manually, but now I know that I was wrong :')

sarxos, 2018-01-18:
@reginavaleria96 

In regards to:

> I was confused about the webcam iteration

Yes, sorry for this. The code I initially wrote was to support multiple cameras, but finally I left only one IP camera in the code.

> If I have the default onboard webcam on my laptop and an IP camera connected via droidCam, how do I tell the program to specifically take one picture from my DroidCam camera?

You have to use composite driver to "compose" two drivers into one, example:

```java
/**
 * Composed webcam driver.
 */
public static class MyCompositeDriver extends WebcamCompositeDriver {

	public MyCompositeDriver() {
		add(new WebcamDefaultDriver());
		add(new IpCamDriver());
	}
}
```

You set it in the same way as you would do it with a different drivers:

```java
static {
	Webcam.setDriver(new MyCompositeDriver());
}
```

You can access cameras by index so the ones from first driver will be first one the list (e.g. index 0), and the ones from second driver will be added later (e.g. index 1, 2, 3, etc).

```java
Webcam webcam = Webcam.getWebcams().get(0);
Webcam ipcam = Webcam.getWebcams().get(1);
```

The other option is to use camera names, but this may be tricky on Windows since it may work well on your computer, but different computers may have different camera names. In Linux this is simple because on different computers every all cameras are simply ```/dev/video0```, ```/dev/video1```, etc, and on Windows this can be ```HD Webcam```, ```Microsoft Webcam```, etc. depending on the webcam hardware.

```java
import java.awt.Dimension;
import java.awt.FlowLayout;
import java.net.MalformedURLException;
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamCompositeDriver;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamPanel.DrawMode;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class DualNativeAndMjpegWebcamExample {

	public static class MyCompositeDriver extends WebcamCompositeDriver {

		public MyCompositeDriver() {
			add(new WebcamDefaultDriver());
			add(new IpCamDriver());
		}
	}

	static {
		Webcam.setDriver(new MyCompositeDriver());
	}

	public static void main(String[] args) throws MalformedURLException {

		IpCamDeviceRegistry.register("Lignano Beach IP Camera", "http://195.31.81.138/mjpg/video.mjpg", IpCamMode.PUSH);

		JFrame window = new JFrame("Live Views From Lignano Beach (Italy)");
		window.setLayout(new FlowLayout());
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);

		Webcam webcam1 = Webcam.getWebcams().get(0); // uvc
		Webcam webcam2 = Webcam.getWebcams().get(1); // ip camera

		webcam1.open();
		webcam2.open();

		WebcamPanel webcamPanel1 = new WebcamPanel(webcam1, false);
		webcamPanel1.setPreferredSize(new Dimension(WebcamResolution.QVGA.getSize()));
		webcamPanel1.setDrawMode(DrawMode.FIT);
		webcamPanel1.start();

		WebcamPanel webcamPanel2 = new WebcamPanel(webcam2, false);
		webcamPanel2.setPreferredSize(new Dimension(WebcamResolution.QVGA.getSize()));
		webcamPanel2.setDrawMode(DrawMode.FIT);
		webcamPanel2.start();

		window.add(webcamPanel1);
		window.add(webcamPanel2);

		window.pack();
		window.setVisible(true);
	}
}
```

[image]

reginavaleria96, 2018-01-18:
Thanks, that is one helpful explanation! I am going to try it now. I hope you don't mind if I still have more questions in the future.

sarxos, 2018-01-18:
@reginavaleria96 

If you link your carrier with Java then Maven is a tool you have to, at least, be familiar with. It has thousands of plugins and configuration options which may be overwhelming at the beginning because different people on the internet may suggest different solution, but after you understand how it works it will become one of the fundamental tools in your toolbox.

When you want to manage you project with Maven it's enough to put ```pom.xml``` file in the project main directory.

The simplest example for your project can be:

```xml
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">

	<modelVersion>4.0.0</modelVersion>

	<groupId>com.github.reginavaleria96</groupId>
	<artifactId>regina-test-project</artifactId>
	<version>0.1</version>

	<dependencies>
		<dependency>
			<groupId>com.github.sarxos</groupId>
			<artifactId>webcam-capture-driver-ipcam</artifactId>
			<version>0.3.12</version>
		</dependency>
	</dependencies>

</project>
```

Instead of providing many JARs I only had to add one dependency in the XML, that is:

```xml
<dependency>
	<groupId>com.github.sarxos</groupId>
	<artifactId>webcam-capture-driver-ipcam</artifactId>
	<version>0.3.12</version>
</dependency>
```

And when I ```cd``` into the project and run ```mvn dependency:tree``` I can see what JARs are used:

```plain
$ cd regina-test-project/
$ mvn dependency:tree
```

Which will display all JARs required by your project:

```plain
[INFO] com.github.reginavaleria96:regina-test-project:jar:0.1
[INFO] \- com.github.sarxos:webcam-capture-driver-ipcam:jar:0.3.12:compile
[INFO]    +- com.github.sarxos:webcam-capture:jar:0.3.12:compile
[INFO]    |  +- com.nativelibs4java:bridj:jar:0.7.0:compile
[INFO]    |  \- org.slf4j:slf4j-api:jar:1.7.2:compile
[INFO]    +- org.apache.httpcomponents:httpclient:jar:4.5.3:compile
[INFO]    |  +- org.apache.httpcomponents:httpcore:jar:4.4.6:compile
[INFO]    |  +- commons-logging:commons-logging:jar:1.2:compile
[INFO]    |  \- commons-codec:commons-codec:jar:1.9:compile
[INFO]    \- org.apache.httpcomponents:httpmime:jar:4.5.3:compile
```

And e.g. if I want to export all required JARs together with your own application JAR I can do the following:

```plain
$ mvn clean instal
$ mvn dependency:copy-dependencies -DoutputDirectory=target
```

And all JARs will be magically be placed in a ```target``` directory:

[image]

I spent many years working in a terminal environments and I'm used to do things from command line, but the same can be done from a decent IDEs, e.g. Eclipse or IntelliJ.

For example when you are using Eclipse IDE for Java EE Developers (basic Eclipse installation does not have Maven plugin installed) you can create Maven project from the mouse menu and manage dependencies with a very nice POM editor.

sarxos, 2018-01-18:
And one last thing - almost all (probably 99.99%) of all decent Java frameworks and libraries are deployed into Maven Central repository which is used as a source from where all dependencies are downloaded.

Jersey, JBoss, Hibernate, JSP, all Apache Commons, SL4J, Logback, Jetty, Netty, Vavr, Akka, and of course Webcam Capture API, all are available from Maven.

avatar31, 2018-01-29:
Hello Sir,
I am trying to use [Droidcam](http://www.dev47apps.com/) as a Webcam. The program is identifying the device but it not showing the panel. I referred [this](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/MjpegLignanoBeachExample.java) example. Please help me.

sarxos, 2018-01-29:
Are you able to view image in Firefox by providing the URL in address bar?
If yes, then you should be able to access your DroidCam as well, so there
is probably some issue in your code or some unknown bug in IP capture
driver. If you are unable to view image in Firefox, then you have to find
out what is the correct address first. What URL are you using when you
register IP camera device?

avatar31, 2018-01-29:
Yes, I am able to view image on browser.  I didn't make any change to the code. I just copied the example program. Here is the code.

import java.net.MalformedURLException;

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class Exa {
	static {
		Webcam.setDriver(new IpCamDriver());
	}
	public static void main(String[] args) throws MalformedURLException {
		System.out.println("M1: Started");
		IpCamDeviceRegistry.register("Droid", "http://192.168.1.100:4747", IpCamMode.PUSH);
		System.out.println("M2: Initialized");  //upto here the program runs
		WebcamPanel panel = new WebcamPanel(Webcam.getWebcams().get(0));
		System.out.println("M3: Panel created");
		panel.setFPSLimit(1);

		JFrame f = new JFrame("Live Views From Lignano Beach");
		f.add(panel);
		f.pack();
		f.setVisible(true);
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
	}
}

avatar31, 2018-01-29:
Yes, I am able to view the Image. I didn't make any changes to the code. I just copied the code from example program. Here is the code.

```import java.net.MalformedURLException;

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class Exa {

	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public static void main(String[] args) throws MalformedURLException {
		System.out.println("M1: Started");
		IpCamDeviceRegistry.register("Droid", "http://192.168.1.100:4747", IpCamMode.PUSH);
		System.out.println("M2: Initialized");  //upto here the program runs
		WebcamPanel panel = new WebcamPanel(Webcam.getWebcams().get(0));
		System.out.println("M3: Panel created");
		panel.setFPSLimit(1);
		
		JFrame f = new JFrame("Live Views From Lignano Beach");
		f.add(panel);
		f.pack();
		f.setVisible(true);
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
	}
}```

sarxos, 2018-01-30:
@avatar31,

If you can view it in browser and you can't in a Java code, means that your code is broken. The URL you provided to IP cam driver:

```
http://192.168.1.100:4747
```

Points to a web page, not MJPEG stream! A static HTML, nothing to stream video feed from. Please take a look at the DridCam web page and _read it carefully_! You will find this information:

> [image]

Therefore, your code should be fixed to use proper URL:

```
http://192.168.1.100:4747/mjpegfeed
```

I installed DroidCam and easily found it myself by checking what is the image source (just press F12 in Firefox):

[image]

This is my code:

```java
import java.net.MalformedURLException;

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class DroidCamExample {

	static {
		Webcam.setDriver(new IpCamDriver());
	}

	public static void main(String[] args) throws MalformedURLException {

		IpCamDeviceRegistry.register("DroidCam", "http://192.168.29.65:4747/mjpegfeed?640x480", IpCamMode.PUSH);

		WebcamPanel panel = new WebcamPanel(Webcam.getWebcams().get(0));
		panel.setFPSLimit(1);

		JFrame f = new JFrame("DroidCam Demo");
		f.add(panel);
		f.pack();
		f.setVisible(true);
		f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
	}
}
```

And it works perfectly well:

[image]

After reading this far please also take a look at the problem I described below.

**POSSIBLE PROBLEM WITH DROIDCAM!**

I noticed that you can connect **only one client** to the DroidCam. Therefore if you have your DroidCam web page open in a browser you **won't be able** to view it from code and vice versa - when you have your DroidCam streaming open from the Java code, you won't be able to view it in a browser. This is DroidCam limitation, not the Webcam Capture issue.

avatar31, 2018-01-31:
Thank You very much Sir. Now it is working