sarxos/webcam-capture #603: import org.slf4j.Logger; does not exist on windows machine ?

Krprashant94, 2017-12-31:
.\com\github\sarxos\webcam\Webcam.java:20: error: package org.slf4j does not exist
import org.slf4j.Logger;
                ^
.\com\github\sarxos\webcam\Webcam.java:21: error: package org.slf4j does not exist
import org.slf4j.LoggerFactory;
                ^
.\com\github\sarxos\webcam\WebcamExceptionHandler.java:5: error: package org.slf4j does not exist
import org.slf4j.Logger;
                ^
.\com\github\sarxos\webcam\WebcamExceptionHandler.java:6: error: package org.slf4j does not exist
import org.slf4j.LoggerFactory;
                ^
.\com\github\sarxos\webcam\WebcamExceptionHandler.java:7: error: package org.slf4j.helpers does not exist
import org.slf4j.helpers.NOPLoggerFactory;

stefan-reich, 2018-01-02:
How/what did you compile?

sarxos, 2018-01-03:
@Krprashant94,

It's nothing to do with a windows or non-windows machines. As @stefan-reich stated, you simply forgot to add required JARs into the classpath (in this case SLF4J JAR).

See this link for more details: 

https://github.com/sarxos/webcam-capture/wiki/java.lang.ClassNotFoundException%3A-org.slf4j.LoggerFactory

Krprashant94, 2018-01-03:
Yes, Thanks for that.

closed 2018-01-03 by Krprashant94