sarxos/webcam-capture #605: Webcam.getWebcams() stuck on macOS

jacktraror, 2018-01-10:
Hello

I am using the sarxos webcam to get pictures and it workes fine on Windows 10 but on Mac OS (high Sierra) it stucks at `Webcam.getWebcams()`, and the same happenes with `Webcam.getDefault()`, so nothing works.

In the logger, the only thing displayed is:
5 [JavaFX Application Thread] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used.

Is there any way to get around this and make this work on mac?

neilyoung, 2018-01-10:
it works perfectly on a Mac.

sarxos, 2018-01-19:
Hi @jacktraror,

You can debug and dig into the code to see what is happening. Enabling [debug logs]() may help you understand what is happening. Community indeed reported problems on _some_ Mac machines, but I'm unable to find a connection since it may not work on your Mac, but works perfectly well on someone's else computer.

vAlmaraz, 2018-03-06:
Hi @jacktraror 

It happens also for me in a MacBook Pro 2017 with Mac OSX 10.13.3.

I have fixed setting a timeout, for example:
```java
// 1 sec timeout
Webcam webcam = Webcam.getDefault(1000);
``` 

The default timeout is 9223372036854775807L.

rhysm94, 2018-03-21:
I’m having this exact same problem on macOS 10.13.3, on a MacBook Pro 13” 2015.
As @vAlmaraz has said, setting a timeout allows code to continue executing, but this leaves no way to access the iSight webcam on macOS.
If anyone has a solution, it would be greatly appreciated!

(P.S. comments like: “It works fine” are not useful to anybody)

migralito, 2018-05-12:
I had several cams plugged in and Webcam.getDefault() hangs. If I set a timeout on that calls, it throws timeout exception. I also subscribed to cams discovery listener but got no notifications about any of them. I booted with linux (have dual boot) and application works just fine: plugged in cameras are recognized though built in did not but thats because I didn't installed the driver. Will check later what happens if I install it (bcwc_pcie).
My point however is that I'm getting the same issue and logs in debug doesn't help at all. It's a 2017 two thunderbolt ports model with macOs 10.13.4

rider87, 2022-07-26:
did you fixed the issue?

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with recent MacOS versions https://github.com/eduramiba/webcam-capture-driver-native