sarxos/webcam-capture #607: IPCamDriver with Android IP WebCam app

neilyoung, 2018-01-12:
Works - kind of.
The frame rate is decent, pretty good. Usually I'm having about 25 "debug repaints" per second (is this kind of "fps"?)

But sometimes the image stutters for several ms. That time the debug display shows "200 debug repaints" or so, a big number. 

The original browser based app doesn't have that problem. Any chance to approach?

Regards

neilyoung, 2018-01-12:
No, much better. It is getting stuck every 5 secs. Some timer running?

neilyoung, 2018-01-12:
@sarxos Do you remember, when I asked you, if it is possible to access the buffered image as like with an ordinary cam? I think to remember, you have confirmed that. 

But it doesn't work:

```
Exception in thread "Thread-32" java.lang.IllegalStateException: Driver com.github.sarxos.webcam.ds.ipcam.IpCamDriver does not support buffer access
	at com.github.sarxos.webcam.Webcam.getImageBytes(Webcam.java:734)
	at com.myCompany.myApp.Capture.run(CameraAPI.java:33)

```

Pointer welcome

neilyoung, 2018-01-15:
Disregard. Found alternatives

sarxos, 2018-01-15:
Hi @neilyoung,

Yes I remember this discussion and indeed I confirmed this to be possible. Obviously I was wrong :( Sorry for misleading you.

Will fix this.

neilyoung, 2018-01-15:
@sarxos No prob. Found an older post here which suggested to patch IPCamDriver. Did that. Works

sarxos, 2018-01-15:
Cool :) Before I read your last comment I did this as well, but the change is not tested. You may take a look and decide which one performs better. I did not test this in any way so it can be laggy or not working. In this case I'm open for suggestions and/or pull requests.

neilyoung, 2018-01-15:
@sarxos Well, my "solution" is a bit different. I was not going the detour via the image again, but using the "frame" byte[] produced earlier. This incorporates more changes, though

sarxos, 2018-01-16:
@neilyoung, but isn't the frame a ```BufferedImage``` image instance? In the code I see:

https://github.com/sarxos/webcam-capture/blob/abaf107a02501ee9a1105f7cf238286b04bbc206/webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/IpCamDevice.java#L97

The ```image``` variable is returned when you invoke ```getImage()```. Therefore to have ```byte[]``` data you still have to _detour via the image_. 

Or are you maybe referring to ```frame``` variable from ```IpCamMJPEGStream```?

https://github.com/sarxos/webcam-capture/blob/abaf107a02501ee9a1105f7cf238286b04bbc206/webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/impl/IpCamMJPEGStream.java#L101

But in this case you have to note that this ```byte[]``` array is not an RGB24, but a JPEG frame.

neilyoung, 2018-01-16:
@sarxos I see. You are right. Will switch to your solution. Thanks

neilyoung, 2018-01-16:
@sarxos OK, works fine. When do you plan to publish it?

sarxos, 2018-01-16:
@neilyoung I can roll SNAPSHOT right now.

neilyoung, 2018-01-16:
@sarxos I see all the capture drivers there, but I don't see a webcam-capture snapshot.

sarxos, 2018-01-16:
@neilyoung Oh, there is (webcam-capture-0.3.12-20180116.162525-8.jar):

https://oss.sonatype.org/content/repositories/snapshots/com/github/sarxos/webcam-capture/0.3.12-SNAPSHOT/

sarxos, 2018-01-16:
@neilyoung Here are all required JARs: [target.zip](https://github.com/sarxos/webcam-capture/files/1636127/target.zip)

neilyoung, 2018-01-16:
Hmm. ok. seems to be not used by Intellij

neilyoung, 2018-01-16:
@sarxos Problem is, that a coworker is also depending on it and I would rather like to let him do the Maven stuff instead of asking him to replace two JARs... .Would be great if there could be an official 0.3.12 asap

sarxos, 2018-01-18:
Hi @neilyoung,

I spent whole yesterday's evening working on this, but finally, after several failed attempts, I was able to roll new official release, that is the [0.3.12](http://search.maven.org/#search%7Cga%7C1%7Cwebcam-capture). :tada: 

The artifacts I deployed yesterday are available already in [Maven Central](http://search.maven.org):

[image]

And the ZIP bundles are available from my [private repository](http://repo.sarxos.pl/maven2/index.php?dir=com%2Fgithub%2Fsarxos%2F):

[image]

[image]

neilyoung, 2018-01-18:
@sarxos Wow, you are my hero. Thanks for your patience with this project. Very much appreciated.

sarxos, 2018-01-19:
@neilyoung The pleasure is mine :) Feel free to open new issue in case of further problems or questions. I'm closing this one.

closed 2018-01-19 by sarxos