sarxos/webcam-capture #608: additional driver needed for usb webcam?

sothawo, 2018-01-14:
Hello,

I have a miscroscope atted to my Mac (macOS 10.13 High Sierra) and this shows up on USB with product id 0x2468 and manufacturer id 0x093a, a pretty common (and old) webcam.

This camera is not recognized as a webcam. Do I have to install special webcam drivers for this cam, or should webcam-capture work with webcams found on USB?

greetings
Peter

sarxos, 2018-01-14:
Hi @sothawo, are you on Windows, Linux or Mac OS?

sarxos, 2018-01-14:
Hi @sothawo, if system recognize it as webcam it should pick it up automatically. I'm not sure however, how this work on Mac since I'm mostly familiar with Linux or Windows. The webcam connection on Mac on the lowest level is done with QTCapture implemented in OpenIMAJ:

https://github.com/openimaj/openimaj/blob/master/hardware/core-video-capture/src-native/macosx/OpenIMAJGrabber/OpenIMAJGrabber.mm

sothawo, 2018-01-14:
Hello @sarxos, thanks for your prompt answer; this is on OSX. I think the problem is that although the device is visible on USB it is not recognized by the OS as a camera and therefore not returned. I found some old unmaintained application on the web (macam) which sees the camera, but the dirver that they provide is not working on actual OSX versions. I'll keep searching.

sothawo, 2019-05-06:
closing this as I am no longer working on this (was an experiment)

closed 2019-05-06 by sothawo