sarxos/webcam-capture #610: [Help Needed] New ffmpeg driver

yuripourre, 2018-01-18:
Hello @sarxos,

I start to write a [new FFMPEG driver](https://github.com/yuripourre/webcam-capture-ffmpegcliv2).
I can read the output from ffmpeg's process as rawvideo (yuv420p) and display the frames. But the refresh rate is very slow.

Maybe I should drop some frames.
Maybe there is a leak on how I am reading the process' output.

Can you please take a look? The repository has some examples (showing the progress).

sarxos, 2018-01-20:
@Hi @yuripourre,

You may try using bgr24 instead of yuv420p. It's bigger but much easier to parse since you have 3 bytes per pixel and you can easily convert raw data into ```BufferedImage``` without tricky loops and array copying you see in ```process()``` method.

yuripourre, 2018-01-20:
You are right, I will try bgr24.

The tricky loop can be avoided just changing a line.
`if (no > 0)` to `if (no > chunk.lenght)`
I missed that.

Thank you for the tips.

sarxos, 2018-01-20:
@yuripourre you may take a look at this:

https://github.com/sarxos/webcam-capture/blob/2e4091f7f055168be9038cd0db355f84162716b2/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L277-L297

This is how RGB24 data is converted to ```BufferedImage``` in default driver. This line indicates the indexes of RGB channels:

https://github.com/sarxos/webcam-capture/blob/2ccfaed4068a24e47a9f822eea3d60f3ff53a62d/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L88

Where 0 = R, 1 = G, 2 = B and therefore if you will be using BGR24 instead of RGB24, the it's enough to switch the indexes to:

```java
private static final int[] BAND_OFFSETS = new int[] { 2, 1, 0 };
```

yuripourre, 2018-01-20:
Thank you for pointing that.
I just made a [Pull Request](https://github.com/sarxos/webcam-capture/pull/612)
I was using:
```java
byte[] bgr = buffer.array();
image.setData(Raster.createRaster(image.getSampleModel(), new DataBufferByte(bgr, bgr.length), new Point()));
```

I start changing the structure to make it more similar to this driver but it might take a while to be finished.

sarxos, 2018-01-20:
Hi @yuripourre, thank you for your pull request. I found some issues when reviewing the code.

yuripourre, 2018-01-29:
Hello @sarxos I made several changes in the driver (following your comments). I think it is good now, I updated the PR with those changes. Can you please take a look later?

I successfully tested in Fedora 26 and Windows 7 64bits.

sarxos, 2018-01-30:
Hi @yuripourre,

I updated your pull request to rebase it on top of master branch, format source code and fix some minor issues. Updated code is already in the repo.

Please check this commit for more details: https://github.com/sarxos/webcam-capture/commit/c3fbe8e3dd14a80af983049ec7ed2ad2da080e95

Thank you for your effort!

yuripourre, 2018-01-30:
Thank you @sarxos. This is a great library, I wanted to contribute somehow and FFmpeg seemed a good option as it was just a "proof-of-concept" driver.

closed 2018-01-30 by yuripourre

sarxos, 2018-01-30:
Thank you @yuripourre :)