sarxos/webcam-capture #613: webcam-capture seems to cause a sudden exceptionless crash

TheRealMarnes, 2018-01-22:
I'm making a little project, a nanohttpd server that serves webcam snapshots as http resources. Functionally, it's all working: I can run an http request, which triggers my webcam (lights up) and responds with a nice jpg image. 👍 Thanks for making this and making it so simple.

However, after a few minutes or a number of requests (I can't tell what the trigger is - 2 to 5 minutes or ~30 calls), the java process crashes. I've narrowed the cause down to webcam-capture: webcam.open, webcam.close, or WebcamUtils.getImageBytes must be the cause because removing those calls prevents the crash. Either that or some background process in webcam-capture.

Strangely, there's no Exception or even Error thrown for me to catch. Could you please look at my service code and see if I've made any mistake in using the webcam-capture API that I don't notice myself?

My server has only a single (at most 2, sometimes) user, making a request every second or 2 or 5. The service is built so that there's no concurrent webcam access: only a single thread can open the webcam and capture an image at a time. Basically, there's no concurrency and no heavy load. The webcam is built-in and no other program uses it, so there are no hardware changes or locks.

I can give you the full code (it's a tiny program, just 6 classes and 2 libraries including webcam-capture) if you'd like.

Thanks for any help

```java
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamUtils;

import java.io.Closeable;
import java.io.IOException;
import java.io.UncheckedIOException;
import java.time.Duration;
import java.time.Instant;
import java.time.temporal.ChronoUnit;
import java.util.Arrays;
import java.util.Comparator;
import java.util.concurrent.atomic.AtomicReference;

public class PictureProvider {
	private static final Duration RENEW_INTERVAL = Duration.of(1, ChronoUnit.SECONDS);
	private static final String FORMAT = "jpg";

	private final Webcam webcam;
	private final AtomicReference<Boolean> renewing = new AtomicReference<>(false);

	private Picture lastPicture = null;

	public PictureProvider() {
		webcam = Webcam.getDefault();
		Arrays.stream(webcam.getViewSizes())
			.max(Comparator.comparing(d -> d.height * d.width))
			.ifPresent(webcam::setViewSize);
	}

	// this method is called by the http server on each request
	public Picture get(boolean renew) {
		if (lastPicture == null) {
			renew();
		}

		if (renew && Duration.between(lastPicture.getTimestamp(), Instant.now()).compareTo(RENEW_INTERVAL) > 0) {
			renew();
		}

		return lastPicture;
	}

	private void renew() {
		if (renewing.compareAndSet(false, true)) {
			// something here crashes after 2-5 minutes or ~30 calls
			try (Closeable c = webcam::close) {
				webcam.open();
				byte[] img = WebcamUtils.getImageBytes(webcam, FORMAT);
				Instant timestamp = Instant.now();
				lastPicture = new Picture(img, timestamp);
			} catch (IOException e) {
				throw new UncheckedIOException(e);
			} finally {
				renewing.set(false);
			}
		}
	}
}

```

The last things that are logged before the crash happens:

> 2018-01-22 22:09:26.416 INFO  [NanoHttpd Request Processor (#2)] net.leaumar.americam.HttpServer -- request from 192.168.0.1
> 2018-01-22 22:09:26.417 DEBUG [NanoHttpd Request Processor (#2)] com.github.sarxos.webcam.WebcamLock -- Lock Webcam Integrated Webcam 0
> 2018-01-22 22:09:26.422 INFO  [atomic-processor-1] com.github.sarxos.webcam.ds.cgt.WebcamOpenTask -- Opening webcam Integrated Webcam 0
> 2018-01-22 22:09:26.422 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Opening webcam device Integrated Webcam 0
> 2018-01-22 22:09:26.423 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Webcam device 0 starting session, size java.awt.Dimension[width=640,height=480]
> 2018-01-22 22:09:26.549 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Webcam device session started
> 2018-01-22 22:09:26.549 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Clear memory buffer
> 2018-01-22 22:09:26.873 DEBUG [webcam-discovery-service] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver -- Searching devices
> 2018-01-22 22:09:27.359 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@20c93cba is now open
> 2018-01-22 22:09:27.360 DEBUG [NanoHttpd Request Processor (#2)] com.github.sarxos.webcam.Webcam -- Webcam is now open Integrated Webcam 0
> 2018-01-22 22:09:27.362 DEBUG [webcam-discovery-service] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver -- Found device Integrated Webcam 0
> 2018-01-22 22:09:27.495 DEBUG [NanoHttpd Request Processor (#2)] com.github.sarxos.webcam.Webcam -- Closing webcam Integrated Webcam 0
> 2018-01-22 22:09:27.569 INFO  [atomic-processor-1] com.github.sarxos.webcam.ds.cgt.WebcamCloseTask -- Closing Integrated Webcam 0
> 2018-01-22 22:09:27.570 DEBUG [atomic-processor-1] com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice -- Closing webcam device

sarxos, 2018-01-22:
Hi, did you try to give it more memory with `-Xmx`? Also, what OS are you using?

sarxos, 2018-01-22:
In regards to:

> no Exception or even Error thrown for me to catch

This is because crash is happening in native code, not the Java.

TheRealMarnes, 2018-01-22:
Neither 512M nor 1024M changed anything. I don't think it's a memory size issue...

Windows 10

TheRealMarnes, 2018-01-22:
I tried 2048 too and it crashed after the same number of requests

sarxos, 2018-01-23:
Hi @TheRealMarnes,

Do you have any crashlog available? To solve this you may try using this capture driver instead of a default one:

https://github.com/frankpapenmeier/webcam-capture-driver-native

It was created specifically for Windows.

Default capture driver uses DirectShow to access webcam hardware where this native driver implemented by Frank Papenmeier use Microsoft Media Foundation which is new way to deal with webcams starting from Windows 8.

TheRealMarnes, 2018-01-23:
I've tried capturing the crash, but there isn't any log of it anywhere.

I'll try the native driver when I get to it, thanks

TheRealMarnes, 2018-01-23:
The native driver seems to resist crashing, but it fails in another way:

No, my room is not pink and green...

[image]

TheRealMarnes, 2018-01-23:
Aha, the color issue seems to be specific to jpg encoding - using png instead works!

Thanks for the workaround, @sarxos 👍

sarxos, 2018-01-23:
Hi @TheRealMarnes, this pink tint is caused by alpha channel in the image. JPG has 3 channels, namely RGB, and PNG has 4, i.e. ARGB where first byte is to encode alpha. This channel can be removed by painting ARGB on RGB image or removing alpha by rewriting pixels (but first option is easier, though).

You can stay with PNG if you are fine with that, but please note that default Java compression for both JPEG and PNG makes PNG much bigger when compared to JPEG.

To get rid of alpha:

```java
BufferedImage image = webcam.getImage();
BufferedImage bi = new BufferedImage(image.getWidth(), image.getHeight(), BufferedImage.TYPE_INT_RGB); // i'm not sure of this type
Graphics2D g2 = bi.createGraphics();
g2.drawImage(image, 0, 0, null);
g2.dispose(); // dispose or you will end up with memory leak
ImageIO.writeImage(bi, "jpg", new File("hello.jpg"));
bi.flush();
```

I wrote this code from my memory and it can be buggy since my ATP-driven biochemical compiler does not function well at this time of night :(

There are probably other methods available in Google, but this one is what I used the most.

TheRealMarnes, 2018-01-23:
Yeah, I've been messing around with the encoding myself too and got basically the same code.

There really seems to be an issue somewhere in the library or driver though: this code results in the pink tint, so the FORMAT_JPG parameter seems to be useless.

```java
try (Closeable c = webcam::close) {
	webcam.open();

	byte[] bytes = WebcamUtils.getImageBytes(webcam, ImageUtils.FORMAT_JPG);

	lastPicture = new Picture(bytes, Instant.now());
} catch (IOException e) {
	throw new UncheckedIOException(e);
}
```

TheRealMarnes, 2018-01-24:
Is there a way to prevent the crash using the regular driver? I'm not liking the native driver's buffer issues much.

sarxos, 2018-01-24:
Hi @TheRealMarnes,

I'm not aware of any other than fixing it. Default driver is from OpenIMAJ:

https://github.com/openimaj/openimaj/tree/master/hardware/core-video-capture/src-native/windows

It's open source so anyone can contribute. Any support is good and wanted.

TheRealMarnes, 2018-01-25:
I tried using lti-civil instead of webcam-capture, but I'm getting dll loading issues that way. Then I noticed webcam-capture has an lti-civil driver, so I set that up according to https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-lti-civil

Webcam-capture's lti-civil library gives me this error when I use it, so I think the 32/64 bit detection is broken. Gonna see if I can find something in the code.

> 2018-01-25 20:56:25.866 ERROR [webcam-discovery-service] com.github.sarxos.webcam.ds.civil.LtiCivilLoader -- Exception when loading DLL library
> java.lang.RuntimeException: Library file cannot be loaded: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:162)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:74)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilDriver.initialize(LtiCivilDriver.java:41)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilDriver.getDevices(LtiCivilDriver.java:67)
> 	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
> 	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
> 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
> 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
> 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
> 	at java.lang.Thread.run(Thread.java:748)
> Caused by: java.lang.UnsatisfiedLinkError: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
> 	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
> 	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
> 	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
> 	at java.lang.Runtime.load0(Runtime.java:809)
> 	at java.lang.System.load(System.java:1086)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:160)
> 	... 9 common frames omitted
> Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: java.lang.RuntimeException: java.lang.RuntimeException: Library file cannot be loaded: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll
> 	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
> 	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:874)
> 	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:847)
> 	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:826)
> 	at net.leaumar.americam.PictureProviderSarxosLtiCivil.main(PictureProviderSarxosLtiCivil.java:9)
> Caused by: java.util.concurrent.ExecutionException: java.lang.RuntimeException: java.lang.RuntimeException: Library file cannot be loaded: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll
> 	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
> 	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
> 	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
> 	... 4 more
> Caused by: java.lang.RuntimeException: java.lang.RuntimeException: Library file cannot be loaded: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:77)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilDriver.initialize(LtiCivilDriver.java:41)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilDriver.getDevices(LtiCivilDriver.java:67)
> 	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
> 	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
> 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
> 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
> 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
> 	at java.lang.Thread.run(Thread.java:748)
> Caused by: java.lang.RuntimeException: Library file cannot be loaded: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:162)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:74)
> 	... 8 more
> Caused by: java.lang.UnsatisfiedLinkError: C:\Users\Marnes\AppData\Local\Temp\civil-1516910185843\civil.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
> 	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
> 	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
> 	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
> 	at java.lang.Runtime.load0(Runtime.java:809)
> 	at java.lang.System.load(System.java:1086)
> 	at com.github.sarxos.webcam.ds.civil.LtiCivilLoader.load(LtiCivilLoader.java:160)
> 	... 9 more

TheRealMarnes, 2018-01-25:
Huh, odd. The loader correctly detects my system, so the selected dll should be the right one... So my first thought, given the error 

> civil.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform

Is that both dlls are 32 bit...

[image]

TheRealMarnes, 2018-01-25:
Well, replacing the misplaced dll by the 64bit dll downloaded from https://sourceforge.net/p/lti-civil/patches/4/ gives me the same error as manually loading lti-civil:

> # A fatal error has been detected by the Java Runtime Environment:
> #
> #  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000006f61ee8d, pid=5980, tid=0x0000000000003a4c
> #
> # JRE version: Java(TM) SE Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
> # Java VM: Java HotSpot(TM) 64-Bit Server VM (25.151-b12 mixed mode windows-amd64 compressed oops)
> # Problematic frame:
> # V  [jvm.dll+0x14ee8d]

sarxos, 2018-01-31:
Hi @TheRealMarnes,

The LTI-CIVIL is dead project for many years now. I did the best to have some stable DLLs in the JARs but it seems I missed something.

sarxos, 2018-12-05:
@TheRealMarnes Can I close this issue?

TheRealMarnes, 2018-12-05:
I guess. If the problem is in the drivers it's not like you can come up with a solution...

closed 2018-12-05 by TheRealMarnes

TheRealMarnes, 2018-12-13:
I ended up coincidentally fixing this crash with the openimaj driver by completely changing the way my project works. I changed it to be more multithreaded and keep the camera open for longer periods of time for multiple pictures, instead of opening it, taking 1 picture, and closing it again. Obviously it has better performance this way, but it also just stopped crashing and stopped repeating the same image. Just fyi :)

sarxos, 2018-12-14:
@TheRealMarnes Thank you for these details. Clues suggest that something may be wrong with native objects allocation or deallocation.

hmmueller, 2021-08-19:
> I ended up coincidentally fixing this crash with the openimaj driver by completely changing the way my project works. I changed it to be more multithreaded and keep the camera open for longer [...]

Thanks very much for this - it helped me to solve the same issues!