sarxos/webcam-capture #615: Download link for binary is broken

joepol, 2018-01-25:


yuripourre, 2018-01-26:
You are right, the right link should be [https://github.com/sarxos/webcam-capture/archive/webcam-capture-parent-0.3.12.zip](https://github.com/sarxos/webcam-capture/archive/webcam-capture-parent-0.3.12.zip)

I will send a pull request to fix it.

closed 2018-01-26 by joepol

sarxos, 2018-01-26:
Thank you @joepol for pointing this out and @yuripourre for the quick fix :)

sarxos, 2018-01-26:
@joepol, @yuripourre 

Guys, I just checked the ZIP and the one from Github contains only source code. No dependencies, no docs, etc.

Here is link complete 0.3.12 dist:

http://repo.sarxos.pl/maven2/com/github/sarxos/webcam-capture/0.3.12/webcam-capture-0.3.12-dist.zip

I need to replace the one in Github with this one.

sarxos, 2018-01-26:
I updated releases page and uploaded missing artifact:

https://github.com/sarxos/webcam-capture/releases/tag/webcam-capture-parent-0.3.12

Also updated README.md to reflect the change.

PR #616 (2018-01-26): Fix webcam-capture-0.3.12-dist.zip link
Closes #615
commit b380a2b0 (2018-01-26): Fix webcam-capture-0.3.12-dist.zip link

Closes #615