sarxos/webcam-capture #617: Java Crash in Spring Boot

ravikrishnasharma, 2018-01-31:
Hello,

I am facing this java crash intermittently. Please suggest a solution.
Has added the crash log.

I am using maven 0.3.12 version. Also I am appending the Java code used to capture the Image.

Any workaround to this also fine, because of this issue whole software is being stopped.

@Controller
public class WebCamController {

	public static byte[] imageByte;
	public static final Object lock = new Object();

	@RequestMapping(value = "/CaptureImage", method = RequestMethod.GET)
	public void captureImage(HttpServletResponse response, ModelMap model, HttpServletRequest request) {

		synchronized (WebCamController.lock) {
			imageByte = null;
			try {
				Webcam webcam = Webcam.getDefault();
				if (webcam.isOpen()) {
					webcam.close();
				}

				int i = 0;
				do {
					if (webcam.getLock().isLocked()) {
						System.out.println("Waiting for lock to be released " + i);
						try {
							Thread.sleep(2000);
						} catch (InterruptedException e1) {
							e1.printStackTrace();
							response.setContentType("image/png");
							response.setContentLength(0);
							response.getOutputStream().write(imageByte);
							return;
						}
					} else {
						break;
					}
				} while (i++ < 2);

				webcam.setViewSize(new Dimension(640, 480));
				webcam.open();

				// get image
				BufferedImage image = webcam.getImage();

				// save image to PNG file
				// ImageIO.write(image, "PNG", new File("test.png"));

				ByteArrayOutputStream baos = new ByteArrayOutputStream();

				webcam.close();
				ImageIO.write(image, "png", baos);
				baos.flush();
				imageByte = baos.toByteArray();
				baos.close();

				response.setContentType("image/png");
				response.setContentLength(imageByte.length);
				response.getOutputStream().write(imageByte);
				return;
			} catch (IOException e) {
				// TODO Auto-generated catch block
				e.printStackTrace();
			} catch (Exception e) {
				// TODO Auto-generated catch block
				e.printStackTrace();
			}
		}
	}
}

[hs_err_pid2768.log](https://github.com/sarxos/webcam-capture/files/1680717/hs_err_pid2768.log)

sarxos, 2018-01-31:
Hi @ravikrishnasharma,

The call stack from your `hs_err_pid` file show it's caused by BridJ:

```plain
Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  org.bridj.JNI.free(J)V+0
j  org.bridj.Pointer$FreeReleaser.release(Lorg/bridj/Pointer;)V+192
j  org.bridj.Pointer$5.release(Lorg/bridj/Pointer;)V+14
j  org.bridj.Pointer$3.release()V+19
j  org.bridj.Pointer$3.finalize()V+1
v  ~StubRoutines::call_stub
J  java.lang.ref.Finalizer.invokeFinalizeMethod(Ljava/lang/Object;)V
J  java.lang.ref.Finalizer.access$100(Ljava/lang/ref/Finalizer;)V
J  java.lang.ref.Finalizer$FinalizerThread.run()V
v  ~StubRoutines::call_stub
```

Does the same happen when you run simple example normally, not with Spring Boot? For example this code:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamPanelExample.java

I'm wondering if this is caused by weird Tomcat's classloading (I had problems with that in the past).

I also I suggest to check [webcam-capture-driver-native](https://github.com/frankpapenmeier/webcam-capture-driver-native) by Frank Papenmeier. It does not use BridJ at all but calls JNI directly and therefore can be used to w/a this BridJ problem.

sarxos, 2018-01-31:
And before you try above, can you make a simple modification in your code and verify it again? I would like you to move this:

```java
Webcam webcam = Webcam.getDefault();
```

To be a static field in your class, such that:

```java
@Controller
public class WebCamController {

	public static final Webcam webcam = Webcam.getDefault();
	public static byte[] imageByte;
	public static final Object lock = new Object();

	@RequestMapping(value = "/CaptureImage", method = RequestMethod.GET)
	public void captureImage(HttpServletResponse response, ModelMap model, HttpServletRequest request) {

		synchronized (WebCamController.lock) {
			imageByte = null;
			try {
				if (webcam.isOpen()) {
					webcam.close();
				}
```

Also, souldn't it be better if you have your webcam constantly open? In this case you can serve images much faster.

You could do:

```java
@PostConstruct
public void init() {
  webcam.open();
}
```

And get rid of this:

```java
if (webcam.isOpen()) {
  webcam.close();
}
```

ravikrishnasharma, 2018-01-31:
Hi sarxos,
Thanks for quick update. 
Definitely I will try and update the results soon.

Also I have a questions.

1. My init method looks like this.

@PostConstruct
	public void init() {
		if (webcam != null) {
			webcam.setViewSize(new Dimension(640, 480));
			webcam.open();
		}
	}

2. Do I need to check for webcam lock before every capture and wait for it to release?? Basically the following code
int i = 0;
			do {
				if (webcam.getLock().isLocked()) {
					System.out.println("Waiting for lock to be released " + i);
					try {
						Thread.sleep(2000);
					} catch (InterruptedException e1) {
						e1.printStackTrace();
						response.setContentType("image/png");
						response.setContentLength(0);
						response.getOutputStream().write(imageByte);
						return;
					}
				} else {
					break;
				}
			} while (i++ < 2);

Thanks,
Ravikrishna

sarxos, 2018-01-31:
Hi @ravikrishnasharma,

In regards to:

> Do I need to check for webcam lock before every capture and wait for it to release?

No. You don't have to. In this case the code will simply wait for other processes to complete. If you do not want this to be blocking then you can open webcam in async mode:

```java
webcam.open(true);
```

Instead of:

```java
webcam.open();
```

In your `init()`.

ravikrishnasharma, 2018-02-13:
Hi sarxos,

Making webcam object as static solved the crash. So the issue is because of tomcat class loading.
Thanks for the support!

Also a quick question, Can I use this library to deal with wi-fi camera as well?

Cheers!
Ravikrishna

sarxos, 2018-02-13:
Hi @ravikrishnasharma,

Good to hear that :)

In regards to your question:

> Can I use this library to deal with wi-fi camera as well?

For me a "wi-fi camera" is just a fancy name for a regular IP camera. I see no difference between linking your cameras with ethernet cords and wi-fi connections. Both (wi-fi or poe) are using TCP/IP to communicate and thus are a regular IP cameras. As long as you can access you wi-fi/IP/whatever camera MJPEG stream by http:// you can use it with Webcam Capture.

To use Webcam Capture API with IP camera you have to use IP camera driver:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam

Or MPJEG driver:

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-mjpeg

ravikrishnasharma, 2018-03-19:
Hi @Sarxos,

When I use the MPJEG Driver I get the following exception

11:25:32.534 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new capture driver com.github.sarxos.webcam.ds.mjpeg.MjpegCaptureDriver@7dc280db
11:25:32.559 [main] INFO com.github.sarxos.webcam.WebcamDiscoveryService - Discovery will not run - driver MjpegCaptureDriver does not support this feature
Exception in thread "main" com.github.sarxos.webcam.WebcamException: Cannot get image frame from tcp://192.168.0.127:5050
	at com.github.sarxos.webcam.ds.mjpeg.MjpegCaptureDevice.getImage(MjpegCaptureDevice.java:78)
	at com.github.sarxos.webcam.ds.mjpeg.MjpegCaptureDevice.getResolutions(MjpegCaptureDevice.java:51)
	at com.github.sarxos.webcam.Webcam.getViewSizes(Webcam.java:540)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:774)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:752)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:737)
	at com.github.sarxos.webcam.WebcamPanel.<init>(WebcamPanel.java:727)
	at com.loctell.crusher.util.IPWebCamCaptureUtil.main(IPWebCamCaptureUtil.java:26)
Caused by: java.io.EOFException
	at java.io.DataInputStream.readUnsignedByte(DataInputStream.java:290)
	at com.github.sarxos.webcam.util.MjpegInputStream.getEndOfSeqeunce(MjpegInputStream.java:65)
	at com.github.sarxos.webcam.util.MjpegInputStream.getStartOfSequence(MjpegInputStream.java:79)
	at com.github.sarxos.webcam.util.MjpegInputStream.readFrame(MjpegInputStream.java:118)
	at com.github.sarxos.webcam.ds.mjpeg.MjpegCaptureDevice.getImage(MjpegCaptureDevice.java:76)
	... 7 more