sarxos/webcam-capture #618: Q: Again a stupid question...

neilyoung, 2018-01-31:
... is there a way feed a webcam instance with RGB data? Like a "software cam"?

sarxos, 2018-02-01:
Hi @neilyoung,

There is no such driver available in the release, but I have something similar available in tests. It's enough to copy the source code and use it in your project.

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/test/java/com/github/sarxos/webcam/ds/test/DummyDriver.java

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/test/java/com/github/sarxos/webcam/ds/test/DummyDevice.java

This `DummyDevice` renders red background with ball bouncing from the edges.

```java
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.test.DummyDriver;

public class DummyDriverExample {

	static {
		Webcam.setDriver(new DummyDriver());
	}

	public static void main(String[] args) {
		JFrame window = new JFrame("Dummy Test");
		window.add(new WebcamPanel(Webcam.getDefault()));
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}

```

[image]

If you want to feed your RGB data to the application and render as image from webcam then `DummyDriver` and `DummyDevice` are a good starting point. Just change `getImage()` method in `DummyDevice` to draw your RGB data on the returned image.

If you want to paint your RGB data on the _real_ image from webcam, then `WebcamImageTransformer` interface can do that for you:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/ImageTransformerExample.java

In this example transformer does nothing more than changing image to grayscale, but in the same way you can paint any RGB data on the image.

neilyoung, 2018-02-01:
@sarxos Thanks, will give it a try

neilyoung, 2018-02-03:
OK, have used this as a template. It works excellent. By using it I was able to write my own MJPEG decoder (tailored to the needs). I found, that the frame rate I can achieve (with 30 fps) is more or less exactly 30 fps on the sink, whereas your MjpegDriver is reaching 18 to 20 fps. Not sure, what it is, but if you like I can share my sources on a private channel.

Because my consuming process has a little problem being fed with one and the same frame several times (the using app is able to achieve 90 fps) I'm now refactoring the code so that "getImage" is hanging on a semaphore and just released on a new "real" frame. 

I will conserve the current state so that I can send it to you if you like. The MIME decoder is pretty much nailed to the project, but it works great.

neilyoung, 2018-02-03:
@sarxos This log will not tell you much, but this is what I see (and currently don't understand):

fps : Measured at driver level, whenever a new frame is decoded and made available
FPS: Measured at driver level, whenever a new image is retrieved via getImage()
APP FPS: Measured at app level as shown above.

```
fps: 29.92
FPS: 29.92
APP FPS: 14.90
fps: 29.97
FPS: 29.97
APP FPS: 14.82
fps: 29.64
FPS: 29.62
APP FPS: 14.87
fps: 30.97
FPS: 30.10
APP FPS: 14.86
fps: 30.42
FPS: 30.30
APP FPS: 15.31
fps: 29.01
FPS: 29.93
APP FPS: 15.00
fps: 30.75
FPS: 30.78
APP FPS: 14.86
fps: 29.25
FPS: 29.27
APP FPS: 14.84
fps: 21.09
FPS: 20.21
APP FPS: 15.46
fps: 39.39
FPS: 39.42
APP FPS: 14.48
fps: 31.31
FPS: 31.31
APP FPS: 15.01
fps: 29.62
FPS: 29.62
APP FPS: 15.72
fps: 29.79
FPS: 29.79
APP FPS: 14.46
fps: 29.24
FPS: 29.24

```

neilyoung, 2018-02-03:
OK, giving up for today. Don't get it really. Here is a raw trace, which I don't understand:

Explanation: The `returnFrame notify` is issued, when a new frame has been decoded and is available as image. The same time the driver triggers a sync object

```
synchronized (syncLock) {
            deliveredFramecount++;
            Util.logger.trace("returnFrame notify");
            syncLock.notify();
}

```

The `getImage leaveWait` trace is issued in the driver's `getImage()` function. The function waits for the release of the object like so:

```
            Util.logger.trace("getImage enter wait");
            synchronized (syncLock) {
                syncLock.wait();
            }
            Util.logger.trace("getImage leave wait");

```

So as far as I can see each `returnFrame notify` is immediately followed by a `getImage leave wait` which returns the image and immediately calls `getImage enter wait` wait again, which means to me, that the `getImage()` is called in a loop. OK for me.

However, only every 2nd delivery ends up in a `Got frame` trace, which is issued by an application thread, which calls `webcam.getImage()` in a loop. In fact the trace totally confirms the weird observation, that I can't achieve full 30 fps at app level, just the half.

```
2018-02-03 23:46:11,812 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,812 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,812 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,824 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,824 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,825 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,825 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:11,873 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,873 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,873 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,886 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,887 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,887 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,887 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:11,939 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,939 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,939 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,953 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:11,953 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:11,953 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:11,953 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:12,005 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,005 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,006 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,017 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,017 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,017 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,017 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:12,072 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,072 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,072 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,086 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,086 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,086 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,086 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:12,147 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,147 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,147 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,157 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,158 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,158 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
[... 47 lines cut ...]
2018-02-03 23:46:12,602 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,602 Capture 63 TRACE [Thread-30]: Got frame
fps: 29.97
2018-02-03 23:46:12,665 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,666 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
FPS: 29.97
2018-02-03 23:46:12,666 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,672 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
[... cut ...]
2018-02-03 23:46:12,736 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
2018-02-03 23:46:12,737 Capture 63 TRACE [Thread-30]: Got frame
2018-02-03 23:46:12,760 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify
2018-02-03 23:46:12,760 MJPEGDirectDevice 202 TRACE [atomic-processor-1]: getImage leave wait
2018-02-03 23:46:12,762 MJPEGDirectDevice 196 TRACE [atomic-processor-1]: getImage enter wait
APP FPS: 13.90
2018-02-03 23:46:12,826 MJPEGDirectDevice 68 TRACE [Thread-27]: returnFrame notify

```

neilyoung, 2018-02-04:
Found our old conversation: https://github.com/sarxos/webcam-capture/issues/599. Still not convinced that it has to be that complicated. Thinking about  a direct frame callback from driver to application...

neilyoung, 2018-02-04:
@sarxos Finally found a better way: Dropped WebcamPanel, doing my own drawing in the capture thread. Can be closed.

closed 2018-02-04 by neilyoung