sarxos/webcam-capture #620: "Cannot change resolution when webcam is open, please close it first"

Zeeebas, 2018-02-05:
Hello,

I made a simple program to take a photo. The photo will be taken when I press this button.
After the photo is taken, I open my Scherm.java and it shows the image that is taken. So in that frame I made an JToggleButton, that opens the first frame (Camera.java). When I press this button, I go back to my frame and press the button to take the photo again. I get this error: 

"Exception in thread "AWT-EventQueue-0" java.lang.IllegalStateException: Cannot change resolution when webcam is open, please close it first at com.github.sarxos.webcam.Webcam.setViewSize(Webcam.java:544)"

I have been searching and I found out that if I remove the webcam.setViewSize everything works. But then my photo is too small so I can't use it. Does someone know how I can fix this problem?

This code is my Camera.java 
`class foto implements ActionListener {
        public void actionPerformed(ActionEvent a) {

            System.out.println("Foto maken... ");
        	 Webcam webcam = Webcam.getDefault();

            if (webcam != null) {
                System.out.println("Webcam: " + webcam.getName());
               
                webcam.setViewSize(new Dimension(640,480)); //If I remove this line, it works fine.
                webcam.open();
                Scherm scherm = new Scherm();
                frame.dispose();

                try {
                    ImageIO.write(webcam.getImage(), "PNG", new File("Test.png"));
                } catch (IOException e) {
                    // TODO Auto-generated catch block
                    e.printStackTrace();                    
                }
            } else {
                System.out.println("No webcam detected");
`

sarxos, 2018-02-05:
Hi @Zeeebas,

You have several options:

1. Close webcam after you took photo.

```java
try { 
  webcam.open();
  // do your stuff here, take photo etc
} finally {
  webcam.close();
}
```

2. Open webcam only once, on application start and set view size only once.
3. Check if webcam is not open before you `setViewSize(..)` (this one is the best IMO).

```java
if (!webcam.isOpen()) {
  webcam.setViewSize(WebcamResolution.VGA.getSize());
}
```

sarxos, 2018-02-12:
Hi @Zeeebas, can I close this ticket? I hope I was able to answer your question in my last response.

Zeeebas, 2018-02-12:
Hey @sarxos
I am so sorry I didnt respond.  My death line was the next day and your solution fixed the problem, I was so happy and glad that I worked I forgot to thank you :)
Thank you so much, yes you can.

closed 2018-02-12 by Zeeebas

sarxos, 2018-02-12:
Cool. Take care!