sarxos/webcam-capture #621: Exception in thread "main" java.lang.NoSuchFieldError: INSTANCE

inferno2246, 2018-02-10:
I am trying to get the video feed from IP Camera App in a JFrame, but i get this error. Please help.

Exception in thread "main" java.lang.NoSuchFieldError: INSTANCE
	at org.apache.http.impl.io.DefaultHttpRequestWriterFactory.<init>(DefaultHttpRequestWriterFactory.java:53)
	at org.apache.http.impl.io.DefaultHttpRequestWriterFactory.<init>(DefaultHttpRequestWriterFactory.java:57)
	at org.apache.http.impl.io.DefaultHttpRequestWriterFactory.<clinit>(DefaultHttpRequestWriterFactory.java:47)
	at org.apache.http.impl.conn.ManagedHttpClientConnectionFactory.<init>(ManagedHttpClientConnectionFactory.java:83)
	at org.apache.http.impl.conn.ManagedHttpClientConnectionFactory.<init>(ManagedHttpClientConnectionFactory.java:96)
	at org.apache.http.impl.conn.ManagedHttpClientConnectionFactory.<init>(ManagedHttpClientConnectionFactory.java:105)
	at org.apache.http.impl.conn.ManagedHttpClientConnectionFactory.<clinit>(ManagedHttpClientConnectionFactory.java:63)
	at org.apache.http.impl.conn.PoolingHttpClientConnectionManager$InternalConnectionFactory.<init>(PoolingHttpClientConnectionManager.java:586)
	at org.apache.http.impl.conn.PoolingHttpClientConnectionManager.<init>(PoolingHttpClientConnectionManager.java:180)
	at org.apache.http.impl.conn.PoolingHttpClientConnectionManager.<init>(PoolingHttpClientConnectionManager.java:164)
	at org.apache.http.impl.client.HttpClientBuilder.build(HttpClientBuilder.java:971)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.createClient(IpCamDevice.java:246)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.<init>(IpCamDevice.java:217)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDevice.<init>(IpCamDevice.java:196)
	at com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry.register(IpCamDeviceRegistry.java:53)
	at IpCam.main(IpCam.java:19)

Here is my code:

import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.ipcam.IpCamDeviceRegistry;
import com.github.sarxos.webcam.ds.ipcam.IpCamDriver;
import com.github.sarxos.webcam.ds.ipcam.IpCamMode;

public class IpCam {

    static {
        Webcam.setDriver(new IpCamDriver());
    }

    public static void main(String[] args) {
        try {
            IpCamDeviceRegistry.register("IP Webcam", "http://192.168.43.1:8080/video", IpCamMode.PUSH);
        }catch(Exception e){
            e.printStackTrace();
        }

        WebcamPanel panel = new WebcamPanel(Webcam.getWebcams().get(0));
        panel.setFPSLimit(1);

        JFrame f = new JFrame("DroidCam Demo");
        f.add(panel);
        f.pack();
        f.setVisible(true);
        f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
    }
}

I've imported many JAR files, in order to solve many errors i have been getting, but now i don't understand how to solve this error. Here are the JAR files i have imported:
1)commons-logging-1.1.1.jar
2)httpclient-4.5.3.jar
3)httpcore.jar
4)bridj-0.6.2.jar
5)httpcore-4.4.8.jar
6)slf4j-api-1.7.2.jar
7)slf4j-binding-1.3.2.jar
8)tinylog-1.3.2.jar
9)webcam-capture-0.3.10.jar
10)webcam-capture-driver-ipcam-0.3.13-20180202.082032-3.jar

Thanks!

sarxos, 2018-02-10:
Hi @inferno2246,

I just guess it's caused by this duplicated (incompatible) dependency you have:

3)httpcore.jar
5)httpcore-4.4.8.jar

Remove the first one and check again.

There is also have problem dependency for webcam-capture.

9)webcam-capture-0.3.10.jar
10)webcam-capture-driver-ipcam-0.3.13-20180202.082032-3.jar

Both should be the same version but one is 0.3.10 and second one is 0.3.13-SNAPSHOT. You should not use different versions of webcam-capture jars. Both latest release JARs are available in Maven Central repository, and if you need snapshot version (0.3.13-20180202.082032-3), you can find both JARs in Sonatype repository. Please let me know in case you can't find them.

sarxos, 2018-02-10:
You can avoid necessity to deal with all these JARs by using Maven which imports all required JARs automatically.

inferno2246, 2018-02-11:
Thank you so much for replying @sarxos, i solved the issue by using JavaCV but still i learnt a lot through this library. Again thanks for helping 👍

sarxos, 2018-02-11:
Cool! Take care :)

closed 2018-02-11 by sarxos