sarxos/webcam-capture #623: How to take a photo usb camera in my android app ?

amittech1, 2018-02-22:
How can take a photo in my android app

amittech1, 2018-02-22:
02/22 14:10:54: Launching app
$ adb push D:\MyApplication\app\build\outputs\apk\debug\app-debug.apk /data/local/tmp/com.cipl.health_tablet.myapplication
$ adb shell pm install -t -r "/data/local/tmp/com.cipl.health_tablet.myapplication"
	pkg: /data/local/tmp/com.cipl.health_tablet.myapplication
Success

$ adb shell am start -n "com.cipl.health_tablet.myapplication/com.cipl.health_tablet.myapplication.MainActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 27536 on device iball-3g_1026_q18-0123456789ABCDEF
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
V/Provider/Settings: invalidate [system]: current 2 != cached 0
D/ActivityThread: hoder:android.app.IActivityManager$ContentProviderHolder@42236208,provider,holder.Provider:android.content.ContentProviderProxy@42236980
V/Provider/Settings: from db cache, name = anr_debugging_mechanism , value = 1
D/asset: AssetManager-->addDefaultAssets CIP path not exsit!
D/Proxy: setHttpRequestCheckHandler
D/dalvikvm: open_cached_dex_file : /data/app/com.cipl.health_tablet.myapplication-1.apk /data/dalvik-cache/data@app@com.cipl.health_tablet.myapplication-1.apk@classes.dex
I/MultiDex: VM with version 1.6.0 does not have multidex support
I/MultiDex: Installing application
I/MultiDex: MultiDexExtractor.load(/data/app/com.cipl.health_tablet.myapplication-1.apk, false, )
I/MultiDex: Blocking on lock /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/MultiDex.lock
I/MultiDex: /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/MultiDex.lock locked
D/dalvikvm: create interp thread : stack size=128KB
D/dalvikvm: create new thread
D/dalvikvm: new thread created
D/dalvikvm: update thread list
D/dalvikvm: threadid=11: interp stack at 0x600dc000
D/dalvikvm: threadid=11: created from interp
D/dalvikvm: start new thread
D/dalvikvm: threadid=11: notify debugger
D/dalvikvm: threadid=11 (SharedPreferencesImpl-load): calling run()
D/dalvikvm: threadid=11: exiting
D/dalvikvm: threadid=11: bye!
I/MultiDex: Detected that extraction must be performed.
I/MultiDex: Extraction is needed for file /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.zip
I/MultiDex: Extracting /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/tmp-com.cipl.health_tablet.myapplication-1.apk.classes60394523.zip
I/AndroidRuntime: zoneinfo is Asia/Calcutta
I/AndroidRuntime: persist.sys.timezone is Asia/Calcutta
I/MultiDex: Renaming to /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.zip
I/MultiDex: Extraction succeeded - length /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.zip: 1387210 - crc: 1311614884
D/dalvikvm: create interp thread : stack size=128KB
D/dalvikvm: create new thread
D/dalvikvm: new thread created
D/dalvikvm: update thread list
D/dalvikvm: threadid=11: interp stack at 0x600dc000
D/dalvikvm: threadid=11: created from interp
D/dalvikvm: start new thread
D/dalvikvm: threadid=11: notify debugger
D/dalvikvm: threadid=11 (SharedPreferencesImpl-load): calling run()
D/dalvikvm: threadid=11: exiting
D/dalvikvm: threadid=11: bye!
I/MultiDex: load found 1 secondary dex files
D/dalvikvm: open_cached_dex_file : /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.zip /data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.dex
D/dalvikvm: DexOpt: --- BEGIN 'com.cipl.health_tablet.myapplication-1.apk.classes2.zip' (bootstrap=0) ---
D/dalvikvm: threadid=11: interp stack at 0x600e5000
D/dalvikvm: DexOpt: --- END 'com.cipl.health_tablet.myapplication-1.apk.classes2.zip' (success) ---
D/dalvikvm: DEX prep '/data/data/com.cipl.health_tablet.myapplication/code_cache/secondary-dexes/com.cipl.health_tablet.myapplication-1.apk.classes2.zip': unzip in 77ms, rewrite 1079ms
I/MultiDex: install done
D/ActivityThread: BIND_APPLICATION handled : 0 / AppBindData{appInfo=ApplicationInfo{422336f0 com.cipl.health_tablet.myapplication}}
I/dalvikvm: Could not find method android.view.Window$Callback.onPointerCaptureChanged, referenced from method android.support.v7.view.WindowCallbackWrapper.onPointerCaptureChanged
W/dalvikvm: VFY: unable to resolve interface method 11708: Landroid/view/Window$Callback;.onPointerCaptureChanged (Z)V
D/dalvikvm: VFY: replacing opcode 0x72 at 0x0002
I/dalvikvm: Could not find method android.view.Window$Callback.onProvideKeyboardShortcuts, referenced from method android.support.v7.view.WindowCallbackWrapper.onProvideKeyboardShortcuts
W/dalvikvm: VFY: unable to resolve interface method 11710: Landroid/view/Window$Callback;.onProvideKeyboardShortcuts (Ljava/util/List;Landroid/view/Menu;I)V
D/dalvikvm: VFY: replacing opcode 0x72 at 0x0002
W/dalvikvm: VFY: unable to find class referenced in signature (Landroid/view/SearchEvent;)
I/dalvikvm: Could not find method android.view.Window$Callback.onSearchRequested, referenced from method android.support.v7.view.WindowCallbackWrapper.onSearchRequested
[... cut ...]
                      at android.app.ActivityThread.main(ActivityThread.java:5292)
                      at java.lang.reflect.Method.invokeNative(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:515)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:824)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:640)
                      at dalvik.system.NativeStart.main(Native Method)
I/Process: Sending signal. PID: 27536 SIG: 9
Application terminated.

sarxos, 2018-02-22:
Hi @amittech1. This library does not support Android. Sorry.