sarxos/webcam-capture #624: Connot show any webcam output

cezzynahurt, 2018-03-04:
Hi i am a student, i have a problem , there's no output or show any panel of webcam in my netbeans...this is my codes 

package pkg123;
import com.github.sarxos.webcam.Webcam;
import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;

public class Main {

    public static void main(String[] args) throws IOException{
  Webcam webcam = Webcam.getDefault();
webcam.open();
ImageIO.write(webcam.getImage(), "PNG", new File("123.PNG"));
    }
    
}

And this is the output when i run the file 
11:23:09.475 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
11:23:09.487 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
11:23:12.366 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Lenovo EasyCamera 0
11:23:12.374 [main] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam Lenovo EasyCamera 0
11:23:12.386 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam Lenovo EasyCamera 0
11:23:12.386 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device Lenovo EasyCamera 0
11:23:12.386 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=176,height=144]
11:23:12.649 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
11:23:12.651 [atomic-processor-1] WARN  c.g.s.w.d.b.WebcamDefaultDevice - Different size obtained vs requested - [176x144] vs [160x120]. Setting correct one. New size is [160x120]
11:23:12.656 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Clear memory buffer
11:23:12.975 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@e1b93e5 is now open
11:23:12.976 [main] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open Lenovo EasyCamera 0
11:23:13.125 [shutdown-hook-1] INFO  c.g.sarxos.webcam.WebcamShutdownHook - Automatic Lenovo EasyCamera 0 deallocation
11:23:13.125 [shutdown-hook-1] INFO  com.github.sarxos.webcam.Webcam - Disposing webcam Lenovo EasyCamera 0
11:23:13.138 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Disposing webcam device Lenovo EasyCamera 0
11:23:13.138 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Closing webcam device
11:23:13.307 [shutdown-hook-1] DEBUG com.github.sarxos.webcam.Webcam - Webcam disposed Lenovo EasyCamera 0
BUILD SUCCESSFUL (total time: 4 seconds)

closed 2018-03-04 by cezzynahurt

sarxos, 2018-03-04:
Hi @cezzynahurt, 

I'm glad you were able to find the problem by yourself :) Happy hacking!

cezzynahurt, 2018-03-05:
Thank you sarxos :)