sarxos/webcam-capture #626: should be "minSize > curSize".

kbaik, 2018-03-10:
https://github.com/sarxos/webcam-capture/blob/7e82df6ba87cdff3edfc4c8318ac0f0769db4a4a/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L247

sarxos, 2018-03-14:
Hi @kbaik,

Indeed, I think it should be `minSize > curSize` instead of `minSize < curSize`.

This is because:
 
* minSize = the minimal size we require to store image (required size)
* curSize = how much buffer size remaining we have (available size)

And available size should always be greater than or equal to required size and error should be throwsn otherwise. Thus `minSize > curSize`.

Can you send pull request with a fix?

sarxos, 2018-03-27:
Fixed by @yuripourre in #631.

PR #631 (2018-03-27): Fix minSize < curSize
Fix #626 
Thanks to @kbaik
closed 2018-03-27 by sarxos