sarxos/webcam-capture #632: How to take image from rtsp?

sandipJava, 2018-03-27:
previusly I used Http IP camera, for that I used following code
` URL myUrl = new URL("http://192.168.15.253/axis-cgi/jpg/image.cgi");
			BufferedImage bufferedImage;
			bufferedImage = ImageIO.read(myUrl); // using axix camera
			java.io.ByteArrayOutputStream baos = new java.io.ByteArrayOutputStream();
			ImageIO.write(bufferedImage, "jpg", baos);`

sarxos, 2018-03-27:
Hi @sandipJava,

I have bad news. Unfortunately there is no possibility in Webcam Capture API to read from RTSP :( The only way to w/a this is to transcode RTSP to MJPEG and use Webcam Capture API on top of transcoded MJPEG.

sandipJava, 2018-03-27:
Can I have sample code, It will be more helpful to me

sarxos, 2018-03-27:
https://stackoverflow.com/questions/29699980/ffmpeg-restream-rtsp-to-mjpeg

sandipJava, 2018-03-28:
is xuggler will work for taking an image from rtsp?

sarxos, 2018-03-28:
Hi @sandipJava,

I'm not sure. Xuggler is discontinued and most of the code migrated to https://github.com/artclarke/humble-video, but please note that I'm not very familiar with this framework and can't give you exact details on how this can be done. 

For Xuggler, I found this: https://gist.github.com/sarxos/03b1196b5d845687d5ba but never test is.

sandipJava, 2018-03-29:
ok sir I will try

sandipJava, 2018-03-29:
Hello Sir I got following error

>SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "main" java.lang.RuntimeException: error Operation not permitted decoding video
	at com.xuggle.mediatool.MediaReader.decodeVideo(MediaReader.java:513)
	at com.xuggle.mediatool.MediaReader.readPacket(MediaReader.java:475)
	at xuggler.XugglerVideoTest.main(XugglerVideoTest.java:55)

jaysentang, 2018-04-06:
Try this code below, it's working fine for my ipcamera.

`

    static {
        String name = "Uniview Mini Dome";
        String rtsp = "rtsp://admin:123456@192.168.1.13/11";
        Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem(name, rtsp, new ArrayList<MediaListItem>()))));
    }

    public static void main(String[] args) throws InterruptedException, IOException {

        Webcam webcam = Webcam.getWebcams().get(0);
        webcam.setViewSize(WebcamResolution.VGA.getSize());
        
        WebcamPanel panel = new WebcamPanel(webcam);
        //panel.setFPSDisplayed(true);
        //panel.setImageSizeDisplayed(true);
        JFrame window = new JFrame("Webcam Panel");
        window.add(panel);
        window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
        window.pack();
        window.setVisible(true);

        Thread.sleep(4000);
      
        try {
			//start get image
			String ImgName = String.format( "testingvlc" + "-%d.jpg", System.currentTimeMillis());
			File FilePath = new File(FileSystems.getDefault().getPath("F:\\Image").toString() + File.separator + ImgName);
			BufferedImage m = webcam.getImage();
			if(m != null) {
				
				ImageIO.write(m, "JPG", FilePath);
				System.out.format("File %s has been saved in %s\n", ImgName, FilePath.getAbsolutePath());
				
			}	
		} catch (IOException t) {
			t.printStackTrace();
		} 
       
    }
`

sandipJava, 2018-06-15:
It is not working is showing below warning

**WARNING: VLCj does not support webcam devices discovery on Windows platform**

sarxos, 2018-06-15:
@sandipJava 

These warnings are just warnings, not errors, and therefore are harmless. Are you sure your RTSP URI is correct? If you can open it in VLC you should also be able to open it via Webcam Capture with VLCJ Capture Driver.

sandipJava, 2018-06-15:
@sarxos  it is opeming in VLC. I used same code that above given by @jaysentang . Its not saving image and not showing video in jframe

sarxos, 2018-06-15:
@sandipJava Can you show me the code sample?

sandipJava, 2018-06-15:
`import java.awt.image.BufferedImage;
import java.io.File;
import java.io.IOException;
import java.nio.file.FileSystems;
import java.util.ArrayList;
import java.util.Arrays;

import javax.imageio.ImageIO;
import javax.swing.JFrame;

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;

import uk.co.caprica.vlcj.medialist.MediaListItem;

public class ImageCapture {
	static {
	    String name = "Uniview Mini Dome";
	    String rtsp = "rtsp://192.168.15.233/StreamingSetting?version=1.0&action=login&username=admin&password=Nipun@123&getRTSPStream&ChannelID=1&ChannelName=Channel1";
	    Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem(name, rtsp, new ArrayList<MediaListItem>()))));
	}

	public static void main(String[] args) throws InterruptedException, IOException {

	    Webcam webcam = Webcam.getWebcams().get(0);
	    webcam.setViewSize(WebcamResolution.VGA.getSize());    
	    WebcamPanel panel = new WebcamPanel(webcam);
	    //panel.setFPSDisplayed(true);
	    //panel.setImageSizeDisplayed(true);
	    JFrame window = new JFrame("Webcam Panel");
	    window.add(panel);
	    window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
	    window.pack();
	    window.setVisible(true);
	    Thread.sleep(4000);
	    try {
			//start get image
			String ImgName = String.format( "testingvlc" + "-%d.jpg", System.currentTimeMillis());
			File FilePath = new File(FileSystems.getDefault().getPath("E:\\Image").toString() + File.separator + ImgName);
			BufferedImage m = webcam.getImage();
			if(m != null) {
				ImageIO.write(m, "JPG", FilePath);
				System.out.format("File %s has been saved in %s\n", ImgName, FilePath.getAbsolutePath());	
			}	
		} catch (IOException t) {
			t.printStackTrace();
		}   
	}
}`

sandipJava, 2018-06-15:
**rtsp://192.168.15.233/StreamingSetting?version=1.0&action=login&username=admin&password=Nipun@123&getRTSPStream&ChannelID=1&ChannelName=Channel1**

Using this link I am able to see video in VLC player

jaysentang, 2018-06-16:
@sandipJava have you modify your vlcjdevice class, please refer this issue #355 , i got it work after i read this issues.

sarxos, 2018-06-18:
@sandipJava These warnings are just _warnings_, not errors, and can be ignored. There is no functional impact due to these informations being printed.

sandipJava, 2018-06-18:
then why not video streaming in not coming

[image]

sandipJava, 2018-06-18:
means , what to do, i did not understood your last action

sarxos, 2018-06-18:
Hi @sandipJava,

Sorry for hanging with this for so long. I had some other important to do first. I did a quick look into the code to check VLCJ args in accordance to the code provided in #355 (thanks @jaysentang for pointing this out).

After these changes provided RTPS endpoints (at least the ones I tested) work fine.

I also prepared [small example](
https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-vlcj/src/example/java/WebcamRtspExample.java) based on your code:

```java
import java.io.IOException;
import java.util.ArrayList;
import java.util.Arrays;
import javax.swing.JFrame;
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.vlcj.VlcjDriver;
import uk.co.caprica.vlcj.medialist.MediaListItem;

public class WebcamRtspExample {

	static {
		String name = "Big Buck Bunny";
		String rtsp = "rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov";
		Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem(name, rtsp, new ArrayList<MediaListItem>()))));
	}

	public static void main(String[] args) throws InterruptedException, IOException {

		Webcam webcam = Webcam.getWebcams().get(0);
		webcam.setViewSize(WebcamResolution.VGA.getSize());
		WebcamPanel panel = new WebcamPanel(webcam);

		JFrame window = new JFrame("Webcam Panel");
		window.add(panel);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
```

After I run it:

[image]

**IMPORTANT!!!**

Please note that this fix has **not been released** yet and is available in SNAPSHOT version only!!! You need to download 0.3.13-SNAPSHOT version of webcam-capture-driver-vlcj from here:

https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture-driver-vlcj&v=0.3.13-SNAPSHOT

Please make sure to update all webcam-capture JARs to 0.3.12 in case you are using older version.

sarxos, 2018-06-18:
@sandipJava please let us know if this is working with your RTSP endpoint.

sandipJava, 2018-06-18:
your code is working Big Buck Bunny video is playing

sarxos, 2018-06-18:
Dammit :(

@sandipJava is there any way you can provide me with the access to the RTSP endpoint you are using? E.g. via public IP or a demo webcam somewhere in the internet?

sandipJava, 2018-06-18:
rtsp://192.168.15.233/StreamingSetting?version=1.0&action=login&username=adm
in&password=Nipun@123&getRTSPStream&ChannelID=1&ChannelName=Channel1

sarxos, 2018-06-19:
Hi @sandipJava 

The address you provided (192.168.15.233) is a [private network](https://en.wikipedia.org/wiki/Private_network) which I'm unable to connect to. It's available only from your own LAN, not from Internet. You have to have public IP with port forwarding configured on your router to temporarily allow users from internet to connect to your camera. The way to configure public IP accessibility strongly depends on your ISP (Internet Service Provider) and the router you have at your place.

sandipJava, 2018-06-19:
@sarxos , I dont have that much permission , Can we find any test IP camera from anywhere(Google)

sarxos, 2018-06-19:
@sandipJava,

In regards to what you wrote:

> Can we find any test IP camera from anywhere(Google)

This would be the best, but I have no time for this (I have my regular job and not much free time to spare), but if you can find such test IP anywhere in the Internet and provide it here with login and password, that would be great and I will use it to debug the problem.

polar1shu, 2019-04-16:
sorry to bother you @sarxos 
when I run WebcamRtspExample demo, it shows no image available
it woks macos 10.14, Is this error in my computer system problem?

megahertzon, 2020-06-24:
Hi friends, 
I have tried to test this code but it does not work:

 rtsp stream: 
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;
import java.io.IOException;
import java.util.ArrayList;
import java.util.Arrays;

import javax.swing.JFrame;
import uk.co.caprica.vlcj.medialist.MediaListItem;
import com.github.sarxos.webcam.ds.vlcj.VlcjDriver;

public class Main {

    static {
        String name = "Big Buck Bunny";
        String rtsp = "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov";
        Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem(name, rtsp, new ArrayList<>()))));
    }

    public static void main(String[] args) throws InterruptedException, IOException {

        Webcam webcam = Webcam.getWebcams().get(0);
        webcam.setViewSize(WebcamResolution.VGA.getSize());
        WebcamPanel panel = new WebcamPanel(webcam);

        JFrame window = new JFrame("Webcam Panel");
        window.add(panel);
        window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
        window.pack();
        window.setVisible(true);
    }
}

The pom file:

<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>
    <groupId>co.com.iotech</groupId>
    <artifactId>WebcamRTSP</artifactId>
    <version>1.0-SNAPSHOT</version>
    <packaging>jar</packaging>
    <properties>
        <project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>
        <maven.compiler.source>1.8</maven.compiler.source>
        <maven.compiler.target>1.8</maven.compiler.target>
        <mainClass>co.com.iotech.webcamrtsp.Main</mainClass>
    </properties>
    
    <pluginRepositories>
        <pluginRepository>
            <id>central</id>
            <name>Central Repository</name>
            <url>https://repo.maven.apache.org/maven2</url>
            <layout>default</layout>
            <snapshots>
                <enabled>false</enabled>
            </snapshots>
            <releases>
                <updatePolicy>daily</updatePolicy>
            </releases>
        </pluginRepository>
    </pluginRepositories>
    
    <build>
        <plugins>
            <plugin>
                <artifactId>maven-dependency-plugin</artifactId>
                <executions>
                    <execution>
                        <phase>install</phase>
                        <goals>
                            <goal>copy-dependencies</goal>
                        </goals>
                        <configuration>
                            <outputDirectory>${project.build.directory}/lib</outputDirectory>
                        </configuration>
                    </execution>
                </executions>
            </plugin>
            <plugin>
                <artifactId>maven-jar-plugin</artifactId>
                <configuration>
                    <archive>
                        <manifest>
                            <addClasspath>true</addClasspath>
                            <classpathPrefix>lib/</classpathPrefix>
                            <mainClass>${mainClass}</mainClass>
                        </manifest>
                    </archive>
                </configuration>
            </plugin>
        </plugins>
    </build>
    

    

    
    <dependencies>
        
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.13-SNAPSHOT</version>
        </dependency>
        
        <dependency>
            <groupId>uk.co.caprica</groupId>
            <artifactId>vlcj</artifactId>
            <version>2.4.1</version>
        </dependency>
        <dependency>
            <groupId>org.slf4j</groupId>
            <artifactId>slf4j-api</artifactId>
            <version>1.7.5</version>
        </dependency>
        <dependency>
            <groupId>org.slf4j</groupId>
            <artifactId>slf4j-log4j12</artifactId>
            <version>1.7.5</version>
        </dependency>
        
        
        <dependency>
            <groupId>group.id</groupId>
            <artifactId>artifact.id</artifactId>
            <version>1.0</version>
        </dependency>
        <dependency>
            <groupId>org.apache.httpcomponents</groupId>
            <artifactId>httpclient</artifactId>
            <version>4.3.4</version>
        </dependency>
        
        <dependency>
            <groupId>vlcjar</groupId>
            <artifactId>vlcjar</artifactId>
            <version>1.0</version>
        </dependency>
    </dependencies>
    <repositories>
        
        <repository>
            <id>central</id>
            <name>Central Repository</name>
            <url>https://repo.maven.apache.org/maven2</url>
            <layout>default</layout>
            <snapshots>
                <enabled>false</enabled>
            </snapshots>
        </repository>
        <repository>
            <id>Sonatype OSS Snapshot Repository</id>
            <url>http://oss.sonatype.org/content/repositories/snapshots</url>
        </repository>
    </repositories>
    
    
</project>

please help me!