sarxos/webcam-capture #633: WebCam Capture Capturing same image again and again

Abdulmoizaftab, 2018-03-27:
Images are being saved to a temp folder before being compressed and saved to a different folder. But the program is making same images over and over in temp folder and with automatically being refreshed sometimes for few hours and then again same old image.

No issue logged at log4j file logger and i have even coded to log unhandled exceptions.
[image]
[image]

Above Images shows reoccurnce.   173651155_1287_1017_0081_1.jpg (First part of image name is timestamp)

Main
File TempDir = new File("./Temp");
     int camdetected= Webcam.getWebcams().size();
          System.out.println("No. of Cameras detected="+camdetected);
          logger.info("No. of Cameras detected="+camdetected);
          System.out.println("Names="+Webcam.getWebcams());
          logger.info("Names="+Webcam.getWebcams());
  if(Camera1==null){
                            cam1.Open(imwidth,imhieght);
                            System.out.println("Camera 1 Opened ("+imwidth+"x"+imhieght+")");
                            logger.info("Camera 1 Opened ("+imwidth+"x"+imhieght+")");
}
 String filename=cam1.GetImage(TempDir+"\\Image1.png");

public class Camera1 {
     public static Logger logger = LogManager.getLogger(Camera1.class.getName());
    public void Open(int w, int h){
        try {
        AISAgent.Camera1 = Webcam.getWebcams().get(0);      
        AISAgent.Camera1.setViewSize(new Dimension(w,h));
        AISAgent.Camera1.open();
          } catch (Exception ex) {
              logger.error( "Error:",ex);
         }
    }
    public String GetImage(String filename) {
        
         try {
             ImageIO.write(AISAgent.Camera1.getImage(), "PNG", new File(filename));
             
             
         } catch (IOException ex) {
              logger.error( "Error:",ex);
         }
         return filename;
    }

sarxos, 2018-03-27:
Did you check if a set of PNG images captured from the webcam also contain duplicates?

Abdulmoizaftab, 2018-03-27:
Yes, Same images appears to be in the temp folder after some time. Check Timestamp same image was captured 09:19 and other images were taken maybe by restarting the program and again at 17:35 and onwards previous email was starting to be generated.

Abdulmoizaftab, 2018-03-28:
I am rewriting the same image in temp folder, which appears to be not being able to rewritten at some stage (not normally) or replaced by an image may be in the program cache(if any). BTW Thanks in advance.

Abdulmoizaftab, 2018-03-29:
Please see below stacktrace, Camera at once started not to respond, ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!

3246750 2018-03-28 00:55:05.557 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3246750 2018-03-28 00:55:05.557 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3249751 2018-03-28 00:55:08.558 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3249752 2018-03-28 00:55:08.559 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3249752 2018-03-28 00:55:08.559 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3252753 2018-03-28 00:55:11.560 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3252753 2018-03-28 00:55:11.560 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3252754 2018-03-28 00:55:11.561 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3255756 2018-03-28 00:55:14.563 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3255757 2018-03-28 00:55:14.564 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3255757 2018-03-28 00:55:14.564 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3258758 2018-03-28 00:55:17.565 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3258759 2018-03-28 00:55:17.566 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3258759 2018-03-28 00:55:17.566 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3261760 2018-03-28 00:55:20.567 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3261761 2018-03-28 00:55:20.568 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3261761 2018-03-28 00:55:20.568 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3264762 2018-03-28 00:55:23.569 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3264763 2018-03-28 00:55:23.570 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3264763 2018-03-28 00:55:23.570 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3267764 2018-03-28 00:55:26.571 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3267765 2018-03-28 00:55:26.572 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3267765 2018-03-28 00:55:26.572 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3270766 2018-03-28 00:55:29.573 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3270767 2018-03-28 00:55:29.574 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3270767 2018-03-28 00:55:29.574 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3273769 2018-03-28 00:55:32.576 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3273770 2018-03-28 00:55:32.577 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3273770 2018-03-28 00:55:32.577 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3276771 2018-03-28 00:55:35.578 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3276772 2018-03-28 00:55:35.579 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3276772 2018-03-28 00:55:35.579 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3279773 2018-03-28 00:55:38.580 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3279774 2018-03-28 00:55:38.581 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
3279774 2018-03-28 00:55:38.581 [main] INFO  c.g.s.webcam.WebcamDiscoveryService - Discovery service has been disabled and thus it will not be started
[... cut ...]
37913142 2018-03-28 10:32:51.949 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
37918148 2018-03-28 10:32:56.955 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
37923148 2018-03-28 10:33:01.955 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
37928148 2018-03-28 10:33:06.955 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!