sarxos/webcam-capture #636: WebcamMotionDetector for IP Camera

sandipJava, 2018-04-16:
WebcamMotionDetector is great feature. But it is only working for web cameras , How I can used ths for IP camera

sarxos, 2018-04-16:
Hi @sandipJava, 

Use [webcam-capture-driver-ipcam](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam) instead of a default one. Examples demonstrating how to use IP camera driver can be found below. I hope you can glimpse the idea.

https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ipcam/src/examples/java

The usage of motion detector remains the same.

sandipJava, 2018-04-16:
I am using simple 
`http://localhost:8090/API/Cam`  IP URL so How I can add this into your code