sarxos/webcam-capture #638: Webcam - Transcode webcam images into live h264 stream example

adelanteDev, 2018-04-24:
Hi, I have modified the above example for my needs. Now the structure of my project looks like this: Client1 sends the image to the server, the server saves it to disk and sends it to Client2. When I want my server to run more threads (I need several ports for different webcams).

My server : 

`public class Server {

    static List<Integer> ports = new ArrayList<Integer>();

    public static void main(String[] args) {
        ports.add(20000);
        ports.add(20001);

        if (args.length == 0) {
            for (int i = 0; i < ports.size(); i++)
                initPorts("localhost", ports.get(i));
        } else {

        }
    }

    public static void init(String host, int port) {
        StreamServerAgent serverAgent = new StreamServerAgent();
        serverAgent.start(new InetSocketAddress(host, port));
    }

    public static void initPorts(final String host, final int port) {
        new Thread() {
            @Override
            public void run() {
                init(host, port);
            }
        }.run();
    }
}`

Error : 

> #
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000006eed40ff, pid=5520, tid=0x00000000000019dc
#
# JRE version: Java(TM) SE Runtime Environment (8.0_161-b12) (build 1.8.0_161-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.161-b12 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [xuggle886819521010380591.dll+0x7940ff]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0x000000001b987800):  JavaThread "New I/O worker #2" [_thread_in_native, id=6620, stack(0x000000001cbb0000,0x000000001ccb0000)]

siginfo: ExceptionCode=0xc0000005, reading address 0xffffffffffffffff

Registers:
RAX=0x0000000000000004, RBX=0x900172005237567b, RCX=0x000000001d3eeeb8, RDX=0x0000000000000018
RSP=0x000000001ccad7a0, RBP=0x000000001ccadb1c, RSI=0x000000001da58170, RDI=0x90017200523758bb
R8 =0x000000001ccadb1c, R9 =0x000000001da57de8, R10=0x0000000000000019, R11=0x0000000000000001
R12=0x000000001da57de8, R13=0x00000000197d10a8, R14=0x000000001ccae458, R15=0x000000001b987800
RIP=0x000000006eed40ff, EFLAGS=0x0000000000010286

Top of Stack: (sp=0x000000001ccad7a0)
0x000000001ccad7a0:   000000001ccad640 000000001ccad820
[... 258 lines cut ...]
java_command: Server.jar
java_class_path (initial): Server.jar
Launcher Type: SUN_STANDARD

Environment Variables:
JAVA_HOME=C:\Program Files\Java\jdk1.8.0_161
PATH=C:\Users\xxx\bin;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\local\bin;C:\Program Files\Git\usr\bin;C:\Program Files\Git\usr\bin;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\Users\xxx\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\Program Files\Git\cmd;C:\apache-maven-3.5.3\bin;C:\Users\xxx\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Git\usr\bin\vendor_perl;C:\Program Files\Git\usr\bin\core_perl
USERNAME=xxx
SHELL=C:\Program Files\Git\usr\bin\bash.exe
DISPLAY=needs-to-be-defined
OS=Windows_NT
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows 10.0 , 64 bit Build 16299 (10.0.16299.15)

CPU:total 4 (initial active 4) (4 cores per cpu, 1 threads per core) family 6 model 60 stepping 3, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, avx2, aes, clmul, erms, lzcnt, tsc, tscinvbit, bmi1, bmi2

Memory: 4k page, physical 6215352k(1888640k free), swap 12244664k(4389836k free)

vm_info: Java HotSpot(TM) 64-Bit Server VM (25.161-b12) for windows-amd64 JRE (1.8.0_161-b12), built on Dec 19 2017 17:52:25 by "java_re" with MS VC++ 10.0 (VS2010)

time: Tue Apr 24 08:44:50 2018
elapsed time: 22 seconds (0d 0h 0m 22s)

sarxos, 2018-06-18:
@adelanteDev Czy ten problem nadal występuje, czy udało się Tobie go ominąć? W razie co służę supportem, jednak z uwagi na mocno ograniczony czas zaznaczam, że moge nie być dostepny od ręki.

adelanteDev, 2018-06-20:
Jestem dużo dalej jeśli chodzi o kod niż w momencie którym to pisałem. W tamtym momencie udało mi się usunąć ten błąd (przynajmniej tak myślałem). W tej chwili mój projekt wysyła z różnych komputerów obraz na jeden serwer, tam jest obraz też zapisywany do plików mp4. Natomiast różni klienci mogą podłączać się pod serwer i oglądać na żywo obraz z kamerek. Niestety okazało się, że błąd nadal występuje. Program się wysypuje po około dniu, dwóch działania. Dodatkowo zauważyłem że im większa liczba fpsów wysyłanych tym szybciej program się wysypuje. W argumencie wywołania oczywiście dałem -Xmx4096. Idąc za radami wyczytanymi w internecie skompilowałem ręcznie bibliotekę xuggler 5.5. Natomiast to również nie rozwiązało problemu. 

Błąd z dziś : 
https://pastebin.com/VG2wypTv

Byłbym wdzięczny za pomoc, bo to blokuje bardzo mój projekt.

sarxos, 2018-06-20:
@adelanteDev,

Wygląda na to, że to nie Xuggler wywala Twoją apkę, tylko BridJ, kiedy przychodzi czas GC i finalizacji śmieci :( Trochę to niefajne, bo jest to jedna z fundamentalnych bibliotek, z których korzysta domyślny Capture Driver projektu Webcam Capture API.

```plain
Stack: [0x000000001d0b0000,0x000000001d1b0000],  sp=0x000000001d1aeed0,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [ntdll.dll+0x4f23c]
C  [kernel32.dll+0x21a0a]
C  [bridj.dll+0xa750]
C  [bridj.dll+0x7288]
C  0x00000000023b7f74
 
Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  org.bridj.JNI.free(J)V+0
j  org.bridj.Pointer$FreeReleaser.release(Lorg/bridj/Pointer;)V+192
j  org.bridj.Pointer$5.release(Lorg/bridj/Pointer;)V+14
j  org.bridj.Pointer$3.release()V+19
j  org.bridj.Pointer$3.finalize()V+1
J 2476% C2 java.lang.ref.Finalizer$FinalizerThread.run()V (55 bytes) @ 0x00000000029828a8 [0x00000000029825c0+0x2e8]
v  ~StubRoutines::call_stub
```

Rozwiązanie tego problemu lezy prawdopodobnie po stronie samej biblioteki BridJ, jednak Olivier (autor tej biblioteki) ostatnimi czasy nie ma za duzo czasu na jej rozwój, więc pozostają Tobie tak naprawdę dwie opcje:

1. Upgrade BridJ do najnowszej wersji z nadzieją, że to magicznie naprawi crashowanie się Twojej apki,
2. Wymianę Capture Driver'a na inny, który nie korzysta z BridJ.

Sprawdź 1, a jeśli to nie pomoże, to od razu idź w opcję 2. Samodzielne debugowanie i naprawianie BridJ jest hardcorowe, bo to naprawdę skomplikowany projekt i może zająć więcej czasu niż się wydaje.

Odnosnie opcji 2, skoro pracujesz na Windowsie, to szczególnie polecam [webcam-capture-driver-native
](https://github.com/frankpapenmeier/webcam-capture-driver-native) autorstwa Franka Papenmeier'a z uwagi na fakt, że na najniższym poziomie wykorzystuje najnowsze API Microsoftu do obsługi video, tj. Media Foundation. Inne, starsze Capture Drivery kompatybilne z Windowsem korzystają z bibliotek, które używają starej technologii Direct3D.

Pamiętaj, że przy wymianie drivera powinieneś usunąć BridJ (wraz z jego zależnościami) z classpath aby nie robił już problemów. Jeśli korzystasz z Mavena, to nie powinien być problem - wystarczy dać exluda na dependency, ale jesli nie korzystasz z Mavena, to musisz je usunąc ręcznie i zweryfikowac czy coś nie zostało popsute.

adelanteDev, 2018-06-21:
@sarxos 

Wielkie dzięki za odpowiedź. Jeśli chodzi o punkt 1, to mam wersje 0.7.0 więc chyba najnowszą. Przeszedłem do punktu drugiego : 
1. Ściągnąłem jar'a webcam-capture-driver-native.jar i dodałem go do projektu.
2. Pliki .dll wrzuciłem do katalogu bin Javy.
3. W programie ustawiłem drivery tak : 

`    public static class MyCompositeDriver extends WebcamCompositeDriver {

        public MyCompositeDriver() {
            add(new NativeWebcamDriver());
            add(new IpCamDriver());
        }
    }

    static {
        Webcam.setDriver(new MyCompositeDriver());
    }`

Inicjuję kilka kamerek. Każda w odzdzielnym wątku. W ten sposób : 

`        Thread thread2 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2000, 0, 10);
                try {
                    Thread.sleep(1000L);
                } catch (InterruptedException e) {
                    e.printStackTrace();
                }

            }
        };
        Thread thread3 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2001, 1, 10);
                try {
                    Thread.sleep(1000L);
                } catch (InterruptedException e) {
                    e.printStackTrace();
                }
            }
        };
        Thread thread1 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2002, 2, 10);
                try {
                    Thread.sleep(1000L);
                } catch (InterruptedException e) {
                    e.printStackTrace();
                }
            }
        };`

Problem polega na tym, że nie wiem dlaczego. Ale czasami uruchamiana jest jedna kamerka, czasami dwie, a czasami wszystkie. Zupełnie random'owo. Żadne wyjątki nie lecą.

adelanteDev, 2018-06-21:
@sarxos

Wielkie dzięki za odpowiedź. Jeśli chodzi o punkt 1, to mam wersje 0.7.0 więc chyba najnowszą. Przeszedłem do punktu drugiego :

    Ściągnąłem jar'a webcam-capture-driver-native.jar i dodałem go do projektu.
    Pliki .dll wrzuciłem do katalogu bin Javy.
    W programie ustawiłem drivery tak :

` public static class MyCompositeDriver extends WebcamCompositeDriver {

    public MyCompositeDriver() {
        add(new NativeWebcamDriver());
        add(new IpCamDriver());
    }
}

static {
    Webcam.setDriver(new MyCompositeDriver());
}`

Inicjuję kilka kamerek. Każda w odzdzielnym wątku. W ten sposób :

      Thread thread2 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2000, 0, 10);
            }
        };
        Thread thread3 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2001, 1, 10);
                }
        };
        Thread thread1 = new Thread() {
            public void run() {
                StreamingClient stream = new StreamingClient();
                stream.init("192.168.1.91", 2002, 2, 10);
                }
        };

Czasami występuje problem, że nie wszystkie kamerki się inicjują, mimo tego że lista kamerek zawsze wynosi 3. Ale to sporadycznie. Przetestuję i dam znać czy błąd dalej występuje.

sarxos, 2018-06-21:
@adelanteDev,

Skoro otwierdzanie w osobnych wątkach nie działa, to spróbuj otworzyć kamerki w głównym wątku, a potem przekaż referencję zamiast tych indeksów do metody `init(...)`.

Przykładowo:

```java
Webcam w0 = Webcam.getWebcams().get(0);
Webcam w1 = Webcam.getWebcams().get(1);
Webcam w2 = Webcam.getWebcams().get(2);

Thread thread0 = new Thread(() -> new StreamingClient().init("192.168.1.91", 2002, w0, 10));
Thread thread1 = new Thread(() -> new StreamingClient().init("192.168.1.91", 2002, w1, 10));
Thread thread2 = new Thread(() -> new StreamingClient().init("192.168.1.91", 2002, w2, 10));
```

adelanteDev, 2018-06-22:
@sarxos 

Dziwna sprawa, na moim komputerze wystarczyło to, że między wątkami dałem `Thread.sleep(1000L).`
Natomiast na innych przy użyciu mojego kodu, działała tylko pierwsza kamerka, a przy użyciu Twojego dostawałem wyjątek "Webcam has already been locked". Dziwna sprawa bo nic nie używało pozostałych kamerek. Użyłem `webcam.getLock().disable(); ` i to rozwiązało problem z ilością działających kamerek. Jeśli chodzi o błąd z EXCEPTION_ACCESS_VIOLATION to dam znać jutro bądź pojutrze, bo zazwyczaj włąśnie tyle musiał chodzić klient wysyłający obraz aby ten wyjątek wyskoczył.

sarxos, 2018-06-22:
@adelanteDev 

Jeśli musisz robić `webcam.getLock().disable()` to upewnij się, że nie masz drugiego procesu w tle, który też korzysta z tych kamerek. To jest najczęściej powód błędu na lockach (webcam has already been locked). Często jest tak, że ktoś używa wątków niedemonicznych (każdy wątek jest domyślnie niedemoniczny), które nie kończą się przy tzw. gracefull shutdown (czyli śmierć wątka `main`) i można je ubić tylko poprzez zakończenie aplikacji za pomocą `System.exit()`. Takie wątki niedemoniczne cały czas utrzymują proces aktywny, a w tym procesie zablokowane kamerki. Można to łatwo zaobserwować jak zrobisz na Linuxie `ps -ef | grep java` a na Windowsie jak wejdziesz w manager procesów i tam będzie więcej procesów java niż powinno być. Innym powodem (potencjalnym) błędu na locku może być bardzo szybki restart procesu po jego wywaleniu się (po crashu). Generalnie ten problem wskazuje, że "coś" jest nie tak i lepiej w to spojrzeć, aby mieć pełną wiedzę na temat działanie aplikacji. Jeśli włączysz dokładne debug logi to tam powinno być trochę więcej informacji kiedy i co jest blokowane.

adelanteDev, 2018-06-26:
@sarxos 

Okay przetestowane, powyższy błąd już nie występuje. Dzięki za poświęcony czas. Będę wprowadzał jeszcze kilka funkcji do swojego projektu. Gdybym sobie z czymś nie radził, będę pytał w innych wątkach. Dzięki i powodzenia!

sarxos, 2018-06-26:
@adelanteDev To super wiadomość :) Powodzenia z projektem!

closed 2018-06-26 by sarxos