sarxos/webcam-capture #640: Biblioteka sarxos, pytanie

marwis95, 2018-05-15:
Witam.
Używam Twojej biblioteki Sarxos do przechwytywania obrazu z kamerki w tablecie.
Potrzebuję zrobić tak w swojej aplikacji, aby mieć podgląd obrazu w małej rozdzielczości i możliwość wykonywania zdjęć w dużej.
Jeżeli zrobię podgląd w dużej rozdzielczości to jest problem z płynnością wyświetlania, dlatego potrzebuję zmniejszyć rozdzielczość podglądu, jednocześnie zachowując możliwość wykonywania zdjęć w pełnej rozdzielczości.
Czy coś takiego jest możliwe w Twojej bibliotece? Jeżeli tak to jak sie zabrać za coś takiego?

Pozdrawiam Marwis95

sarxos, 2018-05-15:
Hej @marwis95,

Pytanie tylko czy problem z płynnością spowodowany jest przez:

* wielkość obrazka, czy
* niższy FPS.

W pierwszym przyp. możesz użyc interfejsu `WebcamPanel.Painter` aby zmienić wielkość obrazka przed jego renderowaniem. W tym przykładzie wykonuję obrót obrazka za pomocą metody `rotate.filter(image, null)` jednak zamiast obracać obraz możesz też zmniejszyć jego rozmiar, tylko musisz zastosować inny filtr lub zrobić to ręcznie (nie jest to trudne, na google jest mnóstwo przykładów). 

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/WebcamPanelRotationExample.java

Natomiast jeśli problem płynności spowodowany jest niskim FPS, co jest dosyć częstym objawem przy wysokich rozdzielczościach (np. HD720p) to niestety rozwiązanie może nie być tak proste.

sarxos, 2018-06-20:
@marwis95 daj znać czy udało się rozwiązać Twój problem.