sarxos/webcam-capture #641: Ambiguty error

mdagraca, 2018-05-23:
Hi, I'm having the same error that [https://github.com/sarxos/webcam-capture/issues/593](https://github.com/sarxos/webcam-capture/issues/593)
 I'm using opencv openpnp version 3.2.0-1, Java 1.8.0_51 and I run my app in OSX 10.11.6 and Windows 10. 
The error message is:
objc[6899]: Class CaptureDelegate is implemented in both /private/var/folders/j1/_q1t_00d0fg66bsgtgf8zjgc0000gp/T/opencv_openpnp5786236370894680416/nu/pattern/opencv/osx/x86_64/libopencv_java320.dylib and /var/folders/j1/_q1t_00d0fg66bsgtgf8zjgc0000gp/T/BridJExtractedLibraries4705987337836921372/libOpenIMAJGrabber.dylib.
One of the two will be used. Which one is undefined.
 Did you find a solution?

yuripourre, 2018-05-25:
Are you using maven to setup your project?
If so, add exclusion tag to webcam-capture:

```
<dependency><groupId>com.github.sarxos</groupId> <artifactId>webcam-capture</artifactId> <version>${project.version}</version> <exclusions> <exclusion> <groupId>com.nativelibs4java</groupId> <artifactId>bridj</artifactId> </exclusion></exclusions> </dependency>
```

mdagraca, 2018-05-28:
Yes,  I use maven. I will test your solution. Thanks