sarxos/webcam-capture #642: Pomijanie klatek przy streamowanie z kamery IP

adelanteDev, 2018-05-24:
Cześć, mam następujący problem. Wykorzystałem Twój przykład odnośnie ipcam tj.  FoscamJpegPullExample.java [https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/FoscamJpegPullExample.java](url) oraz przykład związany z live-streaming tj. webcam-capture-live-streaming [https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-live-streaming](url)

`public class FoscamJpegPullExample {

    static {
        Webcam.setDriver(new IpCamDriver());
    }

    public static void main(String[] args) throws MalformedURLException {

        String name = "Test #255";
        String url = "http://ce3014.myfoscam.org:20054/snapshot.cgi";
        IpCamMode mode = IpCamMode.PULL;
        IpCamAuth auth = new IpCamAuth("user", "user");

        IpCamDeviceRegistry.register(name, url, mode, auth);

        WebcamPanel panel = new WebcamPanel(Webcam.getWebcams().get(0));
        panel.setFPSLimit(1);
        panel.setFPSLimited(true);
        JFrame f = new JFrame("Test #255 PULL");
        f.add(panel);
        f.pack();
        f.setVisible(true);
        f.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
    }
}`

oraz 

`
public class StreamServer {

	/**
	 * @author kerr
	 * @param args
	 */
	public static void main(String[] args) {
		Webcam.setAutoOpenMode(true);
		Webcam webcam = Webcam.getDefault();
		Dimension dimension = new Dimension(320, 240);
		webcam.setViewSize(dimension);

		StreamServerAgent serverAgent = new StreamServerAgent(webcam, dimension);
		serverAgent.start(new InetSocketAddress("localhost", 20000));
	}

}`

I stworzyłem coś takiego : 

` static {
        Webcam.setDriver(new IpCamDriver());
    }

    public static void main(String[] args) throws MalformedURLException {

        String name = "Test #255";
        String url = "http://ce3014.myfoscam.org:20054/snapshot.cgi";
        IpCamMode mode = IpCamMode.PULL;
        IpCamAuth auth = new IpCamAuth("user", "user");

        IpCamDeviceRegistry.register(name, url, mode, auth);

        Webcam webcam = Webcam.getWebcams().get(0);
        Dimension dimension = new Dimension(320, 240);
	webcam.setViewSize(dimension);

	StreamServerAgent serverAgent = new StreamServerAgent(webcam, dimension);
	serverAgent.start(new InetSocketAddress("localhost", 20000));
    }`

Ustawiam liczbę FPS na 1. Problem jest taki, że co jakiś czas (dosyć często np co 3s) jest pomijana jakaś klatka.

sarxos, 2018-06-18:
Cześć @adelanteDev,

Niestety ten przykład do którego się odwołujesz (live streaming example) nie był napisany przeze mnie i nie za bardzo jestem w stanie z głowy powiedzieć co tam jest nie tak i czemu niektóre klatki są pomijane. Podejrzewam, że gdyby podpiąc tam się debuggerem i zobaczyć co się dzieje, to można zobaczyć czy to jakieś timing issue, czy może sieć albo kamera zamula, niestety nie mogę tego zrobić ze względu na brak czasu. Sugeruję w pierwszej kolejności [włączyć bardzo dokładne logi](https://github.com/sarxos/webcam-capture/wiki/How-To-Enable-Debug-Logs) (najlepiej poziom trace) i dokładnie przeanalizować aby zrozumieć co się dzieje pod spodem.

Mikilis, 2018-11-24:
wy jesteście polakami xDDD

sarxos, 2018-11-25:
@Mikilis Właśnie zdobyłeś tytuł geniusza oczywistości. :smile: