sarxos/webcam-capture #645: Cannot execute task.

SynGreis, 2018-06-08:
trying to do a basic device count.

`import com.github.sarxos.webcam.Webcam;

import java.util.List;

public class Test{
    public static void main(String[] args){
        List<Webcam> list = Webcam.getWebcams();
        System.out.println(list.size());
    }
}`

Getting hung up when trying to do anything with webcam.

`[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:789)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:768)
	at Test.main(Test.java:7)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 4 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:118)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NoClassDefFoundError: org/bridj/cpp/CPPObject
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:56)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Caused by: java.lang.ClassNotFoundException: org.bridj.cpp.CPPObject
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 17 more

Process finished with exit code 1`

Not big on words just trying to figure out why this suddenly stopped working.

I am currently using [http://repo1.maven.org/maven2/com/github/sarxos/webcam-capture/0.3.10/webcam-capture-0.3.10.jar](url) and added it to my class path.

I also confirmed that my webcam is indeed working. [https://www.staples.ca/en/Logitech-C170-Webcam/product_100287_1-CA_1_20001](url)

It was working perfectly using pom.xml but i wanted to use jar for clean directories. What am i missing?

sarxos, 2018-06-08:
BridJ JAR is missing in your classpath. You can search other issues. People are forgetting to add it all the time :)

sarxos, 2018-06-08:
Just make sure to add other dependencies as well! You can find ZIP bundle on the project main page. Take care!

closed 2018-06-08 by sarxos

SynGreis, 2018-06-08:
Works great. I had no idea that that jar was needed. I checked your bundle and tried making a jar by using mvn clean package but i was getting compile errors. Your driver-vlcj pom.xml no longer works as it says it no longer exists.

Would you be able to verify this?