sarxos/webcam-capture #646: Multithreading capture images from few cameras

adelanteDev, 2018-06-14:
Hi, I would like to capture images from few cameras in the same time. But I would like the errors or delays of downloading images from one camera to not affect all. At the moment the problem is that when one camera has the exception "timeout getting image" the rest is waiting for it to be downloaded or until the above mentioned exception pops up. I would like to make it independent. I'm using a modified example of : [https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-examples/webcam-capture-live-streaming](url)

sarxos, 2018-06-18:
@adelanteDev You should be able to achieve that, at least partially, by opening webcams in a non-blocking mode. By default when you call `webcam.open()` it's being open in a blocking mode, which means that every call to `webcam.getImage()` will wait for the new image being available. However, when you open webcam with `webcam.open(true)` the webcam will be open in a non-blocking asynchronous mode, which means that `webcam.getImage()` will never wait for new image to be available but return currently available image every time when you call it. Please note that there are some downsides of a non-blocking mode which has to be taken into account! If you invoke `getImage()` few times and very fast in a row you will get the same image (this is because webcam will return current image, not the new one). This mode may require some changes in your program architecture (you may need to switch from blocking and synchronous design to non-blocking asynchronous design).

adelanteDev, 2018-06-20:
Okay, thank you for your help. So I will do that :)

sarxos, 2018-06-20:
Ok, in this case I'm closing the ticket. Good luck!

closed 2018-06-20 by sarxos