sarxos/webcam-capture #647: SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder". SLF4J:     Defaulting to no-operation (NOP) logger implementation

sufyansalim, 2018-06-18:
I have tried all the slf4j files given in the link still get this error on itellij idea ide
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "main" java.lang.IllegalArgumentException: image == null!
	at javax.imageio.ImageTypeSpecifier.createFromRenderedImage(ImageTypeSpecifier.java:925)
	at javax.imageio.ImageIO.getWriter(ImageIO.java:1592)
	at javax.imageio.ImageIO.write(ImageIO.java:1520)
	at Main.main(Main.java:14)

sufyansalim, 2018-06-18:
I have solved it Thank You anyways...

closed 2018-06-18 by sufyansalim

ZhongxuanWang, 2019-10-27:
> I have solved it Thank You anyways...

Sorry, could you plz share how did you solve it? I met the same problem...

sufyansalim, 2019-11-09:
It has been a while i will have to check the code if i still have it ,i think it was the issue with the IDE i used eclipse instead of itnellij