sarxos/webcam-capture #651: Webcam.close() does not return

jackzhp, 2018-07-03:
Webcam cam=...;

                    System.out.println("Cancelled, closing cam");
                    cam.close();
                    System.out.println("Cam closed");

For dummyDevice with jdk 1.8.0_172, Webcam.close() does not return. The above two print lines, first is printed, 2nd is never printed. 

then can not be reopened, and I found a java thread is never terminated.
the java process should be killed, and the lock file under /tmp/ should be deleted, and next open is good. seems has nothing to do with dummydevice,  it is the executor service can not be terminated.

jack

jonnio, 2019-02-20:
I have the same issue. Here is the thread dump where the close task is waiting for the token. Running on Java 11.
 
`java.lang.Thread.State: WAITING
	  at jdk.internal.misc.Unsafe.park(Unsafe.java:-1)
	  at java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
	  at java.util.concurrent.SynchronousQueue$TransferQueue.awaitFulfill(SynchronousQueue.java:759)
	  at java.util.concurrent.SynchronousQueue$TransferQueue.transfer(SynchronousQueue.java:689)
	  at java.util.concurrent.SynchronousQueue.take(SynchronousQueue.java:920)
	  at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:70)
	  at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	  at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	  at com.github.sarxos.webcam.ds.cgt.WebcamCloseTask.close(WebcamCloseTask.java:20)
	  at com.github.sarxos.webcam.Webcam.close(Webcam.java:372)`

nestormariomace, 2019-03-06:
I have a similar problem that shows up randomly (after many minutes).
I also used the two println, before and after the webcamX.close(), and the second never prints.
In my case, using Windows 10, Netbeans 7.3.1 and jdk 1.7, the application shuts down with a message in the Netbeans Output window:   Java result -1073741819

n1mda, 2019-08-20:
Seeing this issue as well, using webcam-capture 0.3.13-SNAPSHOT and vlcj driver for an Axis network camera.