sarxos/webcam-capture #659: picam bridj error dlopen error

SynGreis, 2018-08-29:
I ran this exact jar on my windows machine and laptop that contains raspbian as its OS. works great.

But when i put this on the rpi and use nothing but the picam i get these errors...

```
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
# BridJ: dlopen error when loading /tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so : libjpeg.so.8: cannot open shared object file: No such file or directory
java.lang.RuntimeException: Library 'OpenIMAJGrabber' was not loaded successfully from file '/tmp/BridJExtractedLibraries1203045396730957235/OpenIMAJGrabber.so'
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1072)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1022)
	at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:904)
	at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
	at org.bridj.BridJ.initialize(BridJ.java:1128)
	at org.bridj.NativeObject.<init>(NativeObject.java:50)
	at org.bridj.StructObject.<init>(StructObject.java:46)
	at org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)

```

Is this looking for a different bridj and does this sarxos webcam library work with the picam? I really got attached to this library and would prefer not to go to another api if it can handle the picam.

sarxos, 2018-10-08:
Hi @SynGreis, 

Sorry for late response. I was on a vacation whole August and I simply missed this issue. Did you try to replace BridJ version?

sarxos, 2018-10-08:
@SynGreis,

I see the problem. You are missing `libjpeg` on your RasPi.

This should solve the problem:

```
sudo apt-get install libjpeg
```

SynGreis, 2018-10-13:
tried it, says unable to locate package

sarxos, 2018-10-13:
Try with `libjpeg8`.

SynGreis, 2018-10-14:
i was able to download and install fine. Unfortunately it still doesnt detect my camera is there. No errors that would halt the program but it thinks there is no camera.

`List<Webcam> list = Webcam.getWebcams();
                if (list.size() > 0) {
                    MainPage.CamStatus = true;
                    lblCamR.setText("<html><font color='009900'>Connected!</font></html>");
                } else {
                    Refund.bool = false;
                    MainPage.CamStatus = false;
                    lblCamR.setText("<html><font color='990000'>Disconnected!</font></html>");
                }`

I am unsure if this is the method used for the picam as it isnt USB.

sarxos, 2018-10-15:
Oh, it won't work out-of-the-box if your device is not [UVC (USB Video Class)](https://en.wikipedia.org/wiki/USB_video_device_class).

In case you have a camera module you may try this:

https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

SynGreis, 2018-10-15:
i did 

$ sudo modprobe bcm2835-v4l2

but it couldnt find the directory at all. what would be the install command line for this? I suspect im missing a few installations.