sarxos/webcam-capture #661: Performance issue while using webcam-capture for clicking photo Ask

salmansaifi7, 2018-09-15:
My team is having a requirement to click picture during some workflow, we have integrated webcam-capture API to deal with that, But now we are facing performance issue (system is consuming more than 45% of memory and 30% of CPU).

Here is a code snippet which we are using (similar way):

import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamUtils;
import com.github.sarxos.webcam.ds.vlcj.VlcjDriver;
import com.github.sarxos.webcam.util.ImageUtils;

```
public class WebcamMain {

    public static void main(String[] args) throws InterruptedException {

        long i = 1;
        while (true) {
            try {
                System.out.println("Starting " + i + "th execution...");
                long startTime = System.currentTimeMillis();
                Webcam webcam = Webcam.getDefault();
                webcam.open();
                WebcamUtils.capture(webcam, "test100", ImageUtils.FORMAT_JPG);
                webcam.close();
                long endTime = System.currentTimeMillis();
                i++;
                System.out.println("Photo click time: " + (endTime - startTime) + " ms");
            } catch (Throwable th) {
                System.out.println("Exception occurred... Ignoring exception... " + th.getMessage());
                th.printStackTrace();
            } finally {
                System.out.println("Sleeping for 5 sec.");
                Thread.sleep(5000);
            }
        }
    }
}
```
Not sure why i am facing performance issue, tried changing GC.

sarxos, 2018-09-16:
This is because camera require some time to open. The open() method is time
consuming. If you want to avoid this you can open it at the very beginning
and then simply getImage().

salmansaifi7, 2018-09-17:
Thanks @sarxos , by calling webcam.open() in constructor of my class, i am able to control the memory usage :) , but it resulted in high CPU usage, process is now consuming more 120% of CPU. on enabling TRACE logs, it is refreshing FRAMES, i think that would be useful for capturing video use case, in my use case i just want snap during some flow in my application. is there any way to fix this? maybe controlling the speed of refreshing the FRAME of disabling frame refresh.

sarxos, 2018-09-17:
It's refreshing because there is a small buffer in a camera which holds
captured frames. Without refreshing you would get old frame when getImage()
is invoked, eg. when you did snap 5 minutes ago. The refresh rate can be
configured rtight now by using custom delay calculator, please see:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/Webcam.java#L279

And:

For more details.

sarxos, 2018-09-17:
And:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/WebcamUpdater.java

sarxos, 2018-09-19:
@salmansaifi7, 

By doing delay experiments in the past (something like 5 years ago) using default capture driver I concluded that there is some kind of underlying buffer in the native or hardware layer which holds up to 5 frames. Complete get image operation consist of two operation on the camera - first is a request to refresh frame (which puts frame into buffer), and then request to get it. The buffer works like a fifo queue. When you invoke get image operation you take the oldest frame from the buffer. Therefore, with 500 ms delay in refresh interval, and assuming that nothing has been changed in the native/hardware layer so the buffer still exist after these 5 years, when you do get image, you should get image which is ca. 2500 ms old.

In regards to the memory consumption. Java won't reclaim memory if there are references pointing the object (image buffer in this case). The native space, however, is governed by BridJ (Java to native binding based on JNI/JNA), which I'm not very familiar with, but from what I know is that whole native space should be released after Java object, bound to a native one, is finalized.

The chart you posted is a Java heap or a complete process memory reported by OS?

salmansaifi7, 2018-09-19:
Hi @sarxos ,
Thanks for quick reply.
can we talk directly if possible?

salmansaifi7, 2018-09-19:
this chart is for whole OS MEM consumption.
our process was consuming 700MB of that earlier, after making 25-30 calls, it start shooting up the memory consumption.

apart from 500ms DelayCalculator, i have introduced 500ms delay at following location:
https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/ds/buildin/WebcamDefaultDevice.java#L508