sarxos/webcam-capture #662: hi @MuhammadAmmad ,

sachinrke, 2018-10-08:
Thanks for the code. I have also tried your code and able to record video from multiple cameras. but this code not recording audio with video.so how can we record audio with video together from multiple cameras. please help me.
 i will be very grateful to you.

Thank You 
sachin pal

_Originally posted by @sachinrke in https://github.com/sarxos/webcam-capture/issue_comments#issuecomment-427737822_

sarxos, 2018-10-08:
Hi @sachinrke,

I never intended to add audio support to webcam capture because the main goal of this library is to access video feed from camera.

But if you are interested in audio support then please take a look at this code: https://github.com/ryancheu/webcam-capture/commit/25d63d6c822a5aa3483e22c0ebb16f889276b666, a modified live feed example which includes audio streaming.

sachinrke, 2018-10-08:
hi@sarxos,
Thanks for reply, however  i would like to intimate you that i have recorded audio with video but the problem is that
1.  audio comes in half video and half part is without audio 
2. when i am recording from multiple cameras it give me null pointer exception in method getAudioSamples(int fps) line audioLine.start();

i have send the code.
please help me where i am doing wrong. my knowledge in limited.
your help will be much appreciated.
Thank you.

Exception in thread "Thread-11" java.lang.NullPointerException
	at cam.WebcamViewerExample.getAudioSamples(WebcamViewerExample.java:275)
	at cam.WebcamViewerExample.saveClip(WebcamViewerExample.java:152)
	at cam.WebcamViewerExample.access$000(WebcamViewerExample.java:36)
	at cam.WebcamViewerExample$SaveClip.run(WebcamViewerExample.java:207) 

WebcamApp.java
if("Start Recording".equals(e.getActionCommand())){
            if(dw.availableUSBCameralist.get(0).isOpen() && !dw.availableUSBCameralist.get(1).isOpen()){
                new WebcamViewerExample.Capture(dw.availableUSBCameralist.get(0)).start();
            }
            if(!dw.availableUSBCameralist.get(0).isOpen() && dw.availableUSBCameralist.get(1).isOpen()){
                new WebcamViewerExample.Capture(dw.availableUSBCameralist.get(1)).start();
            }
            if(dw.availableUSBCameralist.get(0).isOpen() && dw.availableUSBCameralist.get(1).isOpen()){
                new WebcamViewerExample.Capture(dw.availableUSBCameralist.get(0)).start();
                 new WebcamViewerExample.Capture(dw.availableUSBCameralist.get(1)).start();
}
}

WebcamViewerExample.java
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;
import com.github.sarxos.webcam.ds.ipcam.IpCamDevice;
import com.xuggle.mediatool.IMediaWriter;
import com.xuggle.mediatool.ToolFactory;
import com.xuggle.xuggler.IAudioSamples;
import com.xuggle.xuggler.ICodec;
import com.xuggle.xuggler.IPixelFormat;
import com.xuggle.xuggler.IVideoPicture;
import com.xuggle.xuggler.video.ConverterFactory;
import com.xuggle.xuggler.video.IConverter;
import java.awt.Dimension;
import java.awt.image.BufferedImage;
import java.io.BufferedReader;
import java.io.File;
import java.io.FileNotFoundException;
import java.io.FileReader;
import java.io.IOException;
import java.text.SimpleDateFormat;
import java.util.ArrayList;
import java.util.Calendar;
import java.util.Date;
import javax.sound.sampled.AudioFormat;
import javax.sound.sampled.AudioSystem;
import javax.sound.sampled.DataLine;
import javax.sound.sampled.LineUnavailableException;
import javax.sound.sampled.TargetDataLine;

public class WebcamViewerExample {
    public static int record_time = 10; // recording time is 10s
    public static int fps = 30;
    public static final int CHANNEL_COUNT = 2;
    public static final int SAMPLE_RATE = 44100;
    public static final IAudioSamples.Format soundFormat = IAudioSamples.Format.FMT_S16;
    public static final int BYTE_DEPTH = soundFormat.swigValue() + 1;
    public static final int BIT_DEPTH = 8 * BYTE_DEPTH;
    public static int audioTime = 0;
    public static TargetDataLine audioLine = null;
    public static AudioFormat audioFormat = null;
[... 219 lines cut ...]
        if (fps <= 0)
            fps = 1;

        IAudioSamples samples = IAudioSamples.make(SAMPLE_RATE / fps, CHANNEL_COUNT, soundFormat);
       
        audioLine.start();
        
        byte[] data = new byte[(SAMPLE_RATE * BYTE_DEPTH * CHANNEL_COUNT) / fps];
        
        int sz = audioLine.read(data, 0, data.length);
        samples.put(data, 0, 0, sz);
        audioTime += (sz);
        samples.setComplete(true, sz / (BYTE_DEPTH * CHANNEL_COUNT), SAMPLE_RATE, CHANNEL_COUNT, soundFormat,
                audioTime / (BYTE_DEPTH * CHANNEL_COUNT));
        
        return samples;
    }

    public static void closeAudioRecorder() {
        if (audioLine != null) {
            audioLine.close();
            audioLine = null;
        }
    }
}