sarxos/webcam-capture #666: Cannot access native resolution

cosmin-l, 2018-11-07:
I have a Logitech c920 webcam that supports natively FHD 1920 x 1080 resolution.
When I try to initialize it to this resolution (or even HD):

```
webcam.setViewSize(WebcamResolution.FHD.getSize());
```
I get the error :

```
java.lang.IllegalArgumentException: Incorrect dimension [1920x1080] possible ones are [176x144] [320x240] [640x480] 
	at com.github.sarxos.webcam.Webcam.setViewSize(Webcam.java:622)
```

I've tried several drivers for webcam-capture and also tried it in Windows 10 and Linux (Ubuntu 18.10)

Any idea if the Webcam can access FHD resolutions or is there a problem with my camera ?

dariyahussain220, 2018-11-09:
try to get sizes from webcam

webcam.getViewSizes();
set highest resolution available for your device -as a webcam dimension like this 

webcam.setViewSize(webcam.getViewSizes()[webcam..getViewSizes().length - 1]);

it will automatically takes your webcam max resolution size.
hope it will work.

eduramiba, 2021-06-17:
Hi,

Please try the new native driver that I just released: https://github.com/eduramiba/webcam-capture-driver-native

It should be able to correctly list all device resolutions.