sarxos/webcam-capture #671: GStreamer 1.X - jar release

Rado9, 2018-11-19:
Hi,
Are they any plans to release jar for GStreamer 1.X? 
https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-gst1
At least beta for testing?

sarxos, 2018-11-19:
Here you are :) https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture-driver-gst1&v=0.3.13-SNAPSHOT

sarxos, 2018-11-19:
It's from 2018-06-18. Please let me know if you need newer one.

Rado9, 2018-11-20:
Hi,
thanks. :)
I really appreciate your help.
I linked these libraries to my test project:
bridj-0.7-20140918.jar
gst1-java-core-0.9.3.jar
jna-5.1.0.jar
jna-platform-5.1.0.jar
webcam-capture-0.3.13.jar
webcam-capture-driver-gst1-0.3.13.jar

But I have this problem:

Exception in thread "main" java.lang.NoSuchFieldError: SIZE
	at org.freedesktop.gstreamer.Gst$NativeArgs.<init>(Gst.java:87)
	at org.freedesktop.gstreamer.Gst.init(Gst.java:292)
	at com.github.sarxos.webcam.ds.gst1.Gst1Driver.init(Gst1Driver.java:61)
	at com.github.sarxos.webcam.ds.gst1.Gst1Driver.<init>(Gst1Driver.java:48)
	at sk.lh.unit.CUnit.<clinit>(CUnit.java:272)

static {
	Webcam.setDriver(new Gst1Driver());  ... this is line CUnit.java:272
}

Can you help me please?

Rado9, 2018-11-20:
I have tried
gst1-java-core-0.9.0.jar
gst1-java-core-0.9.1.jar
gst1-java-core-0.9.2.jar
gst1-java-core-0.9.3.jar
from
https://github.com/gstreamer-java/gst1-java-core/releases
but with no success.

Rado9, 2018-11-20:
Problem was in libraries
jna-5.1.0.jar
jna-platform-5.1.0.jar

correct:
jna-4.5.2.jar
jna-platform-4.5.2.jar

Rado9, 2018-11-20:
I am able to do photo via command:
gst-launch-1.0 imxv4l2src device=/dev/video0 num_buffers=1 ! video/x-raw,format=UYVY,width=1920,height=1080,framerate=30/1 ! jpegenc quality=100 idct-method=2 ! filesink location=picture.jpeg

But my java application stuck on webcam.open()
webcam = Webcam.getWebcamByName("/dev/video0");
webcam.open();

Rado9, 2018-11-21:
My colleague found out that issue is in files
/ds/gs1/Gst1Device.java
/ds/gs1/impl/GsUtils.java    

case LINUX:
return "v4l2src";

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-gst1/src/main/java/com/github/sarxos/webcam/ds/gst1/Gst1Device.java

final Element source = ElementFactory.make("v4l2src", "source");

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-gst1/src/main/java/com/github/sarxos/webcam/ds/gst1/impl/GsUtils.java

We need ‘’imxv4l2src’’ instead ‘’v4l2src’’.
Our PC is TechNexion i.Core 1.5 MX6 Quad with OS Linux pico-imx6 4.1.15-2.0.0-technexion+g08c3634 #1 SMP PREEMPT Tue Oct 30 22:18:51 CET 2018 armv7l armv7l armv7l GNU/Linux

closed 2018-11-30 by Rado9