sarxos/webcam-capture #676: Library 'OpenIMAJGrabber' was not found in path

liubinduo, 2018-12-10:
/Library/Java/JavaVirtualMachines/jdk-11.0.1.jdk/Contents/Home/bin/java -agentlib:jdwp=transport=dt_socket,address=127.0.0.1:49948,suspend=y,server=n --add-opens webcam.capture/com.github.sarxos.webcam.ds.buildin.natives=ALL-UNNAMED --add-modules javafx.base,javafx.graphics --add-reads javafx.base=ALL-UNNAMED --add-reads javafx.graphics=ALL-UNNAMED -javaagent:/Users/liubinduo/Library/Caches/IntelliJIdea2018.3/captureAgent/debugger-agent.jar -Dfile.encoding=UTF-8 -classpath "/Users/liubinduo/Library/Application Support/JetBrains/Toolbox/apps/IDEA-U/ch-0/183.4284.148/IntelliJ IDEA.app/Contents/lib/idea_rt.jar" -p /Users/liubinduo/.m2/repository/org/openjfx/javafx-base/11/javafx-base-11-mac.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-graphics/11/javafx-graphics-11-mac.jar:/Users/liubinduo/Documents/open-source/TestJavaFXAndMaven/target/classes:/Users/liubinduo/.m2/repository/org/openjfx/javafx-controls/11/javafx-controls-11.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-controls/11/javafx-controls-11-mac.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-graphics/11/javafx-graphics-11.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-base/11/javafx-base-11.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-fxml/11/javafx-fxml-11.jar:/Users/liubinduo/.m2/repository/org/openjfx/javafx-fxml/11/javafx-fxml-11-mac.jar:/Users/liubinduo/.m2/repository/com/github/sarxos/webcam-capture/0.3.12/webcam-capture-0.3.12.jar:/Users/liubinduo/.m2/repository/com/nativelibs4java/bridj/0.7.0/bridj-0.7.0.jar:/Users/liubinduo/.m2/repository/org/slf4j/slf4j-api/1.7.2/slf4j-api-1.7.2.jar:/Users/liubinduo/.m2/repository/com/github/sarxos/webcam-capture-driver-openimaj/0.3.12/webcam-capture-driver-openimaj-0.3.12.jar:/Users/liubinduo/.m2/repository/org/openimaj/core-video-capture/1.3/core-video-capture-1.3.jar:/Users/liubinduo/.m2/repository/org/openimaj/core-video/1.3/core-video-1.3.jar:/Users/liubinduo/.m2/repository/org/openimaj/core-image/1.3/core-image-1.3.jar:/Users/liubinduo/.m2/repository/net/sourceforge/jeuclid/jeuclid-core/3.1.9/jeuclid-core-3.1.9.jar:/Users/liubinduo/.m2/repository/commons-logging/commons-logging/1.1.1/commons-logging-1.1.1.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-svg-dom/1.7/batik-svg-dom-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-anim/1.7/batik-anim-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-awt-util/1.7/batik-awt-util-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-css/1.7/batik-css-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-dom/1.7/batik-dom-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-xml/1.7/batik-xml-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-parser/1.7/batik-parser-1.7.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/batik-util/1.7/batik-util-1.7.jar:/Users/liubinduo/.m2/repository/xml-apis/xml-apis-ext/1.3.04/xml-apis-ext-1.3.04.jar:/Users/liubinduo/.m2/repository/org/apache/xmlgraphics/xmlgraphics-commons/1.3.1/xmlgraphics-commons-1.3.1.jar:/Users/liubinduo/.m2/repository/org/openimaj/core/1.3/core-1.3.jar:/Users/liubinduo/.m2/repository/com/google/guava/guava/14.0.1/guava-14.0.1.jar:/Users/liubinduo/.m2/repository/log4j/log4j/1.2.14/log4j-1.2.14.jar:/Users/liubinduo/.m2/repository/commons-lang/commons-lang/2.6/commons-lang-2.6.jar:/Users/liubinduo/.m2/repository/commons-io/commons-io/2.2/commons-io-2.2.jar:/Users/liubinduo/.m2/repository/org/apache/commons/commons-vfs2/2.0/commons-vfs2-2.0.jar:/Users/liubinduo/.m2/repository/org/apache/maven/scm/maven-scm-api/1.4/maven-scm-api-1.4.jar:/Users/liubinduo/.m2/repository/org/codehaus/plexus/plexus-utils/1.5.6/plexus-utils-1.5.6.jar:/Users/liubinduo/.m2/repository/org/apache/maven/scm/maven-scm-provider-svnexe/1.4/maven-scm-provider-svnexe-1.4.jar:/Users/liubinduo/.m2/repository/org/apache/maven/scm/maven-scm-provider-svn-commons/1.4/maven-scm-provider-svn-commons-1.4.jar:/Users/liubinduo/.m2/repository/regexp/regexp/1.3/regexp-1.3.jar:/Users/liubinduo/.m2/repository/commons-httpclient/commons-httpclient/3.1/commons-httpclient-3.1.jar:/Users/liubinduo/.m2/repository/commons-codec/commons-codec/1.2/commons-codec-1.2.jar:/Users/liubinduo/.m2/repository/org/openimaj/core-math/1.3/core-math-1.3.jar:/Users/liubinduo/.m2/repository/org/la4j/la4j/0.4.0/la4j-0.4.0.jar:/Users/liubinduo/.m2/repository/org/openimaj/MatrixLib/1.3/MatrixLib-1.3.jar:/Users/liubinduo/.m2/repository/org/apache/commons/commons-math3/3.3/commons-math3-3.3.jar:/Users/liubinduo/.m2/repository/org/apache/sanselan/sanselan/0.97-incubator/sanselan-0.97-incubator.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/imageio/imageio-jpeg/3.0-rc5/imageio-jpeg-3.0-rc5.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/imageio/imageio-core/3.0-rc5/imageio-core-3.0-rc5.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/imageio/imageio-metadata/3.0-rc5/imageio-metadata-3.0-rc5.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/common/common-io/3.0-rc5/common-io-3.0-rc5.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/common/common-lang/3.0-rc5/common-lang-3.0-rc5.jar:/Users/liubinduo/.m2/repository/com/twelvemonkeys/common/common-image/3.0-rc5/common-image-3.0-rc5.jar:/Users/liubinduo/.m2/repository/org/openimaj/core-audio/1.3/core-audio-1.3.jar:/Users/liubinduo/.m2/repository/org/openimaj/content/animation/1.3/animation-1.3.jar -m TestJavaFXAndMaven/com.v1ok.eprint.Main
Connected to the target VM, address: '127.0.0.1:49948', transport: 'socket'
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.io.FileNotFoundException: Library 'OpenIMAJGrabber' was not found in path '[., /Users/liubinduo/.sdkman/candidates/java/current/bin, /usr/local/bin, /usr/bin, /bin, /usr/sbin, /sbin, /Library/Java/Extensions, /System/Library/Java/Extensions, ., /Library/Java/JavaVirtualMachines/jdk-11.0.1.jdk/Contents/Home/lib, /Library/Java/JavaVirtualMachines/jdk-11.0.1.jdk/Contents/Home/bin, /usr/lib, /usr/local/lib]'
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1074)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1022)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:904)
	at bridj@0.7.0/org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
	at bridj@0.7.0/org.bridj.BridJ.initialize(BridJ.java:1128)
	at bridj@0.7.0/org.bridj.NativeObject.<init>(NativeObject.java:50)
	at bridj@0.7.0/org.bridj.StructObject.<init>(StructObject.java:46)
	at bridj@0.7.0/org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)

msgilligan, 2019-04-29:
I'm seeing something similar. @liubinduo Are you running a modular application?

kayc33, 2019-12-27:
@msgilligan Have you figure it out?