sarxos/webcam-capture #693: Webcam does not Work on RaspberryPi 3B

weltenbranddesign, 2019-02-19:
Hello Sarxos i programmed a Photobox with Java 9 on Windows and want to run it now on a RaspberryPi3 with a RaspiCamv2.1. On my Laptop everthing runs perfectly but on the raspberrypi there are many Errors.

I tried to implement your V4l4j Driver into my code but i got still an Error here ist the latest Error i hope you can help me.

`LF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: com/github/sarxos/v4l4j/V4L4J
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.<clinit>(V4l4jDriver.java:33)
	at FotoBoxHolder.<init>(FotoBoxHolder.java:40)
	at FrameHolder$1.actionPerformed(FrameHolder.java:42)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:269)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6578)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3343)
	at java.desktop/java.awt.Component.processEvent(Component.java:6343)
	at java.desktop/java.awt.Container.processEvent(Container.java:2259)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4961)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2317)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4793)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4539)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4480)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2303)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2758)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4793)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:766)
	at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:99)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:739)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:736)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.ClassNotFoundException: com.github.sarxos.v4l4j.V4L4J
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:582)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:185)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:496)
	... 39 more

`

weltenbranddesign, 2019-02-19:
This is the Code where the Error comes from

`import com.github.sarxos.webcam.Webcam;

import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

import javax.imageio.ImageIO;
import javax.swing.*;

import java.awt.*;
import java.awt.event.ActionEvent;
import java.awt.event.ActionListener;
import java.io.BufferedInputStream;
import java.io.File;
import java.io.FileInputStream;
import java.io.IOException;
import java.util.Properties;
import java.util.TimerTask;
import java.util.Timer;

public class FotoBoxHolder
{

    /*Diese Klasse integriert die Libs WebCam und Timer */
    private final JFrame frame;

    //Webcam Objekt aus der Libery Webcam
    private Webcam webcam;
    private JButton snapbut;
    // Timer wird verwendet um das Foto nicht direkt auszulösen
    private Timer timer = new Timer();
    private TimerTask timerTask;
    private int countdown = 5;
    private File cap;

    public FotoBoxHolder() throws IOException {

        System.setProperty("org.slf4j.simpleLogger.defaultLogLEvel","info");
        System.setProperty("org.slf4j.simpleLogger.log.com.github.sarxos.webcam.ds.v4l4j","trace");
        webcam.setDriver(new V4l4jDriver());
        /*Die Propertie Datei enthält die Hauptkonfiguration für das übertragen der Facebook und Twitterdaten*/
        Properties probs = new Properties();
        BufferedInputStream stream = new BufferedInputStream(new FileInputStream("config.properties"));
        probs.load(stream);
        stream.close();

        //WebCam wird gewählt mit getDefault() erhält man die Standart Webcam des Systems
        webcam = Webcam.getDefault();
            //WebCam Auflösung wird definiert (Wichtig! WebCams haben alle andere Auflösungen!!!)
        /*webcam.setDriver(new V4l4jDriver());*/
            /*webcam.setDriver(new V4l4jDriver());*/
        webcam.setViewSize(new Dimension(640,480));
        //WebCam Stream wird geöffnet
        webcam.open();

        //Frame hat eine Größe von 800 x 580 bezogen auf die Auflösung des 7" Touchdisplays
        frame = new JFrame("FotoBox 0.1 ALPHA");
        frame.setPreferredSize(new Dimension(800,580));
        frame.setResizable(true);

        //Beginn des Optionen Menus
        JMenuBar menuBar = new JMenuBar();

        JMenu menuopt = new JMenu("Optionen");

        // Beginn der UnterMenus
        menuBar.add(menuopt);
        JMenuItem facebook = new JMenuItem("Allgemeine Optionen");
        ActionListener optFacebook = new ActionListener() {
[... 91 lines cut ...]
            }

        };
        snapbut.addActionListener(takesnap);

        butpan.add(snapbut);

        frame.add(vidpan, BorderLayout.NORTH);
        frame.add(butpan, BorderLayout.SOUTH);

        machSchick();

    }

    private void machSchick()
    {
        frame.setExtendedState(JFrame.MAXIMIZED_BOTH);
        frame.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
        frame.setLocationRelativeTo(null);
        frame.pack();
        frame.setVisible(true);
    }

}
`

weltenbranddesign, 2019-02-19:
on my raspberry pi iam using Rasbian

yuripourre, 2019-02-19:
Hey @weltenbranddesign take a look at: https://github.com/sarxos/webcam-capture/wiki/How-To-Configure-Raspberry-Pi

Tell me if you have any progress

weltenbranddesign, 2019-02-20:
Hello now i got this Error

WARNING: Illegal reflective access by com.github.sarxos.v4l4j.NativeUtils (file:/home/pi/Dokumente/FotoBox.jar) to field java.lang.ClassLoader.usr_paths
WARNING: Please consider reporting this to the maintainers of com.github.sarxos.v4l4j.NativeUtils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: /tmp/libs5568890871990368718/libvideo.so: libjpeg.so.8: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
	at java.base/java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2614)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2531)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:873)
	at java.base/java.lang.System.loadLibrary(System.java:1857)
	at com.github.sarxos.v4l4j.NativeUtils.loadLibraryFromJar(NativeUtils.java:98)
	at com.github.sarxos.v4l4j.V4L4J.init(V4L4J.java:50)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.<clinit>(V4l4jDriver.java:33)
	at FotoBoxHolder.<init>(FotoBoxHolder.java:42)
	at FrameHolder$1.actionPerformed(FrameHolder.java:51)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:269)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6578)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3343)
	at java.desktop/java.awt.Component.processEvent(Component.java:6343)
	at java.desktop/java.awt.Container.processEvent(Container.java:2259)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4961)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2317)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4793)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4539)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4480)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2303)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2758)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4793)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:766)
	at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:99)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:739)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:737)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:736)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

yuripourre, 2019-02-20:
Sorry, I have no clue on how to solve it. Maybe @sarxos could help you with that.

Anyway, would you mind to try the ffmpeg driver? It was told that works with Raspberry Pi.

weltenbranddesign, 2019-02-21:
@sarxos can you help me?

weltenbranddesign, 2019-02-21:
Hello i get it working know the libjpeg8.so was not installed!

but know i got this error if i want to take a picture

WT-EventQueue-0] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
OpenJDK Server VM warning: You have loaded library /tmp/BridJExtractedLibraries1957399332863931830/libbridj.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
device doesn't support capture
device doesn't support capture
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam mmal service 16.1 /dev/video0
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
device doesn't support capture
javax.imageio.IIOException: Can't read input file!
	at java.desktop/javax.imageio.ImageIO.read(ImageIO.java:1308)
	at Preview.addImageWatermark(Preview.java:253)
	at Preview.<init>(Preview.java:57)
	at FotoBoxHolder$2$1.run(FotoBoxHolder.java:155)
	at java.base/java.util.TimerThread.mainLoop(Timer.java:556)
	at java.base/java.util.TimerThread.run(Timer.java:506)
device doesn't support capture
device doesn't support capture
[shutdown-hook-1] INFO com.github.sarxos.webcam.WebcamShutdownHook - Automatic mmal service 16.1 /dev/video0 deallocation
[shutdown-hook-1] INFO com.github.sarxos.webcam.Webcam - Disposing webcam mmal service 16.1 /dev/video0

alexmao86, 2019-02-22:
hi all,
I know many people are struggle with raspberrypi usage. please consider use my webcam-capture-driver-raspberrypi/RaspividDriver and RaspividYUVDriver. it is very good in performance. because camera is consuming GPU and java side is really lighweight.

stamanker, 2019-08-17:
@alexmao86 how to use it?

stamanker, 2019-08-17:
@alexmao86 how to install it?

alexmao86, 2019-08-18:
It is very simple, just follow https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-raspberrypi readme to begin. the key statements is: 

> 	static {
> 		Webcam.setDriver(new RaspiYUVDriver());
> 	}

before above code, there is another problem that this driver is not in central repo, so you need to compile the sub-project.