sarxos/webcam-capture #700: Program shut down when closing a webcam

nestormariomace, 2019-03-06:
Hello.
I'm programming an application that requires to read 3 images from 3 webcams, store them in a disk as a "jpg", and store an additional file in the disk once each aprox. 15 seconds.

I'm working in a PC with Windows 10 and developing the program with Netbeans 7.3.1; jdk1.7.

The application is a do-forever that opens one webcam, then reads the image, then closes the webcam.
When running this application, it shuts down in a random way with a message in the netbeans output
window:
    Java result -1073741819 
The problem arises when the program tries to close the webcam.
I tried to get the simplest program that presents the problem:

                while (true){                        
                    for (int k = 0; k < webcams.size(); k++){
                        try{
                            webcam1 = webcams.get(k);
                            if ( ! webcam1.isOpen()){
                                webcam1.setCustomViewSizes(size);
                                webcam1.setViewSize(size[0]);
                                webcam1.open(false);                                    
                            }
                            webcam1.close();
                         }
                        catch(Exception ee){
                            System.out.println("xxxx.");
                        }
                    }                        
                }
This program shuts down after several minutes with the above mentioned message.
Can you please help me?

hooq77, 2019-07-31:
I hava a similar problem with Process finished with exit code -1073740940，do you resolve this yet?

stalactiteDong, 2020-11-20:
兄弟怎么解决的,我的也是关闭摄像头直接给我程序干停了