sarxos/webcam-capture #711: Can I use thos to turn on flashlight on windows 10?

benag, 2019-04-28:


sarxos, 2019-04-29:
Hi. No, unfortunately, there is no such possibility.

benag, 2019-04-29:
Then I will write some native code in c#, but can you explain why is not possible to implement in JAVA?

sarxos, 2019-04-29:
@benag,

I never said it's not possible to implement this in Java. Your question was
clearly directed towards using my Webcam Capture API (this project). A
Webcam Capture API utilizes basic features from UVC device class (
https://en.wikipedia.org/wiki/USB_video_device_class). These features are
exposed by a capture drivers you configure to use (e.g. OpenIMAJ,
GStreamer, OpenCV, etc). The webcam light, however, is not a standard
webcam feature and different vendors may implement it in a different way.
It can be either controlled by a USB side channel or it can be a custom UVC
property. If your webcam exposes it as a property then you can try to use
e.g. gstreamer-java to access it with a probe, but if it's controlled by
USB side channel, then it may be required to write small portion of a code
to communicate with it directly via USB. This could probably be done with
libusb-java or similar. My Webcam Capture API does not support probes and
custom USB communication. It's meant to be simple, and if we try to cover
all possible webcam-related features, then it would become extremely
complicated.

sarxos, 2019-05-10:
Hi @benag, I'm closing this one. Please re-open if you have additional questions in regards to this matter.

closed 2019-05-10 by sarxos